wmbusmeters/wmbusmeters #1293: QUNDIS water meter: meter detection did not match the selected driver qwater! correct driver is: unknown!

c7n, 2024-05-29:
### Type of request

Bug

### OS version

HA

### wmbusmeters version

1.16.1

### Your message goes here

Thanks for this great add-on in HA!!
I think, I have a similar issue with the driver detection for a QUNDIS water meter. 

I got following error:
(meter) water_51220588: meter detection did not match the selected driver qwater! correct driver is: unknown!
(meter) Not printing this warning again for id: 51220588 mfct: (QDS) Qundis, Germany (0x4493) type: Water meter (0x07) ver: 0x19
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|4944934488052251190778_0DFF5F350082930000810007C113FFFF91670400FF2C265402001E34332204000000EE00F201A501DB01C1015401B70178019701B901C9012F046D06091D35|+162

But if I analyze the telegram (https://wmbusmeters.org/analyze/49449344880522511907780DFF5F350082930000810007C113FFFF91670400FF2C265402001E34332204000000EE00F201A501DB01C1015401B70178019701B901C9012F046D06091D35) it says, that the auto driver was not found but the "best driver" and the "used driver" qwater. With this driver, you see the correct data in the analyzing tool. However in the HA add-on, when configuring the water meter with driver qwater, I get the above mentioned error. I am doing something wrong or is that an issue?

Thank you and if you need further assistance / info, just let me know.

Greetings, Christian

weetmuts, 2024-05-29:
The meters are continuously improved by the manufacturers and when they create new versions they update 
the mfct version type (mvt) triplet sent in the telegram. If you hardcode a driver, wmbusmeters will use it for the telegram, however if the mvt triplet is unknown, then wmbusmeters will warn. 

A new mvt can mean that there is new data in the telegram, so that the driver can be improved.
In this case the telegram is proprietary, so we are happy with the values that are correct already. 
I added the new mvt triplet and your telegram as a new test case.

Many thanks!

You can see the change here:
https://github.com/wmbusmeters/wmbusmeters/commit/b3771a9b8e5ee72217a01370956d8923f54f4e78

And you can now analyze from the web:

c7n, 2024-05-29:
Hey weetmuts, thank you very much! I 
really appreciate your engagement!

closed 2024-08-28 by github-actions[bot]