wmbusmeters/wmbusmeters #1296: TAUROM Amiplus update

Mice30, 2024-06-10:
### Type of request

Existing device improvement

### Meter description

Hi,

I just added my Tauron energy meter to Home Assistant using the HA add-on. After entering the decryption key, I received a log message indicating that the Amiplus driver is not suitable for my meter. Some data is present, but not all the data that the telegram contains.

Can you help, please? :)

### Logged telegrams from meter in decrypted format

```shell
telegram=|8E4401061764695601027AE01380052F2F_066DC04B0E0A36800C78176469560AFDC9FC0143020AFDC9FC0243020AFDC9FC0343028E30833C0000000000008E20833C0000000000008E10833C3202000000000BABC8FC100000008E10030648510200008E20030000000000008E30030000000000000B2B9502000BAB3C0000002F2F2F2F2F2F2F2F2F2F2F2F2F2F2F|
```

wrobelda, 2024-06-30:
https://github.com/wmbusmeters/wmbusmeters/pull/1231

wrobelda, 2024-06-30:
You probably have the new generation of AMIPlus meters installed. I previously added a [PR](https://github.com/wmbusmeters/wmbusmeters/pull/1231) to have wmbusmeters actually recognize it as amiplus meter, although as of recent version this is no longer needed for whatever reason. 

Still, it would be nice to have the extra information decrypted as well, many of the basic parameters are currently missing.

pswiatki, 2024-06-30:
But wait! I have NES 73331 - no problems with extra parameters.

wrobelda, 2024-06-30:
> But wait! I have NES 73331 - no problems with extra parameters.

I can't recall, actually, it's been a few months since I last checked this, but my voltage is reported as Phase 3 and I think some tariff information was missing, too. But frankly, I may be wrong here, I didn't investigate this too much as I was after the energy consumption only.

Mice30, 2024-07-01:
It's great to see some feedback! I'll post the type of meter I have later on.

wrobelda, 2024-07-03:
See also this bug report:

https://github.com/wmbusmeters/wmbusmeters/issues/1230

abht234, 2024-08-02:
I can't get my NES 73331 meter to work. I am using the decryption key provided by the meter supplier. Any help would be greatly appreciated. 

```
(meter) myenergy: meter detection did not match the selected driver amiplus! correct driver is: unknown!
(meter) Not printing this warning again for id: 00024595 mfct: (NES) NORA ELK MALZ SAN ve TIC, Turkey (0x38b3) type: Electricity meter (0x02) ver: 0x03
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) myenergy: yes for me
(meter) myenergy(2) amiplus  handling telegram from 00024595
(meter) myenergy 00024595 "8E44B3389545020003027A09008005D6071B712CA9988FCA0CEEB9F412ED456C8FCCCF03795CA70652D6253AC8A5E09C651F75E9EE75BF04870AF71F9CFD8037788EF7E08C6043A9E538ABF9879537F38AB436F685502A057C5E462F4A151718BE525AB7C00F88C067E462C4B65AA79135167430250CE7EC8AB2A616A3518D7E0C5A86784229107AC087F871C2C45F"
(wmbus) parseDLL @0 143
(telegram) DLL L=8e C=44 (from meter SND_NR) M=38b3 (NES) A=00024595 VER=03 TYPE=02 (Electricity meter) (driver unknown!) DEV=rtlwmbus[cmd_0] RSSI=-74
(wmbus) parseELL @10 133
(wmbus) parseNWL @10 133
(wmbus) parseAFL @10 133
(wmbus) parseTPL @10 133
(TPL) num encrypted blocks 8 (128 bytes and remaining unencrypted 0 bytes)
(TPL) AES CBC IV decrypting "D6071B712CA9988FCA0CEEB9F412ED456C8FCCCF03795CA70652D6253AC8A5E09C651F75E9EE75BF04870AF71F9CFD8037788EF7E08C6043A9E538ABF9879537F38AB436F685502A057C5E462F4A151718BE525AB7C00F88C067E462C4B65AA79135167430250CE7EC8AB2A616A3518D7E0C5A86784229107AC087F871C2C45F"
(TPL) IV B3389545020003020909090909090909
(TPL) decrypted  "6ED28C238B831EB72A5FBBA52944834566E447219F234C87BC9AE3BDC3B3AD24487EF51E09AF7DAB1F603BAF3DF0D6C245C5F74E9564986A7FB30B6A8F2B465A21C0525360623ECF47E68D23CDAA8528001E47781513C355B378FA918081798F53E60BDE7B7E679505F70422FF3B0547D0E6B7F61778841C6D6AA8E979D9020F"
(wmbus) WARNING!! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 00024595 mfct: (NES) NORA ELK MALZ SAN ve TIC, Turkey (0x38b3) type: Electricity meter (0x02) ver: 0x03
(telegram) TPL CI=7a ACC=09 STS=00 CFG=0580 (AES_CBC_IV nb=8 cntn=0 ra=0 hc=0)
(wmbus) telegram from 00024595 ignored by all configured meters!
```

Leo-PL, 2024-08-21:
I've got the same complaint for Apator Otus 3, but the decoding process just works otherwise, and correct data is reported.

wrobelda, 2024-08-21:
I am wild guessing that the encryption is part of the envy's standard and indifferent to the underlying device type. Therefore if the software complains about failing to decrypt a message, you almost certainly assumed a wrong device ID as your own meter.

Leo-PL, 2024-08-21:
Nope, not the case. The data matches and also (kinda) does the ID - endianness is switched w.r.t. what I got in Tauron's system. Yes, they finally fixed the online activation.

kitor, 2024-09-13:
Same here - Apator Otus 1, complaint about unknown driver but it works. In my case ID on Tauron site was also screwed / wrong endianess - I found that grepping "supposed ID" over rtl_sdr dump.

```
[2024-09-13_20:57:25] (meter) power: meter detection did not match the selected driver amiplus! correct driver is: unknown!
(meter) Not printing this warning again for id: 55090884 mfct: (APA) Apator, Poland (0x601) type: Electricity meter (0x02) ver: 0x01
[2024-09-13_20:57:25] (meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
[2024-09-13_20:57:25] (meter) to add support for this unknown mfct,media,version combination
```

Seems to be reported as exactly the same type as three phase Opus 3 @Leo-PL has. Thus I guess it can be safely added into auto-detection.

kitor, 2024-09-14:
Just made PR #1362 that includes changes to amiplus code for Opus 1/3 support. I see OP @Mice30 never returned to us with answer about what meter was in the original report - but I see his telegram successfully decodes via amiplus driver and matches APA/0x02/0x01 introduced in that PR so it should work fine.

Leo-PL, 2024-11-14:
I can confirm it works now with my meter too.

kitor, 2024-11-14:
@weetmuts since OP doesn't seem to care and we got independent confirmation my earlier changes work on Opus 1/3 I guess this can be closed.

dslynko, 2024-11-24:
I would add more details. Though all (or almost all, idk) seems is decoded correctly, the main issue I've bumped into is the wrong ID in decoded data, which causes issues - wmbusmeters ignores decoded telegram "not for me", since meter is configured to use another ID. Here is the example:

Decoded datagram:

`
Auto driver    : not found!
Similar driver : amiplus 54/61
Using driver   : amiplus 54/61
000   : 8e length (142 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b338 dll-mfct (NES)
004   : 61910700 dll-id (00079161)
008   : 03 dll-version
009   : 02 dll-type (Electricity meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 09 tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 8005 tpl-cfg 0580 (AES_CBC_IV nb=8 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 06 dif (48 Bit Integer/Binary Instantaneous value)
018   : 6D vif (Date and time type)
019 C!: 390C17173B2F ("device_date_time":"2024-11-23 23:12:57")
025   : 0C dif (8 digit BCD Instantaneous value)
026   : 78 vif (Fabrication no)
027 C?: 61910700
031   : 0A dif (4 digit BCD Instantaneous value)
032   : FD vif (Second extension FD of VIF-codes)
033   : C9 vife (10^0 Volts)
034   : FC combinable vif (CombinableVIFExtension)
035   : 01 combinable extension vife (AtPhase1)
036 C!: 4102 ("voltage_at_phase_1_v":241)
038   : 0A dif (4 digit BCD Instantaneous value)
039   : FD vif (Second extension FD of VIF-codes)
040   : C9 vife (10^0 Volts)
041   : FC combinable vif (CombinableVIFExtension)
042   : 02 combinable extension vife (AtPhase2)
043 C!: 4202 ("voltage_at_phase_2_v":242)
045   : 0A dif (4 digit BCD Instantaneous value)
046   : FD vif (Second extension FD of VIF-codes)
047   : C9 vife (10^0 Volts)
048   : FC combinable vif (CombinableVIFExtension)
049   : 03 combinable extension vife (AtPhase3)
050 C!: 4102 ("voltage_at_phase_3_v":241)
052   : 8E dif (12 digit BCD Instantaneous value)
053   : 30 dife (subunit=0 tariff=3 storagenr=0)
054   : 83 vif (Energy Wh)
055   : 3C combinable vif (BackwardFlow)
056 C!: 000000000000 ("total_energy_production_tariff_3_kwh":0)
062   : 8E dif (12 digit BCD Instantaneous value)
063   : 20 dife (subunit=0 tariff=2 storagenr=0)
064   : 83 vif (Energy Wh)
065   : 3C combinable vif (BackwardFlow)
066 C!: 000000000000 ("total_energy_production_tariff_2_kwh":0)
072   : 8E dif (12 digit BCD Instantaneous value)
073   : 10 dife (subunit=0 tariff=1 storagenr=0)
074   : 83 vif (Energy Wh)
075   : 3C combinable vif (BackwardFlow)
076 C!: 000000000000 ("total_energy_production_tariff_1_kwh":0)
082   : 0B dif (6 digit BCD Instantaneous value)
083   : AB vif (Power W)
084   : C8 combinable vif (UpperLimit)
085   : FC combinable vif (CombinableVIFExtension)
086   : 10 combinable extension vife (AccumulationOfAbsoluteValue)
087 C?: 000000
090   : 8E dif (12 digit BCD Instantaneous value)
091   : 10 dife (subunit=0 tariff=1 storagenr=0)
092   : 03 vif (Energy Wh)
093 C!: 929615000000 ("total_energy_consumption_tariff_1_kwh":159.692)
099   : 8E dif (12 digit BCD Instantaneous value)
100   : 20 dife (subunit=0 tariff=2 storagenr=0)
101   : 03 vif (Energy Wh)
102 C!: 000000000000 ("total_energy_consumption_tariff_2_kwh":0)
108   : 8E dif (12 digit BCD Instantaneous value)
109   : 30 dife (subunit=0 tariff=3 storagenr=0)
110   : 03 vif (Energy Wh)
111 C!: 000000000000 ("total_energy_consumption_tariff_3_kwh":0)
117   : 0B dif (6 digit BCD Instantaneous value)
118   : 2B vif (Power W)
119 C!: 910300 ("current_power_consumption_kw":0.391)
122   : 0B dif (6 digit BCD Instantaneous value)
123   : AB vif (Power W)
124   : 3C combinable vif (BackwardFlow)
125 C!: 000000 ("current_power_production_kw":0)
128   : 2F skip
129   : 2F skip
130   : 2F skip
131   : 2F skip
132   : 2F skip
133   : 2F skip
134   : 2F skip
135   : 2F skip
136   : 2F skip
137   : 2F skip
138   : 2F skip
139   : 2F skip
140   : 2F skip
141   : 2F skip
142   : 2F skip

{
    "media":"electricity",
    "meter":"amiplus",
    "name":"",
    "id":"00079161",
    "current_power_consumption_kw":0.391,
    "current_power_production_kw":0,
    "total_energy_consumption_tariff_1_kwh":159.692,
    "total_energy_consumption_tariff_2_kwh":0,
    "total_energy_consumption_tariff_3_kwh":0,
    "total_energy_production_tariff_1_kwh":0,
    "total_energy_production_tariff_2_kwh":0,
    "total_energy_production_tariff_3_kwh":0,
    "voltage_at_phase_1_v":241,
    "voltage_at_phase_2_v":242,
    "voltage_at_phase_3_v":241,
    "device_date_time":"2024-11-23 23:12:57",
    "timestamp":"2024-11-24T11:38:55Z"
}

Using: wmbusmeters: 1.17.1-29-g3af7a4c
`

The problem is at position 027 C?: 61910700 - this is the port (=ID) of my Tauron energy meter NES 73331, however, json contains:  "id":"00079161", which is not equal to the configuration:

`
name=amiplus-energy
type=amiplus
## This is the real ID from TAURON eLicznik web page
#id=61910700
## This is ID decoded from the datagram, might be fixed soon
id=00079161
key=*****
`
(to fix the meter I had to use decoded ID from the json above, and it works now)

kitor, 2024-11-24:
This was already written in this thread - Tauron provided ID is wrong, they
screwed up endianness in their system. This is not a wmbusmeters issue.

darmach, 2024-12-13:
Thanks to the responses above I managed to configure the new Amiplus meter NES 73331.

I get the following sensors in Homeassistant, total energy consumption and production are `Unknown` though:
[image]

I suspect I know what is the cause of that (please confirm if I'm right):
1. In mqtt `homeassistant/sensor/wmbusmeters/MYID_total_energy_consumption_kwh/config` I see:
```
{
  "device": {
    "identifiers": [
      "wmbusmeters_MYID"
    ],
    "manufacturer": "APATOR",
    "model": "Electricity meter Apator EMU family (driver: amiplus)",
    "name": "Amiplus Meter",
    "hw_version": "MYID" #(lol)
  },
  "name": "total energy consumption",
  "unique_id": "wmbusmeters_MYID_total_energy_consumption_kwh",
  "state_topic": "wmbusmeters/Amiplus Meter",
  "value_template": "{{ value_json.total_energy_consumption_kwh }}",
  "json_attributes_topic": "wmbusmeters/Amiplus Meter",
  "icon": "mdi:transmission-tower-import",
  "unit_of_measurement": "kWh",
  "state_class": "total_increasing",
  "device_class": "energy",
  "enabled_by_default": true
}
```
This points to the source of the data to be "wmbusmeters/Amiplus Meter" topic, "{{ value_json.total_energy_consumption_kwh }}".
I don't seem to find `total_energy_consumption_kwh` in that topic messages, I only have consumption split by tariffs, out of which only one is not empty, as I'm on G11 (no tariffs) plan:

```
{
  "_": "telegram",
  "media": "electricity",
  "meter": "amiplus",
  "name": "Amiplus Meter",
  "id": "MYID",
  "current_power_consumption_kw": 0.744,
  "current_power_production_kw": 0,
  "total_energy_consumption_tariff_1_kwh": 1435.053,
  "total_energy_consumption_tariff_2_kwh": 0,
  "total_energy_consumption_tariff_3_kwh": 0,
  "total_energy_production_tariff_1_kwh": 4.303,
  "total_energy_production_tariff_2_kwh": 0,
  "total_energy_production_tariff_3_kwh": 0,
  "voltage_at_phase_1_v": 224,
  "voltage_at_phase_2_v": 231,
  "voltage_at_phase_3_v": 238,
  "device_date_time": "2024-12-13 17:49:27",
  "timestamp": "2024-12-13T17:49:25Z",
  "device": "rtlwmbus[00000001]",
  "rssi_dbm": 9
}
```

I'd need to change the template for total to agregate those 3 tariffs.
Where is it defined, where should I look and open a PR as I'd like to contribute?

Can you guys advise @kitor @dslynko @Leo-PL ?

darmach, 2024-12-13:
@wrobelda I've seen your issue [here ](https://github.com/wmbusmeters/wmbusmeters/issues/1230) it's interesting, why my telegram is different than yours? In your case the abovementioned templating for energy consumption would work.

I hope I'm not an edge case... :D

Mice30, 2024-12-16:
Hi, Its been a while:)

My temporary solution is to add missing entities in amiplus.json. It is not a full solution but it delivers missing datas.
My meter is Tauron OTUS 3.
Here is a full amiplus.json:
`{
"current_power_consumption_kw":{
     "component":"sensor",
     "discovery_payload":{
        "device":{
           "identifiers":[
              "wmbusmeters_{id}"
           ],
           "manufacturer":"APATOR",
           "model":"Electricity meter Apator EMU family (driver: {driver})",
           "name":"{name}",
           "hw_version":"{id}"
        },
        "name":"current power consumption",
        "unique_id":"wmbusmeters_{id}_{attribute}",
        "state_topic":"wmbusmeters/{name}",
        "value_template":"{{ value_json.{attribute} }}",
        "json_attributes_topic":"wmbusmeters/{name}",
        "icon":"mdi:counter",
        "unit_of_measurement":"kW",
        "state_class":"measurement",
        "device_class":"power",
        "enabled_by_default":true
     }
  },
  "current_power_production_kw":{
     "component":"sensor",
     "discovery_payload":{
        "device":{
           "identifiers":[
              "wmbusmeters_{id}"
           ],
           "manufacturer":"APATOR",
           "model":"Electricity meter Apator EMU family (driver: {driver})",
           "name":"{name}",
           "hw_version":"{id}"
        },
        "name":"current power production",
        "unique_id":"wmbusmeters_{id}_{attribute}",
        "state_topic":"wmbusmeters/{name}",
        "value_template":"{{ value_json.{attribute} }}",
        "json_attributes_topic":"wmbusmeters/{name}",
        "icon":"mdi:counter",
        "unit_of_measurement":"kW",
        "state_class":"measurement",
        "device_class":"power",
        "enabled_by_default":true
     }
  },
  "total_energy_consumption_tariff_1_kwh":{
     "component":"sensor",
     "discovery_payload":{
        "device":{
           "identifiers":[
              "wmbusmeters_{id}"
           ],
           "manufacturer":"APATOR",
           "model":"Electricity meter Apator EMU family (driver: {driver})",
           "name":"{name}",
           "hw_version":"{id}"
        },
        "name":"total energy consumption tariff 1 kwh",
        "unique_id":"wmbusmeters_{id}_{attribute}",
        "state_topic":"wmbusmeters/{name}",
        "value_template":"{{ value_json.{attribute} }}",
[... 241 lines cut ...]
  "rssi_dbm":{
     "component":"sensor",
     "discovery_payload":{
        "device":{
           "identifiers":[
              "wmbusmeters_{id}"
           ],
           "manufacturer":"APATOR",
           "model":"Water meter AT-WMBUS-16-2 (driver: {driver})",
           "name":"{name}",
           "hw_version":"{id}"
        },
        "entity_category":"diagnostic",
        "name":"rssi",
        "unique_id":"wmbusmeters_{id}_{attribute}",
        "state_topic":"wmbusmeters/{name}",
        "value_template":"{{ value_json.{attribute} }}",
        "icon":"mdi:signal",
        "unit_of_measurement":"dBm",
        "device_class":"signal_strength",
        "state_class":"measurement",
        "enabled_by_default":true
     }
  }
}`

darmach, 2025-03-07:
I opened two PRs in HA addon repository, could some of you guys take a look?

https://github.com/wmbusmeters/wmbusmeters-ha-addon/pull/54
https://github.com/wmbusmeters/wmbusmeters-ha-addon/pull/55

closed 2025-06-06 by github-actions[bot]