wmbusmeters/wmbusmeters #1303: WIZE protocol support in 169Mhz ?

MacAre, 2024-06-15:
### Type of request

Question

### OS version

Linux (Home Assistant)

### wmbusmeters version

v1.16.1-40-1

### Your message goes here

Hello,
Are there plans to add decoding of WIZE frames (N mode) in 169Mhz, for DIEHL meters for example ?

The N mode topic is regularly discussed here:
- https://github.com/wmbusmeters/wmbusmeters/issues/447#issuecomment-1455714195
- https://github.com/wmbusmeters/wmbusmeters/issues/677
- https://github.com/wmbusmeters/wmbusmeters/issues/10
- https://github.com/xaelsouth/rtl-wmbus/issues/26
- https://forum.hacf.fr/t/compteur-deau-telereleve-en-169mhz/30050/25

WIZE video here ([Webinar] Protocole et Produits Wize):
- Mode N2: https://youtu.be/3-Z9w-znJqY?si=35gWDecv_AvZMU4z&t=1700
- Frequency: https://youtu.be/3-Z9w-znJqY?si=c_AyjbRmjlD73N7Q&t=2723

How is it possible to help ?

AkdM, 2024-07-08:
I am not expert, but from my experience there's no way to decrypt device message then?

at 51:43: "We don't want anyone to decrypt the content of the message"

[image]

MacAre, 2024-07-19:
Very good observation AkdM
In fact it is said that WIZE provides security-related fields to authenticate frames and perform encryption.
The objective is to be sure of the transmitter, and not to be able to decipher what is circulating on the radio.
This risks putting an end to our research.

Mattarn, 2024-10-15:
Hi. 
I’m Interested too for Wize support but I understand that it can’t be done ?

closed 2025-01-15 by github-actions[bot]