wmbusmeters/wmbusmeters #1306: Hydrocal M4 cooling

GOinfo-Ltd, 2024-06-19:
BMeters Hydrocal M4
Using driver hydrocalm3

The data for heating is decoded properly, but cooling isn't. Same for the current volujme and power.

The device is showing 0.139 kWh of cooling energy and 0.043 m^3 of cooling volume.

Device is setup to send the "combined" data (volume and energy for heating and cooling)
https://wmbusmeters.org/analyze/3a44B409238709051a0d8c00187ac0000000046d26B113360c03060100000c13470000008c1003390100008c1013430000000f6401000000000000 

This is in "instantaneous" mode (sends the heating kWh+m3, volume, power and temperatures)
https://wmbusmeters.org/analyze/3E44B409238709051A0D8C00187ACC000000046D29B113360C03060100000C13470000000B3B0000000B280000000A5998250A5D74250F6402000000000000

and another one with some volume flow > 0
https://wmbusmeters.org/analyze/3E44B409238709051A0D8C00187AE6000000046D2FB113360C03060100000C13470000000B3B6939000B280000000A5966260A5D97250F6402000000000000
(Value could be quite high, I was blowing in the device to get it spinning)

Again, this is a hydrocal M4 and not a M3. Mfg date : 2023.

Thank you!

kzajac83, 2024-09-01:
I was wondering what is difference between below frame types in Hydrocal M4 and now understood, both no coding frames

1. Standard frame, 45 bits, only heating info
https://wmbusmeters.org/analyze/2c44B409418012051a0d8c20837a2f120020046d24B401390c08000000000c13000000000f6400100000000000

3. Extend frane, 59 bits, cool+heating
https://wmbusmeters.org/analyze/3A44B409418012051A0D8C20CD7A79120020046D32A002390C08000000000C13000000008C1008000000008C1013000000000F6401100000000000

This data come from new not used heat meter hence zeros energy and flows.  

Just need upgrade drivers for extend frame in your cause and also you need switch Hydrocal M4 from standard frame to extend frame.

Webbeh, 2024-09-02:
I am 100% confident the cooling frames did work at some point, as I saw that data from wmbusmeter software on a linux machine a few months ago.

However, I am unsure whether bmeters updated the firmware of the devices in the meantime and changed the data format in a new firmware...

kzajac83, 2024-09-02:
No need change FW, just install https://play.google.com/store/apps/details?id=it.gread.bmeters_appnfc and charge via NFC frame type in heat meter.   However next todo is update wmbusmeters driver, probably I cannot do that

sunflowerenergias, 2024-10-01:
I have tried to reconfigure the meter with the BMetering NFC application and with none of the configurations it returns the cold value or the driver is not working correctly.

What would be the correct configuration in the BMetering NFC application?

Does anyone have more information on this?

kzajac83, 2024-10-01:
At this moment, the driver does not recognise cooling-related fields.

sunflowerenergias, 2024-10-02:
I have created a new pull request where this problem is fixed with a new driver

balczezzz, 2024-10-18:
Hi, I'll join the discussion. I've been successfully running few Kamstrup HydrocalM3 and just bought M4 meters. Whereas with the M3 the frames for SUPPLY and RETURN temperature are properly decoded, the M4 does not. The T1/T2 values are null. Is this the same for you as well?

sunflowerenergias, 2024-10-21:
> Hola, me uno a la discusión. He estado utilizando con éxito algunos medidores HydrocalM3 de Kamstrup y acabo de comprar el M4. Mientras que con el M3 los cuadros para la temperatura de SUMINISTRO y RETORNO se decodifican correctamente, el M4 no. Los valores T1/T2 son nulos. ¿Te ocurre lo mismo a ti también?

It may be a problem with the counter configuration. Based on the tests I have been able to perform, the counter has to be configured with the NFC application in "Instantaneous" mode to output the temperature values.

This is a model M4 counter in Instantaneous mode:

https://wmbusmeters.org/analyze/3E44B409381317051A0D8C00497A7C000000046D2BAA153A0C03000000000C13000000000B3B0000000B280000000A5930230A5D08250F6402000000000000

balczezzz, 2024-10-27:
@sunflowerenergias Thanks! This helped and I am able to get the T1/T2 data now.

Webbeh, 2025-01-10:
Stalebot is the worst bot.

kzajac83, 2025-01-14:
@Webbeh  what is the problem here? Asking because also have M4 and seems now all works fine, please check https://github.com/wmbusmeters/wmbusmeters/issues/1456 

After compile current version I checked your 3 telegrams, all expect manufacturer data, work fine. 
Unfortunately on wmbusmeters.org page last patches not yet updated  so there still something missing but please ignore it and compile last version and use it on your system.

closed 2025-04-15 by github-actions[bot]