wmbusmeters/wmbusmeters #1307: Driver for AXIOMA

leifnel, 2024-06-20:
### Type of request

New device support

### Meter description

 (AXI) UAB Axis Industries, Lithuania (0x709) type: Water meter (0x07) ver: 0x01
 
 
 The q400 driver gives good results

### Logged telegrams from meter in decrypted format

```shell
telegram=|614409072497280601077AB7000000_046D04011436042098C77F02041345E5000004933B45E5000004933C00000000023B000002596B07446D000001364413E0CE000044933BE0CE000044933C0000000034FD17000000000424EFC77F0201FD745F|+1963
telegram=|614409072497280601077ABB000000_046D18011436042048CC7F02041345E5000004933B45E5000004933C00000000023B00000259B307446D000001364413E0CE000044933BE0CE000044933C0000000034FD17000000000424C3CC7F0201FD745F|+3200
telegram=|614409072497280601077ABD000000_046D230114360420A0CE7F02041345E5000004933B45E5000004933C00000000023B00000259D007446D000001364413E0CE000044933BE0CE000044933C0000000034FD170000000004242DCF7F0201FD745F|+3817
```

closed 2024-09-18 by github-actions[bot]

leifnel, 2025-07-22:
Please reconsider

geomax2024, 2025-07-22:
It does decrypt... what is the problem?
https://wmbusmeters.org/

614409072497280601077AB7000000_046D04011436042098C77F02041345E5000004933B45E5000004933C00000000023B000002596B07446D000001364413E0CE000044933BE0CE000044933C0000000034FD17000000000424EFC77F0201FD745F

"_":"telegram",
    "media":"water",
    "meter":"q400",
    "name":"",
    "id":"06289724",
    "backward_at_set_date_m3":0,
    "battery_pct":95,
    "consumption_at_set_date_m3":52.96,
    "flow_temperature_c":18.99,
    "forward_at_set_date_m3":52.96,
    "on_time_h":11646.833333,
    "total_m3":58.693,
    "total_backward_m3":0,
    "total_forward_m3":58.693,
    "volume_flow_m3h":0,
    "meter_datetime":"2024-06-20 01:04",
    "set_datetime":"2024-06-01 00:00",
    "status":"OK",
    "timestamp":"2025-07-22T22:56:57Z"