wmbusmeters/wmbusmeters #1313: No topic is published in Mqtt broker

ostracizado, 2024-07-01:
### Type of request

Question

### OS version

Raspberry OS:latest (Pi4)

### wmbusmeters version

1.15.0

### Your message goes here

With new neighbors still waiting for their meter, after months, and asking me to use my water once in a while - it's been quite a chore to have to leave the house and check twice my meter, every time they ask if they can use it.
  
Anyway, this project (with HA) would be quite a treat... But I've been all day at it, and still no luck trying to publish the json data to Mqtt; even do I can see the data.

/etc/wmbusmeters.conf:

```
loglevel=debug
device=rtlwmbus
#logfile=/var/log/wmbusmeters/wmbusmeters.log
logtelegrams=false
format=json
meterfiles=/var/lib/wmbusmeters/meter_readings
meterfilesaction=overwrite
shell=mosquitto_pub -h 192.168.2.100 -u mqtt-user -P Confiancaps5. -t wmbusmeters/ -m "$METER_JSON"
```

/etc/wmbusmeters.d/MyTapWater:

```
name=MyTapWater
id=210fa05d
driver=izar
```

/var/lib/wmbusmeters/meter_readings/MyTapWater:
```
{"media":"water","meter":"izar","name":"MyTapWater","id":"210fa05d","last_month_total_m3":320.597,"remaining_battery_life_y":8.5,"total_m3":324.357,"transmit_period_s":8,"current_alarms":">
```

If I use:
```
wmbusmeters --useconfig=/
```
log:

```
(config) "# Remember to change auto here to the device you are going to use in production." ""
(config) "device" "rtlwmbus"
(config) "#device=/dev/ttyUSB0:t1" ""
(config) "#logfile=/var/log/wmbusmeters/wmbusmeters.log" ""
(config) "# Set to true to capture all received telegrams in log file." ""
(config) "logtelegrams" "false"
(config) "format" "json"
(config) "# Enable the meterfiles to write telegrams to disk." ""
(config) "meterfiles" "/var/lib/wmbusmeters/meter_readings"
(config) "meterfilesaction" "overwrite"
(config) "# Enable execution of a shell command for each received telegram. For example: curl or mqtt" ""
(config) "shell" "mosquitto_pub -h 192.168.2.100 -u user -P password -t wmbusmeters/ -m "$METER_JSON""
(config) "#shell=/usr/bin/mosquitto_pub -h 192.168.2.100 -p 1883 -t wmbusmeters/$METER_ID -m "$METER_JSON"" ""
(config) "#shell=/usr/bin/mosquitto_pub -h 192.168.2.100 -p 1883 -t wmbusmeters/ -m "$METER_JSON"" ""
(config) "# The alarmshell is executed when a problem with the receiving radio hardware is detected." ""
(config) "#alarmshell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters_alarm -m " "" ""
(config) "# The alarmtimeout and expected activity is also used to detect failing receiving radio hardware." ""
(config) "#alarmtimeout=1h" ""
(config) "#alarmexpectedactivity=mon-sun(00-23)" ""
[... 237 lines cut ...]
    Process: 6915 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 6916 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 6917 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
   Main PID: 6918 (wmbusmetersd)
      Tasks: 8 (limit: 3910)
        CPU: 10.490s
     CGroup: /system.slice/wmbusmeters.service
             ├─6918 /usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid
             ├─6936 /bin/sh -c "ERRFILE=\$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=\$ERRFILE ;  date -Iseconds > \$ERRFILE ; tail -f \$ERRFILE & /usr/bin/rtl_sdr  -d 0 -f>
             ├─6939 tail -f /tmp/wmbusmeters_rtlsdr.g7vGXwkB7XeG
             ├─6940 /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 -
             └─6941 /usr/bin/rtl_wmbus -s -f

Jul 01 04:17:34 rtl wmbusmetersd[6918]: (wmbus) parseELL @10 16
Jul 01 04:17:34 rtl wmbusmetersd[6918]: (wmbus) parseNWL @10 16
Jul 01 04:17:34 rtl wmbusmetersd[6918]: (wmbus) parseAFL @10 16
Jul 01 04:17:34 rtl wmbusmetersd[6918]: (wmbus) parseTPL @10 16
Jul 01 04:17:34 rtl wmbusmetersd[6918]: (meter) MyTapWater: for me? 211549a7.M=SAP.V=00.T=07 in 210fa05d
Jul 01 04:17:34 rtl wmbusmetersd[6918]: (meter) MyTapWater: not for me: no match
Jul 01 04:17:34 rtl wmbusmetersd[6918]: (wmbus) telegram from TODO ignored by all configured meters!
Jul 01 04:17:34 rtl wmbusmetersd[6918]: (rtlwmbus) checkRTLWMBusFrame "T1;1;1;2024-07-01 03:17:34.000;27;28;211549A7;0x1944304ca74915219801a1215200135d8b9f0307c045adb66811<0A>"
Jul 01 04:17:34 rtl wmbusmetersd[6918]: (rtlwmbus) received full frame
Jul 01 04:17:34 rtl wmbusmetersd[6918]: (wmbus) skipping already handled telegram leng=26.
```
Any ideas?

ostracizado, 2024-07-02:
I've tested the add-on and it works without a problem.
Unfortunately it doesn't pick my water meter, from where my HA stands - just a couple of ones from some neighbours. 

Seems the problem with the rasp4 must be on my part; probably the config?

ostracizado, 2024-07-03:
@weetmuts, sorry to bother, but could you take a look at it? Maybe a fresh pair of eyes will spot where's the problem...

I've been trying for days now, with no luck.
Like I said before, I can see the data, but I can't seem to find a way to publish it in my MQTT broker.

For instance, if I 'virtually' unplug & plug the usb RTS with uhubctl (I was getting error -6, after a reboot) :

```
sudo uhubctl -l 1-1 -p 3 -a off
```
```
sudo uhubctl -l 1-1 -p 3 -a on
```
Then start wmbusmeters:

```
admin@rtl:~ $ sudo systemctl start wmbusmeters
admin@rtl:~ $ sudo systemctl status wmbusmeters
● wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/lib/systemd/system/wmbusmeters.service; disabled; preset: enabled)
     Active: active (running) since Wed 2024-07-03 17:52:36 WEST; 45s ago
       Docs: https://github.com/wmbusmeters/wmbusmeters
             man:wmbusmeters(1)
    Process: 2426 ExecStartPre=/bin/mkdir -p /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 2427 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 2428 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 2429 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 2430 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 2431 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 2432 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
   Main PID: 2433 (wmbusmetersd)
      Tasks: 8 (limit: 3910)
        CPU: 4.170s
     CGroup: /system.slice/wmbusmeters.service
             ├─2433 /usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid
             ├─2822 /bin/sh -c "ERRFILE=\$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=\$ERRFILE ;  date -Iseconds > \$ERRFILE ; tail -f \$ERRFILE & /usr/bin/rtl_sdr  -d 0 -f>
             ├─2825 tail -f /tmp/wmbusmeters_rtlsdr.i2luEJ2OdIGL
             ├─2826 /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 -
             └─2827 /usr/bin/rtl_wmbus -s -f

Jul 03 17:53:21 rtl wmbusmetersd[2433]: (wmbus) parseELL @10 16
Jul 03 17:53:21 rtl wmbusmetersd[2433]: (wmbus) parseNWL @10 16
Jul 03 17:53:21 rtl wmbusmetersd[2433]: (wmbus) parseAFL @10 16
Jul 03 17:53:21 rtl wmbusmetersd[2433]: (wmbus) parseTPL @10 16
Jul 03 17:53:21 rtl wmbusmetersd[2433]: (meter) MyTapWater: for me? 211549a7.M=SAP.V=00.T=07 in 210fa05d
Jul 03 17:53:21 rtl wmbusmetersd[2433]: (meter) MyTapWater: not for me: no match
Jul 03 17:53:21 rtl wmbusmetersd[2433]: (wmbus) telegram from TODO ignored by all configured meters!
Jul 03 17:53:21 rtl wmbusmetersd[2433]: (rtlwmbus) checkRTLWMBusFrame "T1;1;1;2024-07-03 16:53:21.000;67;67;211549A7;0x1944304ca74915219801a1415200135d1262b6fd04ad796c3e1e<0A>"
Jul 03 17:53:21 rtl wmbusmetersd[2433]: (rtlwmbus) received full frame
Jul 03 17:53:21 rtl wmbusmetersd[2433]: (wmbus) skipping already handled telegram leng=26.
lines 1-32/32 (END)
```

If I check the status after a while:
```
admin@rtl:~ $ sudo systemctl status wmbusmeters
● wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/lib/systemd/system/wmbusmeters.service; disabled; preset: enabled)
     Active: active (running) since Wed 2024-07-03 17:52:36 WEST; 8min ago
       Docs: https://github.com/wmbusmeters/wmbusmeters
             man:wmbusmeters(1)
    Process: 2426 ExecStartPre=/bin/mkdir -p /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 2427 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 2428 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 2429 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 2430 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 2431 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 2432 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
   Main PID: 2433 (wmbusmetersd)
      Tasks: 3 (limit: 3910)
        CPU: 2min 54.009s
     CGroup: /system.slice/wmbusmeters.service
             └─2433 /usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid

Jul 03 17:59:00 rtl wmbusmetersd[2433]: (shell) exec (capture output) "/bin/pidof"
Jul 03 17:59:00 rtl wmbusmetersd[2433]: (shell) arg "wmbusmeters"
Jul 03 17:59:00 rtl wmbusmetersd[2433]: (serial) EVENT thread interrupted
Jul 03 17:59:00 rtl wmbusmetersd[2433]: (shell) output: >>><<<
Jul 03 17:59:00 rtl wmbusmetersd[2433]: (shell) return code 1
Jul 03 17:59:00 rtl wmbusmetersd[2433]: (shell) 3347 exited
Jul 03 17:59:00 rtl wmbusmetersd[2433]: (serial) EVENT thread interrupted
Jul 03 17:59:00 rtl wmbusmetersd[2433]: [ALARM SpecifiedDeviceNotFound] the device rtlwmbus is not working
Jul 03 18:00:02 rtl wmbusmetersd[2433]: [ALARM SpecifiedDeviceNotFound] the device rtlwmbus is not working
Jul 03 18:01:03 rtl wmbusmetersd[2433]: [ALARM SpecifiedDeviceNotFound] the device rtlwmbus is not working
```

weetmuts, 2024-07-03:
I am sorry but I am not an expert on the HA integration. My specialty is the decoding of the telegrams themselves and it seems you do see your meters readings in the logs.

Perhaps you can ask in the HA forums. Perhaps the topic has to be right etc.

ostracizado, 2024-07-03:
Hmmm... maybe it's a question of a bad installation.
It's seems the problem is that "/usr/bin/mosquitto_pub" can not be found.

closed 2024-07-04 by ostracizado