wmbusmeters/wmbusmeters #1317: im871a not found

omaar25, 2024-07-04:
### Type of request

Enhancement

### OS version

11

### wmbusmeters version

1.5

### Your message goes here

Hey,

omar@DESKTOP-5RF2JTS:/mnt/c/Windows/system32$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

omar@DESKTOP-5RF2JTS:/mnt/c/Windows/system32$ ls -l /dev/ttyUSB0
crw-rw-rw- 1 root root 188, 0 Jul  4 15:31 /dev/ttyUSB0

omar@DESKTOP-5RF2JTS:/mnt/c/Windows/system32$ wmbusmeters /dev/ttyUSB0:im871a:t1
Expected im871a on /dev/ttyUSB0 but did not find it! Ignoring tty!
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!

How can i fix this issue ?

weetmuts, 2024-07-04:
Run `wmbusmeters --debug auto:t1` post log here.

omaar25, 2024-07-04:
omar@DESKTOP-5RF2JTS:/mnt/c/Windows/system32$ wmbusmeters --debug auto:t1
(drivers) added builtin driver aquastream
(drivers) added builtin driver elster
(drivers) added builtin driver eltako
(drivers) added builtin driver ime
(drivers) added builtin driver iperl
(drivers) added builtin driver istaheat
(drivers) added builtin driver itronheat
(drivers) added builtin driver kampress
(drivers) added builtin driver sensostar
(drivers) added builtin driver supercal
(drivers) added builtin driver werhlemodwm
(wmbusmeters) version: 1.16.1-101-g901674d
(config) number of meters: 0
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) check if /dev/ttyUSB0 can be accessed
(serial) you do not have the correct permissions to open the tty /dev/ttyUSB0 and you do not share the same access group.
(main) ignoring /dev/ttyUSB0, it does not respond as any of the supported wmbus devices.
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(shell) return code 1
(shell) 56 exited
(serial) EVENT thread interrupted
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>52
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 57 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted

Thank you for your reply and assistance!

weetmuts, 2024-07-04:
It says: "(serial) you do not have the correct permissions to open the tty /dev/ttyUSB0 and you do not share the same access group." Add your user to the dialout group and it should get tty access.

sudo usermod -a -G dialout <username>

Replace <username> with your user name.

Log out and login again.

weetmuts, 2024-07-04:
Are you running Linux under Windows? WSL

omaar25, 2024-07-04:
Yes

weetmuts, 2024-07-06:
I have never tried WSL for wmbusmeters, so I cannot help you with this.

Some searching reveals this:
https://askubuntu.com/questions/1461302/i-need-help-connecting-serial-ports-to-ubuntu-in-wsl

omaar25, 2024-07-09:
Hi Fredrik

I'm running wmbusmeters on ubuntu desktop but still the same issue:

***@***.***:~/GIT/wmbusmeters$ sudo wmbusmeters --debug
/dev/ttyUSB0:im871a:115200:t1
(drivers) added builtin driver aquastream
(drivers) added builtin driver elster
(drivers) added builtin driver eltako
(drivers) added builtin driver ime
(drivers) added builtin driver iperl
(drivers) added builtin driver istaheat
(drivers) added builtin driver itronheat
(drivers) added builtin driver kampress
(drivers) added builtin driver sensostar
(drivers) added builtin driver supercal
(drivers) added builtin driver werhlemodwm
(wmbusmeters) version: 1.16.1-101-g901674d
(config) using device: /dev/ttyUSB0:im871a:115200:t1
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" im871a
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B115200) input() output()
control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output()
control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect im871a)
Expected im871a on /dev/ttyUSB0 but did not find it! Ignoring tty!
(serial) EVENT thread interrupted
(verbose) ignoring device t1
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(shell) return code 1
(shell) 5757 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(serial) EVENT thread interrupted
(shell) output: >>>5754
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 5758 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
^C(serial) stopping manager
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted

AUG-7, 2024-08-10:
i am also still at this point. @omaar25 did you find a solution?

omaar25, 2024-08-10:
No I still have the same issue

AUG-7, 2024-08-10:
@weetmuts Hi Fredrik, sorry to mention you, but we both can't see our im871a dongle. Are there more steps to try? thanks in advance.

closed 2024-11-10 by github-actions[bot]