wmbusmeters/wmbusmeters #1320: RTL dongle very hot

GDXFR, 2024-07-08:
### Type of request

Question

### HAOS version

2024.6.3

### wmbusmeters version

1.16.1

### Your message goes here

Hi, my RTL dongle (Nooelec NESDR) is very hot, is there any way to reduce usage and power it only a few minutes every hours? Thanks

Tpaydt, 2024-07-17:
Which protocol are you using?  c1, s1, t1?

my RTL runs only with c1 but is still hot

weetmuts, 2024-09-16:
I bought one because it looked so cute, but it heats up my laptop to a level where I stopped using it.
Even though the Noelec NSDR might withstand high heat, I am not sure my laptop does....

weetmuts, 2024-09-16:
https://www.reddit.com/r/RTLSDR/comments/ng1zzt/nooelec_smartee_too_hot_with_solutionhopefully/

weetmuts, 2024-09-16:
https://support.nooelec.com/hc/en-us/articles/360005889913-NESDR-Nano-Series

closed 2024-12-16 by github-actions[bot]