wmbusmeters/wmbusmeters #1322: CPU Usage to high with RTL-SDR and Addon

Zuendapp, 2024-07-13:
### Type of request

Question

### HAOS version

2024.7.2

### wmbusmeters version

1.16.1

### Your message goes here

Hello :),

since three days I'm able to get a readout from my Diehl Hydrus Watermeter (Type 171) with a RTL-SDR v3 Dongle.
My Setup is Home Assistant Supervised with the WMbusmeters Add-on.
My CPU Usage is beetwen 18-25% all the time.

I'm using following config:

> device=auto:t1
donotprobe=/dev/ttyAMA0
format=json
logfile=/dev/stdout
loglevel=normal
logtelegrams=false
shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"

Is it better to use a nano CUL 868 Stick instead of the RTL-SDR, or dou you have any idea to get down the CPU usage?

Thank you very much!

BR

Tpaydt, 2024-07-14:
try this:

Here is an example command line that reduces the rtl_wmbus CPU usage if you only need T1/C1 telegrams. It disable S1 decoding (-p s) and trades lower cpu usage for reception performance (-a). You should always add the -f option to enable detection if rtl_sdr has stalled:

device=rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a -f)

Zuendapp, 2024-07-14:
Hi,

thank you, but I tried and the CPU usage is still high (see Screenshot):

[image]

BR
Zuendapp

Zuendapp, 2024-07-28:
Closed - 

> try this:
>
> Here is an example command line that reduces the rtl_wmbus CPU usage if you only need T1/C1 telegrams. It disable S1 decoding (-p s) and trades lower cpu usage for reception performance [...]

closed 2024-07-28 by Zuendapp