wmbusmeters/wmbusmeters #1323: pa-im871a

omaar25, 2024-07-16:
### Type of request

Bug

### OS version

24.04

### wmbusmeters version

1.16.1-101-g901674d

### Your message goes here

Hi, 

How can i resolve this issue?

omar@omar-HP-Laptop-15-dw2xxx:~/GIT/wmbusmeters$ wmbusmeters --debug /dev/ttyUSB0:im871a:t1
(drivers) added builtin driver aquastream
(drivers) added builtin driver elster
(drivers) added builtin driver eltako
(drivers) added builtin driver ime
(drivers) added builtin driver iperl
(drivers) added builtin driver istaheat
(drivers) added builtin driver itronheat
(drivers) added builtin driver kampress
(drivers) added builtin driver sensostar
(drivers) added builtin driver supercal
(drivers) added builtin driver werhlemodwm
(wmbusmeters) version: 1.16.1-101-g901674d
(config) using device: /dev/ttyUSB0:im871a:t1 
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyUSB0" im871a
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect im871a)
Expected im871a on /dev/ttyUSB0 but did not find it! Ignoring tty!
(serial) EVENT thread interrupted
(verbose) ignoring device t1
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 3183 exited
Notice! Wmbusmeters daemon (pid 1199) is running and it might hog any wmbus devices.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 3184 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
^C(serial) stopping manager
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!

Tpaydt, 2024-07-17:
try to add the serial from the im871a

device=/dev/ttyUSB0:im871a[00102759]:t1

and maybe you need to add one meter to the config

(config) number of meters: 0

omaar25, 2024-07-17:
Is there a way to configure wmbus meters to capture packets from multiple
meters without specifying each one individually?

> Message ID: ***@***.***>
>

Tpaydt, 2024-07-17:
i'm not so deep with wmbusmeters but wenn you read the description the is a option for this:

You can use several ids using **id=1111111,2222222,3333333** or you can listen to all meters of a certain type **id=*** or you can suffix with star **id=8765*** to match all meters with a given prefix. If you supply at least one positive match rule, then you can add filter out rules as well. For example **id=*,!2222*** which will match all meter ids, except those that begin with 2222.

You can also specify the exact manufacturer, version and type: **id=11111111.M=KAM.V=1b.T=16** or a subset: **id=11111111.T=16** or all telegrams from 22222222 except those with version 77: **id=22222222,!22222222.V=77** You can also use the fully specified secondary address that is printed by libmbus after doing a bus scan, ie 100002842941011B which is equivalent to 10000284.M=PII.V=01.T=1B

When matching all meters from the command line you can use **ANYID** instead of * to avoid shell quotes.

AUG-7, 2024-08-08:
I am also trying to get this usb dongle to setup, but when i enter this command: wmbusmeters --debug /dev/ttyUSB0:im871a:t1

[ALARM SpecifiedDeviceNotFound] the device /dev/ttyUSB0:im871a:t1 is not working
(lookup) with file/hex "/dev/ttyUSB0" im871a
(serial) check if /dev/ttyUSB0 can be accessed
(serial) tty /dev/ttyUSB0 can be accessed
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyUSB0 is already in use and locked.
(im871a) could not open tty /dev/ttyUSB0 for detection
Expected im871a on /dev/ttyUSB0 but did not find it! Ignoring tty!
(verbose) ignoring device t1

I am trying for 2 days, but can't find out what i do wrong, hope someone can help me out :)

closed 2024-11-08 by github-actions[bot]