wmbusmeters/wmbusmeters #1327: Sensus Water Meter PulseRF failed decryption

Tobiaskquest, 2024-07-20:
### Type of request

Question

### OS version

Ubuntu 24.04 LTS noble

### wmbusmeters version

1.15.0

### Your message goes here

Hello everyone,

I'm new to using wmbusmeters.
I use Ubuntu 24.04 LTS and have managed to get wmbusmeters to work. I use the Amber Wireless 8465-M 868 stick and am trying to read data from a Sensus water meter. Model PulseRF which can also do WMBUS 868 T1.
If I enter the following command : **sudo wmbusmeters --analyze=<key> auto:t1**
(I have of course entered the correct/real key)

Then I get this displayed:

**Auto driver: not found!**
Best driver : unknown 00/00 Using driver : unknown 00/00 000 : 27 length (39 bytes) 001 : 44 dll-c (from meter SND_NR) 002 : ae4c dll-mfct ​​(SEN) 004 : 18651700 dll-id (00176518) 008 : 80 dll-version 009 : 37 d ll-type (Radio converter (meter side)) 010 : 72 tpl-ci-field (EN 13757-3 Application Layer (long tplh)) 011 : 77657410 tpl-id (10746577) 015 : ae4c tpl-mfct ​​(SEN) 017 : 0b tpl-version 018 : 07 tpl-type (Water meter) 019 : 20 tpl-acc-field 020 : 00 tpl-sts-field (OK) 021 : 1005 tpl-cfg 0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0 ) 023 : **738f decrypt check bytes (ERROR should be 2f2f) 025 CE: F4A3C4E736587462D5CB5A73891956 failed decryption. Wrong key?**

{

"media":"water",

"meter":"unknown",

"name":"",

"id":"10746577",

"timestamp":"1970-01-01T00:00:00Z",

"device":"amb8465[00000000]",

"rssi_dbm":-31**

My questions:

1. Is there a driver for Sensus modules somewhere or similar?

2. Can this problem be solved? - 738f decrypt check bytes (ERROR should be 2f2f)

3. Which commands can I use to only wait for this meter and its telegram? Unfortunately it only sends every hour. But I could also have it changed to 15 minutes.

4. Which command gives me more information about this Meter/Module?

I would like to thank you in advance for your help.

weetmuts, 2024-07-22:
The 2f2f warning just means you got the wrong key, or you are listening to the wrong meter.

Do: wmbusmeters --logtelegrams amd8465:t1

Then collect a log of telegrams. Double check that your meter id is correct.
Copy the encrypted telegram content from telegram=|..44........|
and paste it on the command line with different keys:

wmbusmeters ..44.......  MyMeter auto MyMeterId MyKey

Now you do not need to wait one hour for the next test.

Again, you have the wrong or you are trying to listen to the wrong meter. :-)

Tobiaskquest, 2024-07-24:
The manufacturer said the whole time that it was the right key. Only when we spoke to the right technician were they told it was the wrong key. Sensus had been using a standard key the whole time and will be issuing individual keys in the future. That's how it should be from a security perspective.

Thank you again for your help/answer.

This case can therefore be considered closed.

closed 2024-07-26 by Tobiaskquest