wmbusmeters/wmbusmeters #1331: Sensostar U wired MBus - no response

audiogeek123, 2024-07-28:
### Type of request

Bug

### HAOS version

11.5

### wmbusmeters version

1.16.1-101

### Your message goes here

Hello, I have a problem with the Engelmann Sensostar U with MBus Addon.
I am using this [MBus Master to USB converter](https://www.amazon.de/dp/B0CD7G8S53?ref=ppx_yo2ov_dt_b_product_details&th=1) on my Raspi where HA is running.

From the manual I get that the primary adress should be 0, so I am using "p0" in the settings. (This is only possible in the "edge" version of the addon, in the "stable" version p0 is not allowed.

```
[19:29:36] INFO: CONFIG_CONF ...
[19:29:36] INFO: {"device":"MAIN=/dev/ttyUSB1:mbus:2400","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[19:29:36] INFO: CONFIG_METERS ...
[19:29:36] INFO: [{"driver":"sensostar:MAIN:mbus","id":"p0","name":"WMZ","pollinterval":"60s"}]
[19:29:36] INFO: Syncing wmbusmeters configuration ...
[19:29:36] INFO: Registering meters ...
[19:29:37] INFO: Adding meter-0001 ...
[19:29:37] INFO: Generating MQTT configuration ... 
[19:29:37] ERROR: Got unexpected response from the API:
Service not enabled
[19:29:37] INFO: Broker  will be used.
[19:29:37] INFO: 
MQTT Discovery ...
[19:29:37] INFO:  Adding meter: WMZ ...
[19:29:37] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/p0_difference_c/config
Unable to connect (Lookup error.).
[19:29:41] INFO: Starting web configuration service.
[19:29:41] INFO: Running wmbusmeters ...
Started config MAIN=mbus on /dev/ttyUSB1
[19:29:42] INFO: Starting Nginx...
2024/07/28 19:29:42 [notice] 83#83: using the "epoll" event method
2024/07/28 19:29:42 [notice] 83#83: nginx/1.24.0
2024/07/28 19:29:42 [notice] 83#83: OS: Linux 6.1.74-haos
2024/07/28 19:29:42 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/07/28 19:29:42 [notice] 83#83: start worker processes
2024/07/28 19:29:42 [notice] 83#83: start worker process 236
(meter) WMZ p0 did not send a response!
``` 

The "did not send a response" will go on. 

Do you have any idea how to get it running?

audiogeek123, 2024-07-28:
with loglevel=debug:

```
[19:42:03] INFO: CONFIG_CONF ...
[19:42:03] INFO: {"device":"MAIN=/dev/ttyUSB1:mbus:2400","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"debug","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[19:42:03] INFO: CONFIG_METERS ...
[19:42:03] INFO: [{"driver":"sensostar:MAIN:mbus","id":"p0","name":"WMZ","pollinterval":"60s"}]
[19:42:03] INFO: Syncing wmbusmeters configuration ...
[19:42:03] INFO: Registering meters ...
[19:42:03] INFO: Adding meter-0001 ...
[19:42:03] INFO: Generating MQTT configuration ... 
[19:42:03] ERROR: Got unexpected response from the API:
Service not enabled
[19:42:04] INFO: Broker  will be used.
[19:42:04] INFO: 
MQTT Discovery ...
[19:42:04] INFO:  Adding meter: WMZ ...
[19:42:04] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/p0_difference_c/config
Unable to connect (Lookup error.).
[19:42:09] INFO: Starting web configuration service.
[19:42:09] INFO: Running wmbusmeters ...
(config) "logtelegrams" "false"
(wmbusmeters) version: 1.16.1-101-g901674d
(config) using device: MAIN=/dev/ttyUSB1:mbus:2400 
(config) number of meters: 1
(config) template WMZ p0 sensostar:MAIN:mbus
(driver) constructing driver sensostar from already loaded file builtin
(formula) parsing "tariff_counter"
(formula) tokens: FIELD(tariff_counter) 
(formula) handle dventry field tariff_counter into tariff counter
(formula) <DVENTRY tariff_counter>
(formula) parsing "subunit_counter"
(formula) tokens: FIELD(subunit_counter) 
(formula) handle dventry field subunit_counter into subunit counter
(formula) <DVENTRY subunit_counter>
(formula) parsing "tariff_counter"
(formula) tokens: FIELD(tariff_counter) 
(formula) handle dventry field tariff_counter into tariff counter
(formula) <DVENTRY tariff_counter>
(formula) parsing "subunit_counter"
(formula) tokens: FIELD(subunit_counter) 
(formula) handle dventry field subunit_counter into subunit counter
(formula) <DVENTRY subunit_counter>
(formula) parsing "storage_counter"
(formula) tokens: FIELD(storage_counter) 
(formula) handle dventry field storage_counter into storage counter
(formula) <DVENTRY storage_counter>
(meter) created WMZ sensostar p0 not-encrypted
(lookup) with file/hex "/dev/ttyUSB1" mbus
(lookup) driver: mbus
(main) opening MAIN=/dev/ttyUSB1:mbus:2400
Started config MAIN=mbus on /dev/ttyUSB1
(mbus) on /dev/ttyUSB1
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyUSB1 is already in use and locked.
(main) regular reset of mbus  on /dev/ttyUSB1 will happen every 82800 seconds
(wmbus) no alarm (expected activity) for mbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
[19:42:09] INFO: Starting Nginx...
2024/07/28 19:42:09 [notice] 83#83: using the "epoll" event method
2024/07/28 19:42:09 [notice] 83#83: nginx/1.24.0
2024/07/28 19:42:09 [notice] 83#83: OS: Linux 6.1.74-haos
2024/07/28 19:42:09 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/07/28 19:42:09 [notice] 83#83: start worker processes
2024/07/28 19:42:09 [notice] 83#83: start worker process 238
(meter) polling WMZ p0 (primary) with req ud2 fcb=1 on bus MAIN
(serial /dev/ttyUSB1) failed to send "107B007B16"
(meter) WMZ p0 did not send a response!
(meter) polling WMZ p0 (primary) with req ud2 fcb=1 on bus MAIN
(serial /dev/ttyUSB1) failed to send "107B007B16"
(meter) WMZ p0 did not send a response!
(meter) polling WMZ p0 (primary) with req ud2 fcb=1 on bus MAIN
(serial /dev/ttyUSB1) failed to send "107B007B16"
(meter) WMZ p0 did not send a response!
(meter) polling WMZ p0 (primary) with req ud2 fcb=1 on bus MAIN
(serial /dev/ttyUSB1) failed to send "107B007B16"
(meter) WMZ p0 did not send a response!
(meter) polling WMZ p0 (primary) with req ud2 fcb=1 on bus MAIN
(serial /dev/ttyUSB1) failed to send "107B007B16"
(meter) WMZ p0 did not send a response!
(meter) polling WMZ p0 (primary) with req ud2 fcb=1 on bus MAIN
(serial /dev/ttyUSB1) failed to send "107B007B16"
```

weetmuts, 2024-07-29:
the log says: device /dev/ttyUSB1 is already in use and locked.

weetmuts, 2024-07-29:
any other software hogging it?

audiogeek123, 2024-07-29:
Thanks for the hint, connection was probably lost because I disconnected the USB converter (also had to correct, its not connected to a Raspi but to a thin client with proxmox).

```
[20:56:48] INFO: CONFIG_CONF ...
[20:56:48] INFO: {"device":"MAIN=/dev/ttyUSB1:mbus:2400","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[20:56:48] INFO: CONFIG_METERS ...
[20:56:48] INFO: [{"driver":"sensostar:MAIN:mbus","id":"p0","name":"WMZ","pollinterval":"60s"}]
[20:56:48] INFO: Syncing wmbusmeters configuration ...
[20:56:48] INFO: Registering meters ...
[20:56:49] INFO: Adding meter-0001 ...
[20:56:49] INFO: Generating MQTT configuration ... 
[20:56:49] ERROR: Got unexpected response from the API:
Service not enabled
[20:56:49] INFO: Broker  will be used.
[20:56:49] INFO: 
MQTT Discovery ...
[20:56:49] INFO:  Adding meter: WMZ ...
[20:56:49] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/p0_difference_c/config
Unable to connect (Lookup error.).
[20:56:53] INFO: Starting web configuration service.
[20:56:53] INFO: Running wmbusmeters ...
Started config MAIN=mbus on /dev/ttyUSB1
[20:56:54] INFO: Starting Nginx...
2024/07/29 20:56:54 [notice] 83#83: using the "epoll" event method
2024/07/29 20:56:54 [notice] 83#83: nginx/1.24.0
2024/07/29 20:56:54 [notice] 83#83: OS: Linux 6.6.33-haos
2024/07/29 20:56:54 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/07/29 20:56:54 [notice] 83#83: start worker processes
2024/07/29 20:56:54 [notice] 83#83: start worker process 236
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
Unable to connect (Invalid arguments provided.).
(shell) /bin/sh exited with non-zero return code: 1
```

audiogeek123, 2024-07-30:
Somehow I got it to to work (installed mosquitto, set logtelegrams=true). I also found out that the ID is the same as the serialno which can be found out in the menu of the heatmeter. Nevertheless it also works with "p0" at the moment, not sure what exactly fixed it but maybe it will be helpful for someone.
Thanks @weetmuts for your great work!

closed 2024-07-30 by audiogeek123