wmbusmeters/wmbusmeters #1334: help config

foXaCe, 2024-08-02:
### Type of request

Question

### OS version

hass os

### wmbusmeters version

last

### Your message goes here

hello i use the addon and do not work for me
it possible to help me please ?
thanks you

[image]

`[15:45:52] INFO: CONFIG_CONF ...
[15:45:52] INFO: {"device":"cul:c1","donotprobe":"/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0","format":"json","logfile":"/dev/stdout","loglevel":"verbose","logtelegrams":"verbose","shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[15:45:52] INFO: CONFIG_METERS ...
[15:45:52] INFO: []
[15:45:52] INFO: Syncing wmbusmeters configuration ...
[15:45:52] INFO: Registering meters ...
[15:45:52] INFO: Generating MQTT configuration ... 
[15:45:53] INFO: Broker core-mosquitto will be used.
[15:45:53] INFO: 
MQTT Discovery ...
[15:45:53] INFO: MQTT Discovery cleanup...
[15:45:54] INFO: Starting web configuration service.
[15:45:54] INFO: Running wmbusmeters ...
No such logtelegrams setting: "verbose"
(wmbusmeters) version: 1.16.1
(config) using device: cul:c1 
(config) number of meters: 0
(serial) tty /dev/ttyACM0 can be accessed
(serialtty) device /dev/ttyACM0 is already in use and locked.
(main) ignoring /dev/ttyACM0, it does not respond as any of the supported wmbus devices.
(serial) tty /dev/ttyAML0 can be accessed
(serialtty) opened /dev/ttyAML0 fd 4 (detect cul)
[15:45:55] INFO: Starting Nginx...
2024/08/02 15:45:55 [notice] 83#83: using the "epoll" event method
2024/08/02 15:45:55 [notice] 83#83: nginx/1.24.0
2024/08/02 15:45:55 [notice] 83#83: OS: Linux 6.6.33-haos
2024/08/02 15:45:55 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/08/02 15:45:55 [notice] 83#83: start worker processes
2024/08/02 15:45:55 [notice] 83#83: start worker process 211
(serialtty) closed /dev/ttyAML0 (detect cul)
(cul) are you there? no
(main) ignoring /dev/ttyAML0, it does not respond as any of the supported wmbus devices.
(serial) tty /dev/ttyAML1 can be accessed
(serialtty) opened /dev/ttyAML1 fd 4 (detect cul)
(serialtty) closed /dev/ttyAML1 (detect cul)
(cul) are you there? no
(main) ignoring /dev/ttyAML1, it does not respond as any of the supported wmbus devices.
(serial) tty /dev/ttyUSB0 can be accessed
(serialtty) opened /dev/ttyUSB0 fd 4 (detect cul)
(serialtty) closed /dev/ttyUSB0 (detect cul)
(cul) are you there? no
(main) ignoring /dev/ttyUSB0, it does not respond as any of the supported wmbus devices.
(serial) tty /dev/ttyUSB1 can be accessed
(serialtty) opened /dev/ttyUSB1 fd 4 (detect cul)
(serialtty) closed /dev/ttyUSB1 (detect cul)
(cul) are you there? yes
Started config cul on /dev/ttyUSB1 listening on c1
(cul) on /dev/ttyUSB1
(serialtty) opened /dev/ttyUSB1 fd 4 (cul)
(main) regular reset of cul  on /dev/ttyUSB1 will happen every 82800 seconds
(cul) set link mode c
(serial) tty /dev/ttyUSB2 can be accessed
(serialtty) device /dev/ttyUSB2 is already in use and locked.
(main) ignoring /dev/ttyUSB2, it does not respond as any of the supported wmbus devices.
No meters configured. Printing id:s of all telegrams heard!`

foXaCe, 2024-08-02:
i use this how to and this devices 

https://wmbusmeters.github.io/wmbusmeters-wiki/nanoCUL.html

foXaCe, 2024-08-03:
my meter is Kamstrup flowiq2200

BIBOLV, 2024-10-06:
Hi,
Log says that cul device is detected under /dev/ttyUSB1 and listening for telegrams. Are you confident that you have devices that transmits some data?

closed 2025-01-06 by github-actions[bot]