wmbusmeters/wmbusmeters #1338: PA-iM871A dongle support

AUG-7, 2024-08-09:
### Type of request

Question

### wmbusmeters version

1.15.0

### Your message goes here

I have installed wmbusmeters via Snap on a Beaglebone Black (Debian).
In the USB port of the Beaglebone, there is a PA-iM871A W-Mbus dongle.

Next, I modified the wmbusmeters.conf file:

The command grep ttyUSB gives this result: [ 57.668598] usb 1-1: cp210x converter now attached to ttyUSB0

I tried many options in the /etc/wmbusmeters.conf file, including:

device=auto:t1
device=/dev/ttyUSB0:im871a:c1,t1
device=im871a:c1,t1
im871a[00919B58]:t1
But when I restart the service and then check the log files, nothing gives the desired effect.

A message like this always appears:

python
Code kopiëren
No wmbus device detected, waiting for a device to be plugged in.
(shell) invoking /bin/pidof failed!
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
(shell) invoking /bin/pidof failed!
[ALARM SpecifiedDeviceNotFound] the device im871a[00919B58]:t1 is not working
No wmbus device detected, waiting for a device to be plugged in.
Can someone please help me get this working?

AUG-7, 2024-10-03:
I have been trying for over 2 months to read my gas meter using wmbusmeters. This is my setup:

- Raspberry Pi 4;
- PA-iM871A USB stick;
- Flonidan G6SRT gas meter;
- AES key;

I am sure the USB stick works correctly because I can read the gas meter telegrams using a GUI analyser tool on Windows (both T and C messages). But as mentioned, I want to do this via my Raspberry Pi via this software.

My wmbusmeters.conf looks like this:

loglevel=normal
# Remember to change auto here to the device you are going to use in production.
#device=auto:c1,t1
device=/dev/ttyUSB0:im871a[00929B63]:c1,t1
logfile=/var/snap/wmbusmeters/common/logs/wmbusmeters.log
# Set to true to capture all received telegrams in log file.
logtelegrams=false
format=json
# Enable the meterfiles to write telegrams to disk.
#meterfiles=/var/lib/wmbusmeters/meter_readings
#meterfilesaction=overwrite
# Enable execution of a shell command for each received telegram. For example: curl or mqtt
#shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/ -m ""
#shell=psql water -c "insert into consumption values ('',,'') "
# The alarmshell is executed when a problem with the receiving radio hardware is detected.
#alarmshell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters_alarm -m " "
# The alarmtimeout and expected activity is also used to detect failing receiving radio hardware.
#alarmtimeout=1h
#alarmexpectedactivity=mon-sun(00-23)

I also tried #device=auto:c1,t1, but without success.

And I added a meter file in this folder:
/var/snap/wmbusmeters/common/etc/wmbusmeters.d/

It is unclear to me how I should name the file and which extension it should have.
I have named the file **flonidan.conf** for now. The file contains the following content:

name=Flodidan2
driver=auto
id=49660523

After a restart, I do see additional lines in the log file, but I can’t get past this:

Expected im871a on /dev/ttyUSB0 but did not find it! Ignoring tty!
No wmbus device detected, waiting for a device to be plugged in.
Expected im871a on /dev/ttyUSB0 but did not find it! Ignoring tty!
No wmbus device detected, waiting for a device to be plugged in.

And the folder meterreadings remains empty.

Can someone point out what I am doing wrong and help me get my gas meter readings? 
More people here on the forum are getting stuck, I will provide a detailed explanation if I manage to get it working :)

Thanks in advance!

BIBOLV, 2024-10-06:
Could you check that it is really PA-iM871A dongle? That is very specific hardware for monitoring and analyzing. If it really is then I suspect that it could be different from mainstream iM871A dongle and that could be cause.

closed 2025-01-06 by github-actions[bot]