wmbusmeters/wmbusmeters #1342: Add support for BMT Hydrodigit water meter version 0x17

juhkai, 2024-08-12:
### Type of request

Existing device improvement

### Meter description

Add support for new BMT Hydrodigit water meter version: type 0x07, version 0x17.

Driver auto detection suggests to use c5isf, but this is a Hydrodigit meter so hydrodigit driver should be suggested.

https://github.com/wmbusmeters/wmbusmeters/blob/6874b3392fa38171ae70e85827ad0efc24a004dd/src/driver_hydrodigit.cc#L34

The telegram format for water meter version 0x17 is documented in "Protocol specification" section of this document:

https://bmetersuk.com/wp-content/uploads/2022/04/HYDRODIGIT_COMBO_v1.4.3_ENG.pdf

### Logged telegrams from meter in decrypted format

```shell
Received telegram from: 03297611
          manufacturer: (BMT) BMETERS, Italy (0x9b4)
                  type: Water meter (0x07)
                   ver: 0x17
                device: rtlwmbus[00000001]
                  rssi: 81 dBm
                driver: unknown!
telegram=|_2444B4091176290317078C20D27A220000200C1341000000046D0B2C0A380F030000000000|+383
```

jukkapirinen, 2024-09-12:
I have this meter too with both lorawan and wmbus enabled and using with the hydrodigit driver. 
The content is fields are not showing up, maybe the meter has not those enabled.

```
# ./wmbusmeters --analyze=hydrodigit 2444B4090000000017078C204D7A27000020_0C1387790000046D22300C390F030003000000
Auto driver    : not found!
Similar driver : c5isf 08/15
Using driver   : hydrodigit 00/00
000   : 24 length (36 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : b409 dll-mfct (BMT)
004   : 00000000 dll-id (00000000)
008   : 17 dll-version
009   : 07 dll-type (Water meter)
010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : 4d ell-acc
013   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
014   : 27 tpl-acc-field
015   : 00 tpl-sts-field (OK)
016   : 0020 tpl-cfg 2000 (synchronous )
018   : 0C dif (8 digit BCD Instantaneous value)
019   : 13 vif (Volume l)
020 C!: 87790000 ("total_m3":7.987)
024   : 04 dif (32 Bit Integer/Binary Instantaneous value)
025   : 6D vif (Date and time type)
026 C!: 22300C39 ("meter_datetime":"2024-09-12 16:34")
030 C?: 0F manufacturer specific data 030003000000
031  ?: *** 03 frame content unknown, please open issue with this telegram for driver improvement

{
    "media":"water",
    "meter":"hydrodigit",
    "name":"",
    "id":"00000000",
    "meter_datetime":"2024-09-12 16:34",
    "total_m3":7.987,
    "contents":"unknown, please open issue with this telegram for driver improvement",
    "timestamp":"2024-09-12T15:23:57Z"
}

```

jukkapirinen, 2024-09-13:
Working here with the driver
https://github.com/jukkapirinen/wmbusmeters/

Seems the status bits (MS Code) is not according to definition with the examples. The manufacturer specific data might be different between the versions.

so there is from one meter model recorded  (0x13, 0x07) 

0x15
"Backflow, alarms and monthly data"

0x95 
"Backflow, leak date, alarms and monthly data"

0x03 from newer meter (Hydrodigit-s1?)  (0x17, 0x06)
Not sure, maybe it has battery data and and backflow data

But not matching withe example data

| Bit Position (7 to 0) | 7 | 6 | 5 | 4 | 3 | 2 | 1 | 0 |
| --------------------- | - | - | - | - | - | - | - | - |
| **Bit Value (0x15)**  | 0 | 0 | 0 | 1 | 0 | 1 | 0 | 1 |
| **Bit Value (0x95)**  | 1 | 0 | 0 | 1 | 0 | 1 | 0 | 1 |
| **Bit Value (0x03)** |  0 | 0 | 0 | 0 | 0 | 0 | 1 | 1 |
| **Description**       | monthly data| memory2 | memory1 | burst date | leak date | fraud date | backflow data | battery data|

Guessing from these two examples bit-7 would be leak date.

SovakPaleny, 2024-10-11:
Hello, 
I have the hydrodigit version 0x17, but MS Code 8F. Example data:
```
4c44b4097284090317077ad30000000c1343920000046d22370b3a0f8f00a03700000000000000001a00003100004d0000a40000b30000f300002701002d0100e4010000000012000000000000
```

closed 2025-01-11 by github-actions[bot]