wmbusmeters/wmbusmeters #1347: meter detection did not match the selected driver qheat! correct driver is: unknown!

gotsila, 2024-08-20:
### Type of request

Bug

### HAOS version

2024.7.2

### wmbusmeters version

1.16.1

### Your message goes here

Hello,
i use several Qundis Meters Water and Qheat. The last Qheat i install in my House have this warnings in wmbusmeters protokoll:

```
(meter) 68019283_qheat_heizung_dg: for me? 68019283 in 68019283
(meter) 68019283_qheat_heizung_dg: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 68019283 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x46
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
```

can you help me? 
Let me know if you need any other details. 

Thank you very much.

gotsila, 2024-09-04:
i am now updated to HAOS version 2024.8.0 and wmbusmeters version 1.17.1 the same warning exists,

but i have now this: the sensor jumpes from 0,5 to 5 and  back. the real on the display is 5 kWh
it seams to be a rounding problem:

**
[image]
**

weetmuts, 2024-09-04:
You have to post a telegram here for us to debug the problem. Use --logtelegrams on command line or logtelegrams=true in config file.

gotsila, 2024-09-05:
Hello, here the captured log for the unknown meter :
its a Wärmemengenzähler QUNDIS Qheat 5 with Vendor# HMR5000G1020

`(meter) 68019283_qheat_heizung_dg: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 68019283 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x46
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|5844934483920168463778_077983920168934446040DFF5F3500823000005E0107C006FFFF05000000FFFF000000001F38050000000080008000800080008000800080000000000000050000002F02FD170000046D030D0539|+22`

now i found a second various message:

`(meter) 51975583_qwater_kalt_dg_kueche: meter detection did not match the selected driver qwater! correct driver is: wme5
(meter) Not printing this warning again for id: 51975583 mfct: (QDS) Qundis, Germany (0x4493) type: Water meter (0x07) ver: 0x1a
telegram=|49449344835597511A0778_0DFF5F350082390000110007C113FFFF91880100FF2C330000001F382385010000800080030000002F00ED00F900030129015500420161012F046D010D0539|+48`

hope i have the right snippes found
THX

gotsila, 2024-09-18:
hello again,
was this the right String for Telegram? do you need more information?
cu

gotsila, 2024-09-28:
Have to restart the add-on. so a chance for me to take an actual Telegram for the qheat - Problem:
hope it helps

`(meter) 68019283_qheat_heizung_dg: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 68019283 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x46
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|5844934483920168463778_077983920168934446040DFF5F350082680000100007C006FFFF78020000FFFF000000001F38050000000080008000800080008000800080000000000000050000002F02FD170000046D070C1C39|+98`

gotsila, 2024-10-02:
I have ask the vendor because a hardware problem with this meter. but we have anaylized and the meter is ok. 
he said the OMS protocal on this device is more up to date then my other older meters.
Now my question, is it possible to help with programming this driver, or to anaylize where the problem is?

Where can i start to digging into the driver code?

gotsila, 2024-11-05:
Hello again,

can anyone helpme, please?

gotsila, 2024-12-09:
Hello again,
have now upgraded to actual wmbus 1.18.0-RC1
the same error exists. 

`(meter) 68019283_qheat_warmwasser_gesamt: for me? 68019283.M=QDS.V=46.T=37 in 68019283
(meter) 68019283_qheat_warmwasser_gesamt: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 68019283 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x46
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) 68019283_qheat_warmwasser_gesamt: yes for me
(meter) created 68019283_qheat_warmwasser_gesamt qheat 68019283 not-encrypted
(meter) used meter template 68019283_qheat_warmwasser_gesamt 68019283 qheat to match 68019283.M=QDS.V=46.T=37
(meter) started meter 3 (68019283_qheat_warmwasser_gesamt 68019283 qheat) identity mode: id R68019283
(wmbus) parseDLL @0 89
(wmbus) parseELL @10 79
(wmbus) parseNWL @10 79
(wmbus) parseAFL @10 79
(wmbus) parseTPL @10 79
(meter) 68019283_qheat_warmwasser_gesamt: for me? 68019283.M=QDS.V=46.T=37 in 68019283
(meter) 68019283_qheat_warmwasser_gesamt: meter detection did not match the selected driver qheat! correct driver is: unknown!
(meter) Not printing this warning again for id: 68019283 mfct: (QDS) Qundis, Germany (0x4493) type: Radio converter (meter side) (0x37) ver: 0x46
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) 68019283_qheat_warmwasser_gesamt: yes for me
(meter) 68019283_qheat_warmwasser_gesamt(3) qheat  handling telegram from 68019283.M=QDS.V=46.T=37
(meter) 68019283_qheat_warmwasser_gesamt 68019283.M=QDS.V=46.T=37 "5844934483920168463778077983920168934446040DFF5F3500827B00005F0107C006FFFF40090000FFFF000000001E3B7708000000800080008000800000000000000500000055012D01E6002F02FD170000046D170E093C"
(wmbus) parseDLL @0 89
(telegram) DLL L=58 C=44 (from meter SND_NR) M=4493 (QDS) A=68019283 VER=46 TYPE=37 (Radio converter (meter side)) (driver unknown!) DEV=im871a[00103921] RSSI=-41
(wmbus) parseELL @10 79
(wmbus) parseNWL @10 79
(wmbus) parseAFL @10 79
(wmbus) parseTPL @10 79
(telegram) TPL CI=78
telegram=|5844934483920168463778_077983920168934446040DFF5F3500827B00005F0107C006FFFF40090000FFFF000000001E3B7708000000800080008000800000000000000500000055012D01E6002F02FD170000046D170E093C|+29
(meters) trying field info status(Text)[0]...
(meters) using field info status(Text)[0] to extract 02FD17 at offset 81
(meters) trying field info total_energy_consumption(Energy)[1]...
(meters) trying field info last_month_date(Text)[2]...
(meters) trying field info last_month_energy_consumption(Energy)[3]...
(meters) trying field info last_year_date(Text)[4]...
(meters) trying field info last_year_energy_consumption(Energy)[5]...
(meters) trying field info device_date_time(Text)[6]...
(meters) using field info device_date_time(Text)[6] to extract 046D at offset 85
(meters) trying field info device_error_date(Text)[7]...
(meter) AnyEnergyVIF total_energy_consumption_kwh decoded kwh default kwh value 94 (scale 1)
(meter) AnyEnergyVIF last_year_energy_consumption_kwh decoded kwh default kwh value 0 (scale 1)
(meter) AnyEnergyVIF last_month_energy_consumption_kwh decoded kwh default kwh value 87.7 (scale 1)
(qheat) 000   : 58 length (88 bytes)(OK) ("total_energy_consumption_kwh":94) ("last_year_date":"2127-15-31") ("last_year_energy_consumption_kwh":0) ("last_month_date":"2024-11-30") ("last_month_energy_consumption_kwh":87.7)
(qheat) 001   : 44 dll-c (from meter SND_NR)
(qheat) 002   : 9344 dll-mfct (QDS)
(qheat) 004   : 83920168 dll-id (68019283)
(qheat) 008   : 46 dll-version
(qheat) 009   : 37 dll-type (Radio converter (meter side))
(qheat) 010   : 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
(qheat) 011   : 07 dif (64 Bit Integer/Binary Instantaneous value)
(qheat) 012   : 79 vif (Enhanced identification)
(qheat) 013 C?: 8392016893444604
(qheat) 021   : 0D dif (variable length Instantaneous value)
(qheat) 022   : FF vif (Manufacturer specific)
(qheat) 023   : 5F vife (?)
(qheat) 024   : 35 varlen=53
(qheat) 025 C?: 00827B00005F0107C006FFFF40090000FFFF000000001E3B7708000000800080008000800000000000000500000055012D01E6002F
(qheat) 078   : 02 dif (16 Bit Integer/Binary Instantaneous value)
(qheat) 079   : FD vif (Second extension FD of VIF-codes)
(qheat) 080   : 17 vife (Error flags (binary))
(qheat) 081 C!: 0000 ("status":"OK")
(qheat) 083   : 04 dif (32 Bit Integer/Binary Instantaneous value)
(qheat) 084   : 6D vif (Date and time type)
(qheat) 085 C!: 170E093C ("device_date_time":"2024-12-09 14:23")
(qheat) 013 C!: *** 83920168 tpl-id (68019283)
[... 9 lines cut ...]
(shell) arg "-c"
(shell) arg "/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON""
(shell) env "METER_ID=68019283"
(shell) env "METER_NAME=68019283_qheat_warmwasser_gesamt"
(shell) env "METER_TYPE=qheat"
(shell) env "METER_DRIVER="
(shell) env "METER_JSON={"_":"telegram","media":"heat","meter":"qheat","name":"68019283_qheat_warmwasser_gesamt","id":"68019283","last_month_energy_consumption_kwh":87.7,"last_year_energy_consumption_kwh":0,"total_energy_consumption_kwh":94,"device_date_time":"2024-12-09 14:23","last_month_date":"2024-11-30","last_year_date":"2127-15-31","status":"OK","timestamp":"2024-12-09T13:19:23Z","device":"im871a[00103921]","rssi_dbm":-41}"
[... cut ...]
(shell) env "METER_TOTAL_ENERGY_CONSUMPTION_KWH=94"
(shell) env "METER_LAST_MONTH_DATE=2024-11-30"
(shell) env "METER_LAST_MONTH_ENERGY_CONSUMPTION_KWH=87.7"
(shell) env "METER_LAST_YEAR_DATE=2127-15-31"
(shell) env "METER_LAST_YEAR_ENERGY_CONSUMPTION_KWH=0"
(shell) env "METER_DEVICE_DATE_TIME=2024-12-09 14:23"
(shell) env "METER_DEVICE_ERROR_DATE=null"
(shell) env "METER_DEVICE=im871a[00103921]"
(shell) env "METER_RSSI_DBM=-41"
(shell) env "METER_FIRST_TELEGRAM=true"
(shell) waiting for child 493 to complete.
(shell) /bin/sh: return code 0

gotsila, 2025-02-02:
i have no look up at the protokoll. i think the new qheat meter have a new protokoll version:

here ist a list of all my 3 good readable protokolls:
`67370747_qundis_qheat_heizung_dg 67370747.M=QDS.V=23.T=04 "3C4493443969113723377247073767934423042F0000200C05108105004C0590170400426C1F3CCC080541720500C2086C3F31326CFFFF046D1E082232"
67370746_qheat_heizung_og 67370746.M=QDS.V=23.T=04 "3C449344616911372337724607376793442304110000200C05119339004C0510853700426C1F3CCC080587833900C2086C3F31326CFFFF046D20082232"
67370749_qundis_qheat_heizung_eg 67370749.M=QDS.V=23.T=04 "3C449344916911372337724907376793442304C50000200C05605212004C0520821000426C1F3CCC080541441200C2086C3F31326CFFFF046D1F082232"
`
and this one ist the problem protokoll:
`68019283_qheat_warmwasser_gesamt 68019283.M=QDS.V=46.T=37 "5844934483920168463778077983920168934446040DFF5F3500829400007E0007C006FFFF781300001F3C411100003F3168130000008000800000000000000500000055012D01E6000801E3002F02FD170000046D1F092232"`

i think wmbusmeters can't read the new V=46.T=37 properly

gotsila, 2025-02-02:
in this moment iam get a good readable Telegram from the meter:
`68019283_qheat_warmwasser_gesamt 68019283.M=QDS.V=46.T=04 "414493448392016846377283920168934446049B0000200C06781300004C0641110000426C1F3CCC080668130000C2086C3F3102FD170000326CFFFF046D110A2232"`

the T=37 is a long telegram
the T=04 is a shorter telegram (same lengt as my other 3 meters)

Could it be the meter sends two protokolls?

gotsila, 2025-02-02:
Ok, i think i got the problem. i am only should use T=04

`(meter) 68019283_qheat_warmwasser_gesamt(6) qheat  handling telegram from 68019283.M=QDS.V=46.T=37
(meter) 68019283_qheat_warmwasser_gesamt 68019283.M=QDS.V=46.T=37 "5844934483920168463778077983920168934446040DFF5F350082A800000F0007C006FFFF781300001F3C411100003F3168130000008000800000000000000500000055012D01E6000801E3002F02FD170000046D000C2232"
(shell) env "METER_NAME=68019283_qheat_warmwasser_gesamt"
(shell) env "METER_JSON={"_":"telegram","media":"heat","meter":"qheat","name":"68019283_qheat_warmwasser_gesamt","id":"68019283","last_month_energy_consumption_kwh":136.8,"last_year_energy_consumption_kwh":114.1,"total_energy_consumption_kwh":137.8,"device_date_time":"2025-02-02 12:00","device_error_date":"2127-15-31","last_month_date":"2025-01-31","last_year_date":"2024-12-31","status":"OK","timestamp":"2025-02-02T10:55:50Z","device":"im871a[00103921]","rssi_dbm":-39}"
(meter) 68019283_qheat_warmwasser_gesamt: for me? 51969119.M=QDS.V=1a.T=07 in 68019283
(meter) 68019283_qheat_warmwasser_gesamt: not for me: no match
(meter) 68019283_qheat_warmwasser_gesamt: for me? 68019283.M=QDS.V=46.T=37,68019283.M=QDS.V=46.T=04 in 68019283
(meter) 68019283_qheat_warmwasser_gesamt: yes for me
(meter) 68019283_qheat_warmwasser_gesamt(6) qheat  handling telegram from 68019283.M=QDS.V=46.T=04
(meter) 68019283_qheat_warmwasser_gesamt 68019283.M=QDS.V=46.T=04 "41449344839201684637728392016893444604A90000200C06781300004C0641110000426C1F3CCC080668130000C2086C3F3102FD170000326CFFFF046D010C2232"
(shell) env "METER_NAME=68019283_qheat_warmwasser_gesamt"
(shell) env "METER_JSON={"_":"telegram","media":"heat","meter":"qheat","name":"68019283_qheat_warmwasser_gesamt","id":"68019283","last_month_energy_consumption_kwh":1368,"last_year_energy_consumption_kwh":1141,"total_energy_consumption_kwh":1378,"device_date_time":"2025-02-02 12:01","device_error_date":"2127-15-31","last_month_date":"2025-01-31","last_year_date":"2024-12-31","status":"OK","timestamp":"2025-02-02T10:56:05Z","device":"im871a[00103921]","rssi_dbm":-39}"`

wrong:
the T=34 has "total_energy_consumption_kwh":137.8 

right:
the T=04 has "total_energy_consumption_kwh":1378

how can i now dismiss the T=34?

please i am stuck here, can anyone help me further?
could i script anything to detect and ignore the T=34?

gotsila, 2025-02-22:
hello @weetmuts greets from Germany.

i saw you are worked at a Problem with to telegrams in your file simulation_qheat_bad.txt located here: [wmbusmeters](https://github.com/wmbusmeters/wmbusmeters/tree/master)/[simulations](https://github.com/wmbusmeters/wmbusmeters/tree/master/simulations)

Please let me now if i should test anything, i will help as good as i can.

good luck and greets again

schmyd, 2025-02-27:
Hi @gotsila und Grüße,

did you made any progress? I also see that there are two different message types. Ideally the 'wrong' message can be interpreted correctly, or at least filtered. But I have no clue how to start here.
As I noticed that there are multiple threads about the qheat issue I am trying to combine the analysis in #1497.

gotsila, 2025-03-01:
Hello,
i have digged in deep for the last few month. 

Now at the end: the solution for me was just easy: almost too easy :

in HA / wmbusmeters-Add-on i changed the driver from qheat to qheat_55_us
as i try this first, mont ago, i didn't received any data. so I thought it didn't work that way.

But i have forgotten to make a json file with the same name as the driver.

0. make a Backup ;-)
1. driver = qheat_55_us
2. in \\my_ha\config\wmbusmeters\etc\mqtt_discovery\  copy the file "qheat.json" to "qheat_55_us.json"
3. restart wmbusmeters Addon

[image]

closed 2025-03-01 by gotsila

eporsche, 2025-03-02:
@gotsila great find! 👍