wmbusmeters/wmbusmeters #1348: wmbusmeters creates lots of zombie processes

probutus, 2024-08-21:
### Type of request

Enhancement

### OS version

Debian Bookworm

### wmbusmeters version

1.16.1 @ commit b3771a9b8e5ee72217a01370956d8923f54f4e78 

### Your message goes here

wmbusmeters created over 32.000 zombie processes while running for several weeks:

I'm using rtlsdr dongle with rtlsdr 
● wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/lib/systemd/system/wmbusmeters.service; enabled; preset: enabled)
     Active: active (running) since Tue 2024-07-09 06:33:23 CEST; 1 month 13 days ago
       Docs: https://github.com/wmbusmeters/wmbusmeters
             man:wmbusmeters(1)
   Main PID: 15988 (wmbusmetersd)
      Tasks: 751 (limit: 3932)
        CPU: 3w 2d 6h 7min 28.510s
     CGroup: /system.slice/wmbusmeters.service
             ├─15988 /usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid
             ├─16002 /bin/sh -c "ERRFILE=\$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=\$ERRFILE ;  date -Iseconds > \$ERRFILE ; tail >
             ├─16005 tail -f /tmp/wmbusmeters_rtlsdr.a92IsHPBjxZn
             ├─16006 /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 -
             └─16007 /usr/bin/rtl_wmbus -s -f

Jul 09 06:33:22 andromeda systemd[1]: Starting wmbusmeters.service - "wmbusmeters service"...
Jul 09 06:33:23 andromeda wmbusmetersd[15987]: (wmbusmeters) started /run/wmbusmeters/wmbusmeters.pid
Jul 09 06:33:23 andromeda systemd[1]: Started wmbusmeters.service - "wmbusmeters service".

`
wmbusme+ 31928 15988  0 Aug18 ?        00:00:00 [sh] <defunct>

wmbusme+ 32001 15988  0 Jul14 ?        00:00:00 [sh] <defunct>

wmbusme+ 32005 15988  0 Jul23 ?        00:00:00 [sh] <defunct>

wmbusme+ 32026 15988  0 Jul28 ?        00:00:00 [sh] <defunct>

wmbusme+ 32053 15988  0 Aug03 ?        00:00:00 [sh] <defunct>

wmbusme+ 32076 15988  0 Jul12 ?        00:00:00 [sh] <defunct>

wmbusme+ 32101 15988  0 Aug14 ?        00:00:00 [sh] <defunct>

wmbusme+ 32103 15988  0 Jul29 ?        00:00:00 [sh] <defunct>

wmbusme+ 32144 15988  0 Aug09 ?        00:00:00 [sh] <defunct>

wmbusme+ 32209 15988  0 Jul20 ?        00:00:00 [sh] <defunct>

wmbusme+ 32308 15988  0 06:22 ?        00:00:00 [sh] <defunct>

wmbusme+ 32319 15988  0 Jul14 ?        00:00:00 [sh] <defunct>

wmbusme+ 32348 15988  0 Jul13 ?        00:00:00 [sh] <defunct>

wmbusme+ 32420 15988  0 Jul22 ?        00:00:00 [sh] <defunct>

wmbusme+ 32427 15988  0 Aug20 ?        00:00:00 [sh] <defunct>

wmbusme+ 32439 15988  0 Jul14 ?        00:00:00 [sh] <defunct>

wmbusme+ 32443 15988  0 Aug14 ?        00:00:00 [sh] <defunct>

wmbusme+ 32476 15988  0 Jul29 ?        00:00:00 [sh] <defunct>

wmbusme+ 32482 15988  0 Jul25 ?        00:00:00 [sh] <defunct>

wmbusme+ 32483 15988  0 Aug11 ?        00:00:00 [sh] <defunct>

wmbusme+ 32564 15988  0 Jul21 ?        00:00:00 [sh] <defunct>

wmbusme+ 32586 15988  0 Jul15 ?        00:00:00 [sh] <defunct>

wmbusme+ 32642 15988  0 Aug09 ?        00:00:00 [sh] <defunct>

wmbusme+ 32706 15988  0 Jul27 ?        00:00:00 [sh] <defunct>
`

probutus, 2024-08-21:
I assume it is this process who dies and does not exit cleanly:

 /bin/sh -c ERRFILE=$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus -s -f

weetmuts, 2024-08-22:
32000 is a bit odd. It should restart once per day. Can you see exactly which process is a zombie, is it the shell?

weetmuts, 2024-08-22:
Can you try running with:

rtlwmbus:CMD(/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s -f)
This will pipe stderr to null and avoid the silly workaround that the shell script is about.

The shell scripts ERRFILE etc is a workaround for this bug in rtl_sdr:
https://github.com/osmocom/rtl-sdr/commit/142325a93c6ad70f851f43434acfdf75e12dfe03

Bug: If stderr is sent to a pipe, then rtl_sdr cannot be terminated using sigterm/sigint etc. it goes into 100% cpu usage. Send stderr to a file and it can be restarted, therefore the odd script.

probutus, 2024-08-22:
From what I see it's the shell process itself. To double confirm I will have a look on the next Zombie. I will also check your recommendation. Thanks for the quick feedback!

probutus, 2024-08-22:
also the new cmdline did not help. There are still popping up "sh" zombie processes as children of wmbusmeters:

root@andromeda:/proc/1101# ps ax | grep def
 1101 ?        Z      0:00 [sh] <defunct>
 5094 ?        Z      0:00 [sh] <defunct>
 5397 pts/1    S+     0:00 grep def
root@andromeda:/proc/1101# systemctl restart wmbusmeters
root@andromeda:/proc/1101# ps ax | grep def
 5447 pts/1    S+     0:00 grep def
root@andromeda:/proc/1101# systemctl status wmbusmeters
● wmbusmeters.service - "wmbusmeters service"
     Loaded: loaded (/lib/systemd/system/wmbusmeters.service; enabled; preset: enabled)
     Active: active (running) since Fri 2024-08-23 00:45:28 CEST; 15s ago
       Docs: https://github.com/wmbusmeters/wmbusmeters
             man:wmbusmeters(1)
    Process: 5426 ExecStartPre=/bin/mkdir -p /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 5427 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/lib/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
    Process: 5428 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 5429 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 5430 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 5431 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 5432 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
   Main PID: 5434 (wmbusmetersd)
      Tasks: 7 (limit: 3932)
        CPU: 7.796s
     CGroup: /system.slice/wmbusmeters.service
             ├─5434 /usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid
             ├─5439 /bin/sh -c "/usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s -f"
             ├─5440 /usr/bin/rtl_sdr -d 0 -f 868.625M -s 1.6e6 -
             └─5441 /usr/bin/rtl_wmbus -s -f

Aug 23 00:45:28 andromeda systemd[1]: Starting wmbusmeters.service - "wmbusmeters service"...
Aug 23 00:45:28 andromeda systemd[1]: Started wmbusmeters.service - "wmbusmeters service".

-> It looks like somewhere else within a 'sh' process is forked but not properly closed

weetmuts, 2024-08-23:
What do you see if you do loglevel=debug
does the shell restart all the time? Do you get any meter readings?

probutus, 2024-08-24:
still running with loglevel debug since yesterday evening but no Zombies appeared

I noticed the following in the logfile where loglevel was still on normal:
(wmbusmeters) started meter 2 (zenner_wmz 32002044 c5isf) identity mode: id R32002044
(wmbusmeters) started meter 3 (zenner_ww_kuche_1OG 55155387 minomess) identity mode: id R24888061
(wmbusmeters) started meter 4 (strom_eg 61067911 esyswm) identity mode: id R63290879
(wmbusmeters) started meter 5 (zenner_kw_kuche_1OG 55152544 minomess) identity mode: id R15227402
(wmbusmeters) started meter 6 (zenner_ww_2OG_L 55024123 minomess) identity mode: id R24821193
(wmbusmeters) started meter 7 (zenner_ww_bad_1OG 55134110 minomess) identity mode: id R25298872
[2024-08-22_01:54:10] (shell) /bin/sh exited with non-zero return code: 224
(wmbusmeters) started meter 8 (zenner_kw_2OG 55016267 minomess) identity mode: id R15143045
[2024-08-22_02:47:47] (shell) /bin/sh exited with non-zero return code: 96
[2024-08-22_04:00:58] (shell) /bin/sh exited with non-zero return code: 64
[2024-08-22_07:52:17] (shell) /bin/sh exited with non-zero return code: 32
[2024-08-22_10:31:42] (shell) /bin/sh exited with non-zero return code: 128
[2024-08-22_11:17:14] (shell) /bin/sh exited with non-zero return code: 192
[2024-08-22_12:07:09] (shell) /bin/sh exited with non-zero return code: 192
[2024-08-22_12:21:48] (shell) /bin/sh exited with non-zero return code: 96
[2024-08-22_14:58:47] (shell) /bin/sh exited with non-zero return code: 64
[2024-08-22_16:23:25] (shell) /bin/sh exited with non-zero return code: 192
[2024-08-22_17:40:17] (shell) /bin/sh exited with non-zero return code: 192
[2024-08-22_17:58:42] (shell) /bin/sh exited with non-zero return code: 192
[2024-08-22_18:08:17] (shell) /bin/sh exited with non-zero return code: 192
[2024-08-22_19:51:43] (wmbus) resetting /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BJEFn114J20-if00-port0:mbus[?]
[2024-08-22_19:51:48] (wmbus) reset completed /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BJEFn114J20-if00-port0:mbus[?]
[2024-08-22_19:55:42] (shell) /bin/sh exited with non-zero return code: 192
[2024-08-22_19:59:47] (shell) /bin/sh exited with non-zero return code: 224
(wmbusmeters) logging started 2024-08-22 20:16:43 using 1.6.0-45-g19834d8
[2024-08-22_20:16:43] Started config MAIN=mbus on /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BJEFn114J20-if00-port0
[2024-08-22_20:16:43] Started config rtlwmbus[00000001] listening on c1,t1
(wmbusmeters) waiting for telegrams
[2024-08-22_20:16:45] (memory) rss 3743744 peak 3.57 MiB
[2024-08-22_20:16:45] Lost 00000001 closing rtlwmbus[00000001]
[2024-08-22_20:16:45] Started config rtlwmbus[00000001] listening on c1,t1
[2024-08-22_20:16:47] Lost 00000001 closing rtlwmbus[00000001]
[2024-08-22_20:16:47] Started config rtlwmbus[00000001] listening on c1,t1
[2024-08-22_20:16:49] Lost 00000001 closing rtlwmbus[00000001]
[2024-08-22_20:16:49] Started config rtlwmbus[00000001] listening on c1,t1
[2024-08-22_20:22:15] (meter) allgemeinstrom 24450291 did not send a response!

I will update as soon as I have a Zombie process with debuglog

probutus, 2024-08-24:
got the first to Zombie processes:

13580 ?        Ssl   48:30 /usr/bin/python3 /usr/local/bin/solaredge2mqtt
25771 ?        Z      0:00 [sh] <defunct>
29991 ?        Ssl  211:39 /usr/bin/tvheadend -f -p /run/tvheadend.pid -u hts -g video --satip_xml http://192.168.10.2:49000/satipdesc.xml
31329 ?        I      0:00 [kworker/2:2]
32586 ?        Z      0:00 [sh] <defunct>
32764 ?        Ssl   14:26 /usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid

[Uploading wmbusmeters.log.tar.bz2.txt…]()

with the mentioning in debuglog:
(meter) polling allgemeinstrom 24450291 (secondary) with req ud2 bus MAIN
(shell) waiting for child 25771 to complete.
(serial /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BJEFn114J20-if00-port0) sent "105BFD5816"
(shell) /bin/sh: return code 0
(mbus) parse MBUS DLL @0 152
(mbus)  "6892926808017291024524941501029A0000008C1004651204008C1104651204008C2004000000008C21040000000002FDC9FF01E40002FDDBFF01000002ACFF0101008240ACFF01010002FDC9FF02E30002FDDBFF02000002ACFF0200008240ACFF02000002FDC9FF03E40002FDDBFF03020002ACFF0304008240ACFF03000002FF68010002ACFF0005008240ACFF00010001FF1300EB16"
(meter) strom_eg: for me? p1.M=@@@.V=00.T=00,24450291.M=ELT.V=01.T=02 in 61067911,R63290879
(meter) strom_eg: not for me: no match
(mbus) parse MBUS DLL @0 152

and

(meter) polling allgemeinstrom 24450291 (secondary) with req ud2 bus MAIN
(shell) waiting for child 32586 to complete.
(serial /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BJEFn114J20-if00-port0) sent "105BFD5816"
(shell) /bin/sh: return code 0
(wmbus) parseDLL @0 108
(wmbus) parseELL @10 98
(wmbus) parseNWL @13 95
(wmbus) parseAFL @13 95
(wmbus) parseTPL @30 78

I have the complete logfile attached aswell

closed 2024-11-23 by github-actions[bot]