wmbusmeters/wmbusmeters #1350: Support for iM891A (replacement for discontinued iM871A)

Bob2345de, 2024-08-26:
### Type of request

Enhancement

### OS version

debian

### wmbusmeters version

current sable

### Your message goes here

IMST has discontinued and replaced the iM871a with the iM891a-xl, which is based on a Cypress chipset and is not compatible with older versions.
Can we expect drivers for this device into be included in wmbusmeters?

weetmuts, 2024-08-26:
Probably, but I need to get a physical device first. :-)

weetmuts, 2024-08-26:
Do you have one?

weetmuts, 2024-08-26:
I.e. can you participate in testing?

Bob2345de, 2024-08-26:
Yes, I have one and would be happy to help with testing.

Henry221164, 2024-08-28:
I do have also one and i try to get it to work.Only with the WM-Bus-Gateway-Package-R13 it seems to work.I also  would be happy to help with testing

weetmuts, 2024-08-28:
Great! It has got a completely new serial protocol, so a whole new wmbus_im891a.cc must be written.

kongomongo, 2024-08-28:
hi there, I would also offer to help if I can be helpful. 

Does it means this device has to be reverse engineered or is there an official serial comm document that can be used? Can anyone guesstimate how much time this will take? days? weeks? months?

Thanks!

ElizabethHoffman, 2024-08-29:
I also got iU891A. The serial protocol is well described here:
https://wireless-solutions.de/downloadfile/wireless-m-bus-software-im891a-iu891a/ 

> hi there, I would also offer to help if I can be helpful.
> 
> Does it mean this device has to be reverse engineered or is there an official serial comm document that can be used? Can anyone guesstimate how much time this will take? days? weeks? months?
> 
> Thanks!

Hausi91, 2024-09-11:
I also got my hands on an IMST iU891A-XL. I found this thread while trying to set it up, encountered some issues, and now I know. I could also participate in testing, as I would really like to use wmbusmeters with this hardware

dkadioglu, 2024-09-12:
I also have one and I also offer to participate in testing and experimenting!

mckar, 2024-09-12:
> I also got an iU891A. The serial protocol is well described here: https://wireless-solutions.de/downloadfile/wireless-m-bus-documents-im891a-iu891a/ (WM-Bus Gateway HCI Protocol Specification V2.1)

The document mentions:
"API Example Code: IMST offers C/C++ and Python example code, including message encoders and decoders for HCI messages outlined in this document."
Has anyone contacted IMST for this sample code already?
@weetmuts Would that help maybe or would it help more if you would have the physical device and someone should send it to you for testing?

weetmuts, 2024-09-12:
@mckar Thanks, but IMST actually sent me a device already! I can start developing two weeks from now.
I have to do some real work first, you know, the kind where you get paid. :-)
It is great to see so many eager beta-testers!

ElizabethHoffman, 2024-09-13:
> The document mentions: "API Example Code: IMST offers C/C++ and Python example code, including message encoders and decoders for HCI messages outlined in this document." Has anyone contacted IMST [...]

I think the C/C++ and Python example code referred to in the documentation is included in the software package:
https://wireless-solutions.de/downloadfile/wireless-m-bus-software-im891a-iu891a/

mckar, 2024-09-17:
> I do have also one and i try to get it to work.Only with the WM-Bus-Gateway-Package-R13 it seems to work.I also would be happy to help with testing
Did you get it working with the WM Bus Studio App or python example? When starting Studio App on Windows 11 I only get an error with missing ftd2xx.dll. After installing some FTDI drivers (like written in StudioReleaseNotes.html - also it does not makes sense because new USB stick is Cypress based) and copying DLLs to Studio Installation directory I get another application error 0xc000007b.

Hausi91, 2024-09-17:
@mckar I’m having the same issues, but the Python sample did work for me—though only on Windows 10 (I also tried Windows 11 and the Studio app, dll, drivers etc.). In fact i need those drivers for Windows 10 as well to get the device corretly working. All attempts to add the missing DLL failed with the same error. With python i can scan for my device and successfully find my meter, but the documentation states that the firmware is not capable of reading meters—only devices of type 'other'. It also suggests contacting support if further assistance is needed. That's where stopped in the weekend.

From FirmwareReleaseNotes.html/pdf
"Target Devices:  This firmware is specifically intended for WM-Bus devices that fall under the category of "Other." This likely refers to devices that are not typically meters (e.g., data concentrators, repeaters, or other communication modules within the WM-Bus network)."

juhkai, 2024-09-18:
@Hausi91 I asked about this from the IMST and got this response:

`The iU891A-XL is pre-programmed with the wM-Bus Gateway firmware. It is suitable to collect the wM-Bus messages from devices working as “meter”. In other words, the iU891A with wM-Bus Gateway firmwares is intended to be used as "Other."`

weetmuts, 2024-09-21:
Support pushed to wmbusmeters. Please test!

dkadioglu, 2024-09-21:
> Support pushed to wmbusmeters. Please test!

Great news, thank you very much! Unfortunately, my meters are all configured to not transmit during the night and on sundays, so I will test next week.

Andi252, 2024-09-23:
Hello, I tried the new driver and on my ubuntu 22.04 and it worked without problems.
The only point I found: I could not use **/dev/ttyUSB0** as device.
I had to use device=**/dev/ttyACM0**:iu891a:c1,t1

dkadioglu, 2024-09-24:
I had the chance to test. Unfortunately, I only receive my water meters. My heating meters are not received by the iM891a, although my iM871a receives them perfectly. Also, I see many CRC errors in the debug log. What kind of output/logging do you need from me?

Andi252, 2024-09-24:
@dkadioglu what antenna do you use for your new usb stick? I used the antenna the IMST offered in their webshop.
I get about the same RSSI values on both devices. I receive message from different meters with different RSSI values.

The "(iu8 80b) bad crc got cb49 expected 0284" looks like a copy-paste leftover.

dkadioglu, 2024-09-24:
> @dkadioglu what antenna do you use for your new usb stick? I used the antenna the IMST offered in their webshop.
> I get about the same RSSI values on both devices. I receive message from different meters with different RSSI values.

I also ordered the one IMST is offering in their shop.

weetmuts, 2024-09-24:
@dkadioglu It is probably something wrong in the removal of the slip framing....perhaps I can fix it based on the log you already posted but  if you can catch som log lines with debug enabled where we can see the serial bytes arriving, that would help as well.

dkadioglu, 2024-09-24:
> @dkadioglu It is probably something wrong in the removal of the slip framing....perhaps I can fix it based on the log you already posted but  if you can catch som log lines with debug enabled where we can see the serial bytes arriving, that would help as well. 

Does this help:
```
(serial) received binary "C00920C6"
(iu891a) partial frame, expecting more. "C00920C6"
(serial) received binary "F5D666020505B09E44FA123882"
(iu891a) partial frame, expecting more. "C00920C6F5D666020505B09E44FA123882"
(serial) received binary "222200077A4B0090250285DBDCB219ABD2FB2A04"
(iu891a) partial frame, expecting more. "C00920C6F5D666020505B09E44FA123882222200077A4B0090250285DBDCB219ABD2FB2A04"
(serial) received binary "569A4D76ACE64117542738FFB47DCC9500B1368351"
(iu891a) partial frame, expecting more. "C00920C6F5D666020505B09E44FA123882222200077A4B0090250285DBDCB219ABD2FB2A04569A4D76ACE64117542738FFB47DCC9500B1368351"
(serial) received binary "0682B527EEA57ADBDC593C13381F4EDF9540EDF023"
(iu891a) partial frame, expecting more. "C00920C6F5D666020505B09E44FA123882222200077A4B0090250285DBDCB219ABD2FB2A04569A4D76ACE64117542738FFB47DCC9500B13683510682B527EEA57ADBDC593C13381F4EDF9540EDF023"
(serial) received binary "BA827D1CDFBFDD6E6B3F9E4840E933B0DF20FCB3"
(iu891a) partial frame, expecting more. "C00920C6F5D666020505B09E44FA123882222200077A4B0090250285DBDCB219ABD2FB2A04569A4D76ACE64117542738FFB47DCC9500B13683510682B527EEA57ADBDC593C13381F4EDF9540EDF023BA827D1CDFBFDD6E6B3F9E4840E933B0DF20FCB3"
(serial) received binary "D1281DF4E418C3EC8C506EA6D04FEB752FA578CBB1"
(iu891a) partial frame, expecting more. "C00920C6F5D666020505B09E44FA123882222200077A4B0090250285DBDCB219ABD2FB2A04569A4D76ACE64117542738FFB47DCC9500B13683510682B527EEA57ADBDC593C13381F4EDF9540EDF023BA827D1CDFBFDD6E6B3F9E4840E933B0DF20FCB3D1281DF4E418C3EC8C506EA6D04FEB752FA578CBB1"
(serial) received binary "61DE55EEDC69D6FAEB83D749DBDC4DCB9B1D54A4"
(iu891a) partial frame, expecting more. "C00920C6F5D666020505B09E44FA123882222200077A4B0090250285DBDCB219ABD2FB2A04569A4D76ACE64117542738FFB47DCC9500B13683510682B527EEA57ADBDC593C13381F4EDF9540EDF023BA827D1CDFBFDD6E6B3F9E4840E933B0DF20FCB3D1281DF4E418C3EC8C506EA6D04FEB752FA578CBB161DE55EEDC69D6FAEB83D749DBDC4DCB9B1D54A4"
(serial) received binary "D5CE5BF22BAE5288B6A15B7B3F6279FEB04A7493FB"
(iu891a) partial frame, expecting more. "C00920C6F5D666020505B09E44FA123882222200077A4B0090250285DBDCB219ABD2FB2A04569A4D76ACE64117542738FFB47DCC9500B13683510682B527EEA57ADBDC593C13381F4EDF9540EDF023BA827D1CDFBFDD6E6B3F9E4840E933B0DF20FCB3D1281DF4E418C3EC8C506EA6D04FEB752FA578CBB161DE55EEDC69D6FAEB83D749DBDC4DCB9B1D54A4D5CE5BF22BAE5288B6A15B7B3F6279FEB04A7493FB"
(serial) received binary "0DF0B2B44AADBE8B4418F80B745DC0"
(iu880b) bad crc got 5d74 expected a7a7
(iu891a) bad frame, clearing. "C00920C6F5D666020505B09E44FA123882222200077A4B0090250285DBDCB219ABD2FB2A04569A4D76ACE64117542738FFB47DCC9500B13683510682B527EEA57ADBDC593C13381F4EDF9540EDF023BA827D1CDFBFDD6E6B3F9E4840E933B0DF20FCB3D1281DF4E418C3EC8C506EA6D04FEB752FA578CBB161DE55EEDC69D6FAEB83D749DBDC4DCB9B1D54A4D5CE5BF22BAE5288B6A15B7B3F6279FEB04A7493FB0DF0B2B44AADBE8B4418F80B745DC0"
```

weetmuts, 2024-09-25:
Thanks @dkadioglu ! As expected, the slip frame removal had a bug. Fixed pushed. Please test!

mckar, 2024-09-26:
I did a fresh installation of edge ha-addon, but see no telegrams. Is the HA addon already using latest version of wmbusmeters or only after you create a new release? As it shows iu880b, I think it's not using latest version.

2024/09/26 12:43:48 [notice] 83#83: start worker process 207
Started auto iu880b[00062dbe] on /dev/ttyACM0 listening on t1
Warning! Desired link modes t1 cannot be set for device /dev/ttyACM0:iu880b[00062dbe]
No meters configured. Printing id:s of all telegrams heard!

Hausi91, 2024-09-26:
Sadly, I havent had enough time to test yet. Since I didnt have a running Linux environment, I tried to build from source and run it as a daemon in a VM, but something went wrong while building. I havent had enough time and won't find any in the coming two weeks.

@mckar I think you need to test the latest wmbusmeters build. I checked the HA add-on releases, and it doesnt seem to be updated yet.

Henry221164, 2024-09-26:
I had the same issue as @mckar.Clean install of ha-addon edge 1.17.1-1
2024/09/26 16:01:51 [notice] 83#83: start worker process 206
Started auto iu880b[00062dbe] on /dev/ttyACM0 listening on t1
Warning! Desired link modes t1 cannot be set for device /dev/ttyACM0:iu880b[00062dbe]
No meters configured. Printing id:s of all telegrams heard!
This happend with auto detect.With /dev/ACM0:iu891a:c1,t1 as discript in https://github.com/wmbusmeters/wmbusmeters/commit/26760f3838aaffe0830a73316fc6b0a36ec2300b the Output is:
[16:08:42] INFO: CONFIG_CONF ...
[16:08:42] INFO: {"device":"/dev/ttyACM0:iu891:c1,t1","donotprobe":"/dev/ttyUSB0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[16:08:42] INFO: CONFIG_METERS ...
[16:08:42] INFO: []
[16:08:42] INFO: Syncing wmbusmeters configuration ...
[16:08:42] INFO: Registering meters ...
[16:08:43] INFO: Generating MQTT configuration ... 
[16:08:43] INFO: Broker core-mosquitto will be used.
[16:08:43] INFO: 
MQTT Discovery ...
[16:08:43] INFO: MQTT Discovery cleanup...
[16:08:44] INFO: Starting web configuration service.
[16:08:44] INFO: Running wmbusmeters ...
Not a valid device "/dev/ttyACM0:iu891:c1,t1"
s6-rc: info: service legacy-services: stopping
[14:08:44] INFO: Service restart after closing

It seems that the iU891a dongle dosen´t exit in the system or it wasn´t regoniced by the addon.

weetmuts, 2024-09-27:
I am no an expert on the ha-addon, my focus is on the c++ decoder code. 

However I added the /dev/ttyACM0 to the config.json in the wmbusmeters-ha-addon since the new iu891a dongle appears here.

There is also something strange with the upload part of the build process. 
https://github.com/wmbusmeters/wmbusmeters-ha-addon/actions/runs/11027851498/job/30628300104

I do not know how to fix this. @BIBOLV do you know why this happens?

weetmuts, 2024-09-27:
Latest attempt:
https://github.com/wmbusmeters/wmbusmeters-ha-addon/actions/runs/11065928513/job/30747442612

Henry221164, 2024-09-27:
After Update the Addon(https://github.com/wmbusmeters/wmbusmeters-ha-addon/commit/a4825a452e302b0afe782bf474708786a26aa438),i tried once more.Her is the the result:

[10:55:06] INFO: CONFIG_CONF ...
[10:55:06] INFO: {"device":"/dev/ttyACM0:iu891a:c1,t1","donotprobe":"/dev/ttyUSB0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[10:55:06] INFO: CONFIG_METERS ...
[10:55:06] INFO: []
[10:55:06] INFO: Syncing wmbusmeters configuration ...
[10:55:06] INFO: Registering meters ...
[10:55:06] INFO: Generating MQTT configuration ... 
[10:55:06] INFO: Broker core-mosquitto will be used.
[10:55:07] INFO: 
MQTT Discovery ...
[10:55:07] INFO: MQTT Discovery cleanup...
[10:55:08] INFO: Starting web configuration service.
[10:55:08] INFO: Running wmbusmeters ...
Not a valid device "/dev/ttyACM0:iu891a:c1,t1"
s6-rc: info: service legacy-services: stopping
[08:55:08] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
this is with device= /dev/ttyACM0:iu891a:c1,t1 same as before.
and here is the result with device:auto:t1

[11:03:19] INFO: CONFIG_CONF ...
[11:03:19] INFO: {"loglevel":"normal","device":"auto:t1","donotprobe":"/dev/ttyAMA0","logtelegrams":"false","format":"json","logfile":"/dev/stdout","shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[11:03:19] INFO: CONFIG_METERS ...
[11:03:19] INFO: []
[11:03:19] INFO: Syncing wmbusmeters configuration ...
[11:03:19] INFO: Registering meters ...
[11:03:19] INFO: Generating MQTT configuration ... 
[11:03:19] INFO: Broker core-mosquitto will be used.
[11:03:19] INFO: MQTT Discovery cleanup...
[11:03:20] INFO: Starting web configuration service.
[11:03:20] INFO: Running wmbusmeters ...
[11:03:21] INFO: Starting Nginx...
2024/09/27 11:03:21 [notice] 83#83: using the "epoll" event method
2024/09/27 11:03:21 [notice] 83#83: nginx/1.24.0
2024/09/27 11:03:21 [notice] 83#83: OS: Linux 6.6.46-haos
2024/09/27 11:03:21 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/09/27 11:03:21 [notice] 83#83: start worker processes
2024/09/27 11:03:21 [notice] 83#83: start worker process 206
Started auto iu880b[00062dbe] on /dev/ttyACM0 listening on t1
Warning! Desired link modes t1 cannot be set for device /dev/ttyACM0:iu880b[00062dbe]
No meters configured. Printing id:s of all telegrams heard!

Here are the parameters i use for the Test.
[image]
[image]

weetmuts, 2024-09-28:
After som fixes in the github workflow, which stopped working all by itself some time ago.
The final step of updating the version nr in the edge addon worked.
It is a weird version, but it is at least different to the previous one. "version": "1.17.1-1-1"

 @Hausi91 @Henry221164 check again. If you see any reference to iu880 in the log, then your docker/addon has not updated.

BIBOLV, 2024-09-28:
Version number is correct. If some fixes will still be needed, it will have to wait when also my dongle will arrive.

Henry221164, 2024-09-28:
Test with Version edge 1.17.1-1-2 and this parameter:
[image]
and this is the output:
[14:42:54] INFO: CONFIG_CONF ...
[14:42:54] INFO: {"device":"auto:t1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[14:42:54] INFO: CONFIG_METERS ...
[14:42:54] INFO: []
[14:42:54] INFO: Syncing wmbusmeters configuration ...
[14:42:54] INFO: Registering meters ...
[14:42:54] INFO: Generating MQTT configuration ... 
[14:42:54] INFO: Broker core-mosquitto will be used.
[14:42:55] INFO: 
MQTT Discovery ...
[14:42:55] INFO: MQTT Discovery cleanup...
[14:42:56] INFO: Starting web configuration service.
[14:42:56] INFO: Running wmbusmeters ...
[14:42:56] INFO: Starting Nginx...
2024/09/28 14:42:56 [notice] 83#83: using the "epoll" event method
2024/09/28 14:42:56 [notice] 83#83: nginx/1.24.0
2024/09/28 14:42:56 [notice] 83#83: OS: Linux 6.6.46-haos
2024/09/28 14:42:56 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/09/28 14:42:56 [notice] 83#83: start worker processes
2024/09/28 14:42:56 [notice] 83#83: start worker process 209
172.30.32.2 - - [28/Sep/2024:14:42:57 +0200] "GET / HTTP/1.1" 200 1300 "http://192.168.178.111:8123/c8a990ad_wmbusmeters-ha-addon-edge" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
172.30.32.2 - - [28/Sep/2024:14:42:57 +0200] "GET /bootstrap.min.css HTTP/1.1" 200 221221 "http://192.168.178.111:8123/api/hassio_ingress/O_IX67h7XgsJ2Vyb1MX730BX5JU6PWOPh-lmHcz_MWU/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
172.30.32.2 - - [28/Sep/2024:14:42:57 +0200] "GET /bootstrap.bundle.min.js HTTP/1.1" 200 80761 "http://192.168.178.111:8123/api/hassio_ingress/O_IX67h7XgsJ2Vyb1MX730BX5JU6PWOPh-lmHcz_MWU/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
172.30.32.2 - - [28/Sep/2024:14:42:57 +0200] "GET /main.js HTTP/1.1" 200 9981 "http://192.168.178.111:8123/api/hassio_ingress/O_IX67h7XgsJ2Vyb1MX730BX5JU6PWOPh-lmHcz_MWU/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
172.30.32.2 - - [28/Sep/2024:14:42:57 +0200] "GET /wmbusmeters.png HTTP/1.1" 200 55338 "http://192.168.178.111:8123/api/hassio_ingress/O_IX67h7XgsJ2Vyb1MX730BX5JU6PWOPh-lmHcz_MWU/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
172.30.32.2 - - [28/Sep/2024:14:42:57 +0200] "GET /get_json HTTP/1.1" 200 320 "http://192.168.178.111:8123/api/hassio_ingress/O_IX67h7XgsJ2Vyb1MX730BX5JU6PWOPh-lmHcz_MWU/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
Started auto iu891a[00202021] on /dev/ttyACM0 listening on t1
No meters configured. Printing id:s of all telegrams heard!

it seems to work.My Watermeter works next on Monday at 06:00.

Next try with other parameter:
device: /dev/ttyACM0:iu891a:t1
here is the Output:
[14:53:03] INFO: CONFIG_CONF ...
[14:53:03] INFO: {"device":"/dev/ttyACM0:iu891a:t1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[14:53:03] INFO: CONFIG_METERS ...
[14:53:03] INFO: [{"driver":"qwater","id":"07288946","key":"18D5EB48F5B8137EDDF1F6D65DFFCC42","name":"Wasserzähler"}]
[14:53:03] INFO: Syncing wmbusmeters configuration ...
[14:53:03] INFO: Registering meters ...
[14:53:03] INFO: Adding meter-0001 ...
[14:53:03] INFO: Generating MQTT configuration ... 
[14:53:03] INFO: Broker core-mosquitto will be used.
[14:53:04] INFO: 
MQTT Discovery ...
[14:53:04] INFO:  Adding meter: Wasserzähler ...
[14:53:04] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/07288946_rssi_dbm/config
[14:53:04] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/07288946_total_m3/config
[14:53:04] INFO: MQTT Discovery cleanup...
[14:53:05] INFO: Starting web configuration service.
[14:53:05] INFO: Running wmbusmeters ...
Started config iu891a[00202021] on /dev/ttyACM0 listening on t1
[14:53:06] INFO: Starting Nginx...
2024/09/28 14:53:06 [notice] 85#85: using the "epoll" event method
2024/09/28 14:53:06 [notice] 85#85: nginx/1.24.0
2024/09/28 14:53:06 [notice] 85#85: OS: Linux 6.6.46-haos
2024/09/28 14:53:06 [notice] 85#85: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/09/28 14:53:06 [notice] 85#85: start worker processes
2024/09/28 14:53:06 [notice] 85#85: start worker process 239

it seems also work.
My Axioma Watermeter is only for the week configured,that means it only works from Mo-Fr 06:00-18:00 so i dont get
any comunication yet.

dkadioglu, 2024-09-28:
> Thanks @dkadioglu ! As expected, the slip frame removal had a bug. Fixed pushed. Please test!

Thank you very much. As far as I can see, the CRC errors are gone. However, I can still not see any messages of my heating meters Sensostar Q. Could there be anything missing in the implementation? With the iM871a I can receive them.

BIBOLV, 2024-09-28:
You use same config for both? Do you have debug enabled and can provide some logs?

dkadioglu, 2024-09-29:
> You use same config for both? Do you have debug enabled and can provide some logs?

Yes, both wmbusmeters instances use the same config, except regarding the line for the device. Unfortunately I cannot provide debug output from both currently, as my meters only transmit weekdays. I will try to do that as soon as possible.

Henry221164, 2024-09-30:
Monday first Test: parameter: device=/dev/serial/by-id/usb-IMST_iU891A-if00:iu891a:t1
Output: 
[09:06:36] INFO: CONFIG_CONF ...
[09:06:36] INFO: {"device":"/dev/serial/by-id/usb-IMST_iU891A-if00:iu891a:t1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":true,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[09:06:36] INFO: CONFIG_METERS ...
[09:06:36] INFO: [{"driver":"qwater","id":"07288946","key":"18D5EB48F5B8137EDDF1F6D65DFFCC42","name":"Wasserzähler"}]
[09:06:36] INFO: Syncing wmbusmeters configuration ...
[09:06:36] INFO: Registering meters ...
[09:06:36] INFO: Adding meter-0001 ...
[09:06:36] INFO: Generating MQTT configuration ... 
[09:06:36] INFO: Broker core-mosquitto will be used.
[09:06:37] INFO: 
MQTT Discovery ...
[09:06:37] INFO:  Adding meter: Wasserzähler ...
[09:06:37] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/07288946_rssi_dbm/config
[09:06:37] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/07288946_total_m3/config
[09:06:37] INFO: MQTT Discovery cleanup...
[09:06:38] INFO: Starting web configuration service.
[09:06:38] INFO: Running wmbusmeters ...
Expected iu891a on /dev/serial/by-id/usb-IMST_iU891A-if00 but did not find it! Ignoring tty!
No wmbus device detected, waiting for a device to be plugged in.
[09:06:39] INFO: Starting Nginx...
2024/09/30 09:06:39 [notice] 83#83: using the "epoll" event method
2024/09/30 09:06:39 [notice] 83#83: nginx/1.24.0
2024/09/30 09:06:39 [notice] 83#83: OS: Linux 6.6.46-haos
2024/09/30 09:06:39 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/09/30 09:06:39 [notice] 83#83: start worker processes
2024/09/30 09:06:39 [notice] 83#83: start worker process 241
172.30.32.2 - - [30/Sep/2024:09:06:40 +0200] "GET / HTTP/1.1" 200 1290 "http://192.168.178.111:8123/hassio" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
172.30.32.2 - - [30/Sep/2024:09:06:40 +0200] "GET /bootstrap.min.css HTTP/1.1" 200 221221 "http://192.168.178.111:8123/api/hassio_ingress/O_IX67h7XgsJ2Vyb1MX730BX5JU6PWOPh-lmHcz_MWU/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
172.30.32.2 - - [30/Sep/2024:09:06:40 +0200] "GET /bootstrap.bundle.min.js HTTP/1.1" 200 80762 "http://192.168.178.111:8123/api/hassio_ingress/O_IX67h7XgsJ2Vyb1MX730BX5JU6PWOPh-lmHcz_MWU/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
172.30.32.2 - - [30/Sep/2024:09:06:40 +0200] "GET /main.js HTTP/1.1" 200 9981 "http://192.168.178.111:8123/api/hassio_ingress/O_IX67h7XgsJ2Vyb1MX730BX5JU6PWOPh-lmHcz_MWU/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
172.30.32.2 - - [30/Sep/2024:09:06:40 +0200] "GET /wmbusmeters.png HTTP/1.1" 200 55338 "http://192.168.178.111:8123/api/hassio_ingress/O_IX67h7XgsJ2Vyb1MX730BX5JU6PWOPh-lmHcz_MWU/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
172.30.32.2 - - [30/Sep/2024:09:06:40 +0200] "GET /get_json HTTP/1.1" 200 463 "http://192.168.178.111:8123/api/hassio_ingress/O_IX67h7XgsJ2Vyb1MX730BX5JU6PWOPh-lmHcz_MWU/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:130.0) Gecko/20100101 Firefox/130.0"
Started config iu891a[00202021] on /dev/serial/by-id/usb-IMST_iU891A-if00 listening on t1
(meter) Wasserzähler: meter detection did not match the selected driver qwater! correct driver is: q400
(meter) Not printing this warning again for id: 07288946 mfct: (AXI) UAB Axis Industries, Lithuania (0x709) type: Water meter (0x07) ver: 0x10
telegram=|434409074689280710078C203C900F002C25C42D01009ACACD6D8C8E2ED27A3C002007102F2F_046D09081E390413D02C0000446D000001394413740F00

It looks like it works.
First look says there be no device but after a few seconds it run and get a telegram.

travismarius, 2024-09-30:
I can approve that. I´ve just installed the update and now the iU891A ist correct discovered and receives telegrams from heat meter. Thanks. I´ll post an update if a have configured the meters.

dkadioglu, 2024-10-02:
> Yes, both wmbusmeters instances use the same config, except regarding the line for the device. Unfortunately I cannot provide debug output from both currently, as my meters only transmit weekdays. I [...]

This is the relevant output for one of my heating meters, received via iM871a:
```
[2024-10-02 18:52:33] (serial) received binary "A5C203A944C5144473352200047ABB00902519ED95B2B5934D025798E2E21E6A6698C64C58F22D43C0D1870441470E6C2346AA08B4CA3B65760762F6A74D94BF6FC786BD87437C0FE9163A3BD54A68783C8EE9D53214800CEF8C8C54722F4CAB516F734F8860A48D227BCFF89C68650C4FE9ACA10052B4EE8EBF5F053FFD9239589B27BF57181DF27B2908617A13EAC092F2C96535DEBFC6283ED0A6063AE5EF8AA503FD0C05010002FD0B01113FCBB9"
[2024-10-02 18:52:33] (im871a) checkIM871AFrame "A5C203A944C5144473352200047ABB00902519ED95B2B5934D025798E2E21E6A6698C64C58F22D43C0D1870441470E6C2346AA08B4CA3B65760762F6A74D94BF6FC786BD87437C0FE9163A3BD54A68783C8EE9D53214800CEF8C8C54722F4CAB516F734F8860A48D227BCFF89C68650C4FE9ACA10052B4EE8EBF5F053FFD9239589B27BF57181DF27B2908617A13EAC092F2C96535DEBFC6283ED0A6063AE5EF8AA503FD0C05010002FD0B01113FCBB9"
[2024-10-02 18:52:33] (im871a) has_timestamp=0 has_rssi=1 has_crc16=1
[2024-10-02 18:52:33] (im871a) endpoint 2
[2024-10-02 18:52:33] (im871a) msgid 3
[2024-10-02 18:52:33] (im871a) rssi 63 (-91 dBm)
[2024-10-02 18:52:33] (im871a) got crc16 b9cb expected b9cb
[2024-10-02 18:52:33] (im871a) received full frame
[2024-10-02 18:52:33] (wmbus) parseDLL @0 170
[2024-10-02 18:52:33] (wmbus) parseELL @10 160
[2024-10-02 18:52:33] (wmbus) parseNWL @10 160
[2024-10-02 18:52:33] (wmbus) parseAFL @10 160
[2024-10-02 18:52:33] (wmbus) parseTPL @10 160
[2024-10-02 18:52:33] (meter) WMZoben: for me? 22357344.M=EFE.V=00.T=04 in 22357344
[2024-10-02 18:52:33] (meter) WMZoben: yes for me
[2024-10-02 18:52:33] (meter) WMZoben(1) sensostar  handling telegram from 22357344.M=EFE.V=00.T=04
[2024-10-02 18:52:33] (meter) WMZoben 22357344.M=EFE.V=00.T=04 "A944C5144473352200047ABB00902519ED95B2B5934D025798E2E21E6A6698C64C58F22D43C0D1870441470E6C2346AA08B4CA3B65760762F6A74D94BF6FC786BD87437C0FE9163A3BD54A68783C8EE9D53214800CEF8C8C54722F4CAB516F734F8860A48D227BCFF89C68650C4FE9ACA10052B4EE8EBF5F053FFD9239589B27BF57181DF27B2908617A13EAC092F2C96535DEBFC6283ED0A6063AE5EF8AA503FD0C05010002FD0B0111"
[2024-10-02 18:52:33] (wmbus) parseDLL @0 170
[2024-10-02 18:52:33] (telegram) DLL L=a9 C=44 (from meter SND_NR) M=14c5 (EFE) A=22357344 VER=00 TYPE=04 (Heat meter) (driver sensostar) DEV=im871a[01001652] RSSI=-91
[2024-10-02 18:52:33] (wmbus) parseELL @10 160
[2024-10-02 18:52:33] (wmbus) parseNWL @10 160
[2024-10-02 18:52:33] (wmbus) parseAFL @10 160
[2024-10-02 18:52:33] (wmbus) parseTPL @10 160
[2024-10-02 18:52:33] (TPL) num encrypted blocks 9 (144 bytes and remaining unencrypted 11 bytes)
[2024-10-02 18:52:33] (TPL) AES CBC IV decrypting "19ED95B2B5934D025798E2E21E6A6698C64C58F22D43C0D1870441470E6C2346AA08B4CA3B65760762F6A74D94BF6FC786BD87437C0FE9163A3BD54A68783C8EE9D53214800CEF8C8C54722F4CAB516F734F8860A48D227BCFF89C68650C4FE9ACA10052B4EE8EBF5F053FFD9239589B27BF57181DF27B2908617A13EAC092F2C96535DEBFC6283ED0A6063AE5EF8AA503FD0C05010002FD0B0111"
[2024-10-02 18:52:33] (TPL) IV C514447335220004BBBBBBBBBBBBBBBB
[2024-10-02 18:52:33] (TPL) decrypted  "2F2F046D3431023A0406FB06000004132B4E010001FD1700426CFF2C4406D6020000840106FB060000840206FB060000840306FB060000840406FB060000840506FB060000840606FB060000840706A8060000840806CE050000840906E4040000840A06D6020000840B0619010000840C0604000000840D0600000000840E0600000000840F06000000002F2F2F2F2F"
[2024-10-02 18:52:33] (TPL) appended   "2F2F046D3431023A0406FB06000004132B4E010001FD1700426CFF2C4406D6020000840106FB060000840206FB060000840306FB060000840406FB060000840506FB060000840606FB060000840706A8060000840806CE050000840906E4040000840A06D6020000840B0619010000840C0604000000840D0600000000840E0600000000840F06000000002F2F2F2F2F03FD0C05010002FD0B0111"
[2024-10-02 18:52:33] (telegram) TPL CI=7a ACC=bb STS=00 CFG=2590 (synchronous AES_CBC_IV nb=9 cntn=0 ra=0 hc=0)
[2024-10-02 18:52:33] telegram=|A944C5144473352200047ABB0090252F2F_046D3431023A0406FB06000004132B4E010001FD1700426CFF2C4406D6020000840106FB060000840206FB060000840306FB060000840406FB060000840506FB060000840606FB060000840706A8060000840806CE050000840906E4040000840A06D6020000840B0619010000840C0604000000840D0600000000840E0600000000840F06000000002F2F2F2F2F03FD0C05010002FD0B0111|+2769
[2024-10-02 18:52:33] (meters) trying field info fabrication_no(Text)[0]...
[2024-10-02 18:52:33] (meters) trying field info model_version(Text)[1]...
[2024-10-02 18:52:33] (meters) using field info model_version(Text)[1] to extract 03FD0C at offset 162
[2024-10-02 18:52:33] (meters) trying field info parameter_set(Text)[2]...
[2024-10-02 18:52:33] (meters) using field info parameter_set(Text)[2] to extract 02FD0B at offset 168
[2024-10-02 18:52:33] (meters) trying field info on_time(Time)[3]...
[2024-10-02 18:52:33] (meters) trying field info meter_datetime(Text)[4]...
[2024-10-02 18:52:33] (meters) using field info meter_datetime(Text)[4] to extract 046D at offset 19
[2024-10-02 18:52:33] (meters) trying field info status(Text)[5]...
[2024-10-02 18:52:33] (meters) using field info status(Text)[5] to extract 01FD17 at offset 38
[2024-10-02 18:52:33] (meters) trying field info total(Energy)[6]...
[2024-10-02 18:52:33] (meters) using field info total(Energy)[6] to extract 0406 at offset 25
[2024-10-02 18:52:33] (meter) AnyEnergyVIF total_kwh decoded kwh default kwh value 1787 (scale 1)
[... cut ...]
[2024-10-02 18:52:39] (meter) Waschmaschine: for me? 21030710.M=TCH.V=94.T=08 in 22146660
[2024-10-02 18:52:39] (meter) Waschmaschine: not for me: no match
[2024-10-02 18:52:39] (meter) Warm: for me? 21030710.M=TCH.V=94.T=08 in 22076627
[2024-10-02 18:52:39] (meter) Warm: not for me: no match
[2024-10-02 18:52:39] (meter) WMZoben: for me? 21030710.M=TCH.V=94.T=08 in 22357344
[2024-10-02 18:52:39] (meter) WMZoben: not for me: no match
[2024-10-02 18:52:39] (wmbus) telegram from TODO ignored by all configured meters!
```
If you need anything else, please ask.

Eschioche, 2024-11-18:
Still having the issue that iU891A cannot be configured in the wmbusmeters HA addon (version 1.17.1) on Rpi5.

The stick is properly recognized
[image]

The setting device= auto:t1 
leads to a wrong recognition of the device as iu880b

[11:31:17] INFO: CONFIG_CONF ...
[11:31:17] INFO: {"loglevel":"normal","device":"auto:t1","donotprobe":"/dev/ttyAMA0","logtelegrams":"false","format":"json","logfile":"/dev/stdout","shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[11:31:17] INFO: CONFIG_METERS ...
[11:31:17] INFO: []
[11:31:17] INFO: Syncing wmbusmeters configuration ...
[11:31:17] INFO: Registering meters ...
[11:31:17] INFO: Generating MQTT configuration ... 
[11:31:17] INFO: Broker core-mosquitto will be used.
[11:31:17] INFO: MQTT Discovery cleanup...
[11:31:18] INFO: Starting web configuration service.
[11:31:18] INFO: Running wmbusmeters ...
[11:31:19] INFO: Starting Nginx...
2024/11/18 11:31:19 [notice] 83#83: using the "epoll" event method
2024/11/18 11:31:19 [notice] 83#83: nginx/1.24.0
2024/11/18 11:31:19 [notice] 83#83: OS: Linux 6.6.31-haos-raspi
2024/11/18 11:31:19 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/11/18 11:31:19 [notice] 83#83: start worker processes
2024/11/18 11:31:19 [notice] 83#83: start worker process 206
**Started auto iu880b[00062dbe] on /dev/ttyACM1 listening on t1
Warning! Desired link modes t1 cannot be set for device /dev/ttyACM1:iu880b[00062dbe]
No meters configured.** Printing id:s of all telegrams heard!

Other trials and errors with direct references:

device = /dev/ttyACM1:iu981a:t1   --> Not a valid device "/dev/ttyACM1:iu891a:t1"
or
device = /dev/serial/by-id/usb-IMST_iU891A-if00:iu891a:t1  --> Not a valid device "/dev/serial/by-id/usb-IMST_iU891A-if00:iu891a:t1"
s6-rc: info: service legacy-services: stopping

Any hints? 
Thank you!

travismarius, 2024-11-18:
Same here. I´m using the "edge" Add-On.
With that the iu891a is recognized and working fine.

[image]

Eschioche, 2024-11-18:
Wow, just uninstalled the non-edge version and installed the [edge] addon and everything works fine. 
Thanks @travismarius for the hint!
May anyone explain this ?

boyi01, 2024-12-01:
Hello, i'm a little bit confused, i got a iU891A-XL, but no chance to get data from it... 
I work on a Raspberry pi, and i can see the usb device, any advice?

Normaly this should just work by starting (and stop al other processes):
wmbusmeters auto:t1

Or?

travismarius, 2024-12-01:
> Hello, i'm a little bit confused, i got a iU891A-XL, but no chance to get data from it... I work on a Raspberry pi, and i can see the usb device, any advice?
> 
> Normaly this should just work by starting (and stop al other processes): wmbusmeters auto:t1
> 
> Or?

Hi, do you have tried the edge Version of wmbusmeter addon?
Regards, travis

boyi01, 2024-12-01:
@travismarius I just compiled it myself from this repo

joberreiter, 2025-01-02:
I have an iU891A-XL from IMST, and a problem that the device is not always recognized.

Calling `wmbusmeters --verbose --logtelegrams --nodeviceexit /dev/ttyACM0:iu891a:t1`, this is the output:
```
(wmbusmeters) version: _
(config) using device: /dev/ttyACM0:iu891a:t1
(config) number of meters: 0
(serial) tty /dev/ttyACM0 can be accessed
(serialtty) opened /dev/ttyACM0 fd 3 (detect iu891a)
(iu891a) are you there? no.
(serialtty) closed /dev/ttyACM0 (detect iu891a)
Expected iu891a on /dev/ttyACM0 but did not find it! Ignoring tty!
No wmbus device detected. Exiting!
(serial) event loop stopped!
```
Running the exact command immediately after above output once (or twice) again:
```
(wmbusmeters) version: _
(config) using device: /dev/ttyACM0:iu891a:t1
(config) number of meters: 0
(serial) tty /dev/ttyACM0 can be accessed
(serialtty) opened /dev/ttyACM0 fd 3 (detect iu891a)
(serialtty) closed /dev/ttyACM0 (detect iu891a)
(iu891a) are you there? yes 00202074
Started config iu891a[00202074] on /dev/ttyACM0 listening on t1
(iu891a) on /dev/ttyACM0
(serialtty) opened /dev/ttyACM0 fd 3 (iu891a)
(iu891a) get device info
(iu891a) device info:  type=iu891a uid=00000626
(iu891a) get wmbus address
(iu891a) device info: 00202074  type=iu891a uid=00000626
(iu891a) get config
(iu891a) config: RCV_ALL RCV_IND SND_IND RECALIB led flash: 50 ms recalibrate: 1414 ms link modes: t1
(iu891a) got device id 00202074
(main) regular reset of iu891a  on /dev/ttyACM0 will happen every 82800 seconds
(iu891a) set config
(iu871a) set config to set link mode 02
No meters configured. Printing id:s of all telegrams heard!
(telegram) DLL L=43 C=44 (from meter SND_NR) M=0709 (AXI) A=07002065 VER=10 TYPE=07 (Water meter) (driver q400) DEV=iu891a[00202074] RSSI=-50
```

The behaviour (detection / no detection) is not predictable, if i run it through the daemon service, the device is mostly not found.

BIBOLV, 2025-01-02:
Such behavior is not normal. Check dmesg/systems logs and look for frequent usb connect/disconnect messages, power related or other issues. Check if more than one instance of wmbusmeters is not running, inspect carefully wmusmeters logs for some clues. Try using different usb port, do not use usb extenders.

rjurca, 2025-01-11:
I think I have a similar problem:
I'm running wmbusmeter as a Home Assistant add-on, running on a Raspberry pi:
Wmbusmeters (W-MBus to MQTT)
Current version: 1.18.0-RC1

When running with device=auto:t1 I get the following lines in the log:

[22:46:23] INFO: Starting web configuration service.
[22:46:23] INFO: Starting wmbusmeters service.
[22:46:28] INFO: CONFIG_CONF ...
[22:46:28] INFO: {"device":"auto:c1,t1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":true,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
2025/01/11 22:46:30 [notice] 92#92: nginx/1.24.0
2025/01/11 22:46:30 [notice] 92#92: OS: Linux 6.6.62-haos-raspi
2025/01/11 22:46:30 [notice] 92#92: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2025/01/11 22:46:30 [notice] 92#92: start worker processes
2025/01/11 22:46:30 [notice] 92#92: start worker process 180
[22:46:40] INFO: Running wmbusmeters ...
Started auto iu891a[00202095] on /dev/ttyACM0 listening on t1,c1
No meters configured. Printing id:s of all telegrams heard!

However, if I replace device=/dev/ttyACM0:iu891a[00202095]:c1,t1 in the configuration, I get:
[22:49:22] INFO: Running wmbusmeters ...
Expected iu891a on /dev/ttyACM0 but did not find it! Ignoring tty!
No wmbus device detected, waiting for a device to be plugged in.

Any idea why I cannot configure the right device?
BTW, device=iu891a:t1 does not work either.

Thank you
Radu

weetmuts, 2025-01-11:
@rjurca Ah, can you enable loglevel=debug so that we can see when it talks to the iu891a
I might have forgotten to test the iu891a selection by its id.... so there can be bugs.

Do you have multiple iu891a that you want to differentiate between?

rjurca, 2025-01-12:
Thanks for the reply Frederik. I have a single iu891a dongle.

Here's the log output with loglevel=debug:

[19:25:31] INFO: Starting web configuration service.
[19:25:31] INFO: Starting wmbusmeters service.
[19:25:36] INFO: CONFIG_CONF ...
[19:25:36] INFO: {"device":"/dev/ttyACM0:iu891a:t1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"debug","logtelegrams":true,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[19:25:36] INFO: CONFIG_METERS ...
[19:25:36] INFO: []
[19:25:36] INFO: Syncing wmbusmeters configuration ...
[19:25:37] INFO: Starting Nginx...
2025/01/12 19:25:37 [notice] 92#92: using the "epoll" event method
2025/01/12 19:25:37 [notice] 92#92: nginx/1.24.0
2025/01/12 19:25:37 [notice] 92#92: OS: Linux 6.6.62-haos-raspi
2025/01/12 19:25:37 [notice] 92#92: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2025/01/12 19:25:37 [notice] 92#92: start worker processes
2025/01/12 19:25:37 [notice] 92#92: start worker process 173
[19:25:37] INFO: Registering meters ...
[19:25:38] INFO: Generating MQTT configuration ... 
[19:25:40] ERROR: Got unexpected response from the API:
Service not enabled
[19:25:43] INFO: Broker  will be used.
[19:25:45] INFO: 
MQTT Discovery ...
[19:25:46] INFO: MQTT Discovery cleanup...
[19:25:47] INFO: Running wmbusmeters ...
(config) "logtelegrams" "true"
(wmbusmeters) version: 1.18.0
(config) using device: /dev/ttyACM0:iu891a:t1 
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyACM0" iu891a
(serial) check if /dev/ttyACM0 can be accessed
(serial) tty /dev/ttyACM0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,72,84,171,127,250) modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,64,68,244,238,127,248) modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect iu891a)
(serial) received binary "C009206B596D38020502BA4E44B40593547570F1077A9F004005419D2482ECECAB41BAD4031181A72E17D2EF9ADBDCE15C05DAD76E161E19D0F0E244FF21ECC942E150F229C1FB5DF4303AF1CF9DAFE337DBDC8354991464B54D033E880CC0C009206C596D3803FF02CB19442423860358352100A2515F001486C71AB9631ADBDD24658BE712E8C0C0092074596D3803FF02CB1944242386"
(serial /dev/ttyACM0) sent "C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0"
(serial) EVENT thread interrupted
(serial /dev/ttyACM0) sent "C001030424C0"
(serial) EVENT thread interrupted
(serial) received binary "0358352100A2615F001486FE6B639E78AF4E08A0E044E5C0C009207B596D38020502BA4E44B40593547570F1077A9F0040053970813DC87BB99049C3FEDF81210B3A7AC222BE0470C00104006ED2040000BE2D060073050000855AC0"
(iu891a) bad crc got e544 expected c9d9
(iu891a) are you there? no.
(serialtty) closed /dev/ttyACM0 (detect iu891a)
Expected iu891a on /dev/ttyACM0 but did not find it! Ignoring tty!
(serial) EVENT thread interrupted
(verbose) ignoring device t1
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 1
(shell) 233 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>230
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 234 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted

I've also noticed that device=iu891a:t1 works only if loglevel=debug.
Thanks again
Radu

jarkkouitti, 2025-01-13:
Same problem here. It was working fine for few months with version 1.17.

After upgrading to 1.18.0-26 got this error:

[15:37:23] INFO: Running wmbusmeters ...
(config) "logtelegrams" "false"
(wmbusmeters) version: 1.18.0-26-ga316380
(config) using device: /dev/ttyACM0:iu891a:c1 
(config) number of meters: 1
(config) template Vesimittari XXXXXXXX multical21
(lookup) with file/hex "/dev/ttyACM0" iu891a
(serial) check if /dev/ttyACM0 can be accessed
(serial) tty /dev/ttyACM0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,136,20,186,127,188) modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(127,128,131,232,244,127,248) modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect iu891a)
(serial) received binary "C009203B6F7138028015A122442D2C599158681D168D207660E84524CA9A9C7FD4CAE4BD1924C23273F12394738822F3C0C009203D6F7138028015A023442D2C095773751B168D2033427F6D2439716CCBDCBF1BDA05FF56B438AD7227573A8EBDB7C0C009203E6F7138028015BC23442D2C058063741B168D20EAB1459924CA9B9D3BB4E577EE693991DBDD3784305F824947E0BBC0C009"
(serial /dev/ttyACM0) sent "C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0"
(serial) EVENT thread interrupted
(serial /dev/ttyACM0) sent "C001030424C0"
(serial) EVENT thread interrupted
(serial) received binary "20466F7138028015AD23442D2C475873751B168D2080907A6D24201F9576FE49A506475ADA04F444E16E8D3A1051E2C0C009204B6F71380280159E22442D2C599158681D168D2077C00104006EFF040000BE2D0600620400003C7EC0"
(iu891a) bad crc got e251 expected 0409
(iu891a) are you there? no.
(serial) EVENT thread interrupted
(serialtty) closed /dev/ttyACM0 (detect iu891a)
Expected iu891a on /dev/ttyACM0 but did not find it! Ignoring tty!
(verbose) ignoring device c1
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 469 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>466
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 470 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted

lilabeste, 2025-09-28:
Hi there, i also have problems getting wmbusmeters working in homeassistant with an imst iu891a-xl on a pi 5. Are there any solutions yet?

Best regards

weetmuts, 2026-04-01:
I think the stick shouldbe working now. We fixed some detection issues when radio telegrams were being
received during detection.

https://github.com/wmbusmeters/wmbusmeters/issues/1630

Ping @steven1234561