wmbusmeters/wmbusmeters #1354: iU891A – Wireless M-Bus USB Adapter Replacement for iM871A-USB

Henry221164, 2024-08-28:
### Type of request

Question

### HAOS version

2024.8.3

### wmbusmeters version

1.17.1 

### Your message goes here

Is it possible to include the replacement USB-Stick iU891A-XL for the iM871A-USB to use with wmbusmeters addon.Because i get one.It looks like it
was reconized as :
Started auto iu880b[00062dbe] on /dev/ttyACM0 listening on t1
Warning! Desired link modes t1 cannot be set for device /dev/ttyACM0:iu880b[00062dbe]
And it won´t do anything

kongomongo, 2024-08-28:
I have the same iU891A since the iM871A is EOL and is now unavailable from the manufacturer.

I get this from dmesg:
```
[Aug28 14:41] usb 1-1.2: new full-speed USB device number 6 using xhci-hcd
[  +0.163370] usb 1-1.2: New USB device found, idVendor=04b4, idProduct=0003, bcdDevice= 0.00
[  +0.060972] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  +0.024694] usb 1-1.2: Product: iU891A
[  +0.011840] usb 1-1.2: Manufacturer: IMST
[  +0.061312] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device

```

and this in wmbusmeters.log:
```
[2024-08-28_14:41:59] Started auto iu880b[00062dbe] on /dev/ttyACM0 listening on t1
[2024-08-28_14:42:00] Warning! Desired link modes t1 cannot be set for device /dev/ttyACM0:iu880b[00062dbe]

```

Henry221164, 2024-08-28:
Take a Look here:
https://github.com/wmbusmeters/wmbusmeters/issues/1350

weetmuts, 2024-09-21:
Duplicate

closed 2024-09-21 by weetmuts