wmbusmeters/wmbusmeters #1357: Compile wmbusmeters for openWRT - footprint

dkelmatic, 2024-09-03:
### Type of request

Question

### OS version

RutOs7.xx

### wmbusmeters version

1.17.1

### Your message goes here

Hello folks,
first of all thanks to all outhors for the great software! I compiled wmbusmeters for RutOs, which is just a flavour of OpenWrt.
Everything worked out so far, I got all libs and the program itserlf compiled. Now I realize that one of the libs, namely libicu from which libxml2 depends that is needed by wmbusmeters is insanely huge for an embedded system. Everything together is about 12MB and most of it is libicu. I ask myself if wmbusmeters would still work if I compile libxml2 without libicu? Would I have to face problems? I would be fine if I just get a decoded MBus packet out of the program, I actually don't need necessarily the conversion to xml.

I'd be super thankful if someone with a bit more knowledge/overview over the code could answer my questions.

Have a good day everyone,

David

weetmuts, 2024-09-03:
Please try to compile libxml2 without libicu! And let me know how you did it and if it worked. 
I do not think any complicated unicode/locale stuff are used so it should be ok. 

The driver files for wmbusmeters (check wmbusmeters/drivers/src) are parsed using my https://libxmq.org (https://github.com/libxmq/xmq) Imported as xmq.h and xmq.c into wmbusmeters.

weetmuts, 2024-09-03:
The libxml2 and libxslt libraries are already available on normal Linux distributions and on MacOS.

That is why the xmq binary download from https://github.com/libxmq/xmq/releases/tag/2.13.0
is only ~64KB compressed for Linux and Mac! 

I had been thinking about a #define that would replace the full libxml2/libxslt with my own implementation of a DOM that is much more limited, but would be good enough when it is used for config files.

dkelmatic, 2024-09-04:
Hello Frederik,
thanks for your quick response and the details about how the xml lib is used and that it might work without libicu.
I will try it and I will happily share what I figured out as well as my package recipe(s)

Best, 
David

kzajac83, 2024-09-09:
For OpenWRT topics like compile I suggest ask on eko.one.pl forum page or on official OpenWRT.

weetmuts, 2024-09-09:
You should run ./configure before running make. I am pretty sure the OpenWRT packaging process does not require patching of the makefiles (I suspect this might be a ChatGPT distraction), just run the makefile with the right settings. But I am not an experton OpenWRT builds so it is better if you follow @kzajac83 advice and ask in the other forums. 

Wmbusmeters can be built as a deb package so building a OpenWRT package should be possible without major changes. Also remember to add libxslt as a package dep.

closed 2024-12-09 by github-actions[bot]