wmbusmeters/wmbusmeters #1361: Build from master complains on auto detection of every single driver?

kitor, 2024-09-13:
### Type of request

Question

### OS version

Debian 12.5

### wmbusmeters version

wmbusmeters: 1.17.1-4-g3eee107

### Your message goes here

I was going to update `amiplus` driver with new IDs and test cases for Opus 1 and 3 (as described by me and Leo-PL in #1296)

I built wmbusmeters as usual:
```
apt install librtlsdr-dev libxml2-dev rtl-sdr
./configure
make clean
make -j6
make install
systemctl daemon-reload
systemctl enable wmbusmeters
mkdir -p /var/log/wmbusmeters/meter_readings
chown wmbusmeters:wmbusmeters /var/log/wmbusmeters/meter_readings
systemctl start wmbusmeters
```

What I'm surprised is that now every driver complains to me about auto detection failure, like my very own `apatoreitn`:

```
[2024-09-13_21:48:11] (meter) heat: meter detection did not match the selected driver apatoreitn! correct driver is: unknown!
(meter) Not printing this warning again for id: 35365047 mfct: (APA) Apator, Poland (0x8601) type: Heat Cost Allocator (0x08) ver: 0x04
[2024-09-13_21:48:11] (meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
[2024-09-13_21:48:11] (meter) to add support for this unknown mfct,media,version combination
telegram=|19440186475036350408A0_A100003200000000002D310815B018|+375
```
 
while [src/driver_apatoretin.cc](https://github.com/wmbusmeters/wmbusmeters/blob/3eee107dd8f50ca9b5cbfab46a50f3103991240c/src/driver_apatoreitn.cc#L40) clearly matches:
`        di.addDetection(0x8614 /* APT? */, 0x08,  0x04);`

Similar for `apator162`:

```
[2024-09-13_21:42:53] (meter) woda: meter detection did not match the selected driver apator162! correct driver is: unknown!
(meter) Not printing this warning again for id: 00058812 mfct: (APT) Apator, Poland (0x8614) type: Water meter (0x07) ver: 0x05
[2024-09-13_21:42:53] (meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
[2024-09-13_21:42:53] (meter) to add support for this unknown mfct,media,version combination
```

where driver [src/driver_apator162.cc ](https://github.com/wmbusmeters/wmbusmeters/blob/3eee107dd8f50ca9b5cbfab46a50f3103991240c/src/driver_apator162.cc#L39) should match:
`        di.addDetection(MANUFACTURER_APA,  0x07,  0x05);`

Is there something wrong in my env or you also see this issue?

kitor, 2024-09-14:
Bisection points to fccdea9. 288da71 works fine.
```
fccdea9345ee40450b93ccb58b4042efba61906a is the first bad commit
commit fccdea9345ee40450b93ccb58b4042efba61906a
Author: Fredrik Öhrström <oehrstroem@gmail.com>
Date:   Wed Jun 19 21:19:27 2024 +0200

    Fix for meters with weird mfct value, like apator08 and itronheat.

 src/driver_apator08.cc | 4 ++--
 src/meters.cc          | 4 +++-
 2 files changed, 5 insertions(+), 3 deletions(-)
```

Looks like this change fails:
```diff
diff --git a/src/meters.cc b/src/meters.cc
index 05427e4..be3f7d9 100644
--- a/src/meters.cc
+++ b/src/meters.cc
@@ -112,7 +112,9 @@ bool DriverInfo::detect(uint16_t mfct, uchar type, uchar version)
     for (auto &dd : detect_)
     {
         if (dd.mfct == 0 && dd.type == 0 && dd.version == 0) continue; // Ignore drivers with no detection.
-        if (dd.mfct == mfct && dd.type == type && dd.version == version) return true;
+        //some meters send mfct where the first character ist lower case, which results in mfct which are bigger
+        //than 32767, therefore restrict mfct to correct range
+        if (dd.mfct == (mfct & 0x7fff) && dd.type == type && dd.version == version) return true;
     }
     return false;
 }
```

The problem is pretty obvious to spot :) This fixed one case and broke all other cases since `dd.mfct` is not given the same treatment as `mfct` variable (ie: for the test both should be made lowercase).

That pretty much fixes it:
```diff
diff --git a/src/meters.cc b/src/meters.cc
index bc87b6d..e8293db 100644
--- a/src/meters.cc
+++ b/src/meters.cc
@@ -115,7 +115,7 @@ bool DriverInfo::detect(uint16_t mfct, uchar type, uchar version)
         // Some weird meters (aptor08 and itronheat) send a mfct where the first character is lower case,
         // which results in mfct which are bigger than 32767, therefore restrict mfct to correct range
         // and the normal check will work.
-        if (dd.mfct == (mfct & 0x7fff) && dd.type == type && dd.version == version) return true;
+        if ((dd.mfct & 0x7fff) == (mfct & 0x7fff) && dd.type == type && dd.version == version) return true;
     }
     return false;
 }
```

Let me prepare a PR then :)

kitor, 2024-09-14:
I created PR #1362 that includes that fix among other things.

weetmuts, 2024-09-14:
Thanks, but why did not all the test cases fail then?

kitor, 2024-09-14:
Idk how your test system works, but since tests are held in each driver source file I guess it just silently ignores the warning?

kitor, 2024-09-14:
In fact, that is interesting. Built from master head again. From log:

```
[2024-09-14_17:10:29] (meter) woda: meter detection did not match the selected driver apator162! correct driver is: unknown!
(meter) Not printing this warning again for id: 00058812 mfct: (APT) Apator, Poland (0x8614) type: Water meter (0x07) ver: 0x05
[2024-09-14_17:10:29] (meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
[2024-09-14_17:10:29] (meter) to add support for this unknown mfct,media,version combination
telegram=|5A441486128805000507B6_0AFFFFF5450106F41BA8767AF2004085EE13D538A092EFB7206EB937456DD16C8A50D05E9054AD3E628A5AC526948002BC14CAB2F1C74DE09EF30E1F3F9189ADA3F711D3194A3F8E108C27AEA69404A0|+70
```

Running telegram via wmbusmeteres like tests are run:
```
$ wmbusmeters 4E4401061288050005077AF20040852F2F_0F098ED99800000010B79B020041555B4204678101433F067301B6960200CD890200EA7B0200B1700200A03E40F003FFFFFFFFFFFFFFFFFFFFFFFFFF45F5 test apator162 00058812 NOKEY
test	00058812	170.935 m³	2024-09-14 17:12:19
```

```
$ wmbusmeters 4E4401061288050005077AF20040852F2F_0F098ED99800000010B79B020041555B4204678101433F067301B6960200CD890200EA7B0200B1700200A03E40F003FFFFFFFFFFFFFFFFFFFFFFFFFF45F5 test auto 00058812 NOKEY
test	00058812	170.935 m³	2024-09-14 17:12:44
```

For reference:
```
$ cat /etc/wmbusmeters.conf 
loglevel=normal
device=rtlwmbus:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a)
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=append
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=/opt/grafana/meters.py "$METER_JSON"
$ cat /etc/wmbusmeters.d/woda.conf 
name=woda
type=apator162
id=00148686,00058812
key=00000000000000000000000000000000
```

So idk... seems those warnings were going only to wmbusmeters.log while doing regular listen, but they didn't appear when parsing telegram from command args... I guess that may be a hint where to look as this looks like some oversight.

I see that my PR is merged already, that's cool. Regarding your comment there:

> Thanks! I do not think you need (dd.mfct & 0x7fff)
because the driver mfct should be correct, it was broken with 0x8601 /* APA? */
but you fixed this. But it does not hurt either. :-)

As you see I made update for `amiplus` and `apatoreitn` yet in the example above I had this warning also from `apator162` :) And that `(dd.mfct & 0x7fff)` fix also fixed it. Thus, as said earlier - I think something works differently when parsing data from arguments vs while running wmbusmeters as a service and you may want to check this.

weetmuts, 2024-09-14:
Ah, yes the stderr is piped separately. Most of the tests check that the stdout produces the expected json output.

weetmuts, 2024-09-14:
Many thanks for fixing this!

kitor, 2024-09-14:
I guess we may close this issue then :)

closed 2024-09-14 by kitor

weetmuts, 2024-09-14:
Could it be that the amiplus sends one APA which can be decoded, and one APT which cannot be decoded?

Previously you could only specify the id 00058812 . But now you can specify:

00058812.M=APA
and even more specific:
00058812.M=APA.V=04.T=08

So we can filter away the broken telegrams from the same id.

kitor, 2024-09-14:
Ok, since there's some confusion here :)

- My `amiplus` meter indeed sends two telegrams, one with `APA`, one with `APT`; First one works, 2nd is not decoded - I added definition in my PR but left that commented out. Good to know one can filter that in config itself.
- In my earlier message I gave example on `apator162` driver but copied wrong part of the log - indeed those water meters also send two telegrams `APA` and `APT` with same caveat (APT not understood/decoded) - hovewer `driver_apator162` binds to both versions so the warning should never happen seen in that case.

After the fix is merged, the only warning left is the one from `amiplus` with `APT` mfct code, but that was expected, as above :) 
Before the fix each time I started wmbus I had 7 warnings in total, as I listen to 7 devices (4x apatoretin, 2x apator162, 1x amiplus)

weetmuts, 2024-09-14:
Good. Yes, I remember that the APT telegrams were annoying and I think that the print warning only once, was the solution I came up with. Now you can get rid of it by using the id 00058812.M=APA if you like.