wmbusmeters/wmbusmeters #1363: Hex string is not proper! Ignoring telegram!

alientec, 2024-09-14:
### Type of request

Question

### HAOS version

12,4

### wmbusmeters version

1.17.1

### Your message goes here

Hey there,
I have an nanoCUL and I am using the ha-addon for receiving my meter data.

I read that the nanoCUL can receive at max. 148 Bytes per telegram. 
- Is here the problem that the telegram would be longer than that? 
- Why I get at the end the `<0D><0A>` if its no finished? 
- How can I eliminate the CRC Bytes to decode is manually on [http://wmbusmeters.org](url)

Regards,
Alex

Example 1:
```
(serial) received binary "6241433434323433343830333530313233353030373632464137413836303030303230"
(cul) checkCULFrame "bAC44243480350123500762FA7A86000020"
(cul) no eol found yet, partial frame
(serial) received binary "32463246303431333939313830303030303436443246363839443244304533393034464431373030303030303030304537383030303030303030414444"
(cul) checkCULFrame "bAC44243480350123500762FA7A860000202F2F041399180000046D2F689D2D0E3904FD17000000000E7800000000ADD"
(cul) no eol found yet, partial frame
(serial) received binary "3330303030343431333636304430303030343236434646324338343031313330303831314331383030303038323031364331463338443330313342374130313030433430314135454536443143333031353335383130344644323830313832303436433146333838343843413630343133303031383030"
(cul) checkCULFrame "bAC44243480350123500762FA7A860000202F2F041399180000046D2F689D2D0E3904FD17000000000E7800000000ADD300004413660D0000426CFF2C84011300811C18000082016C1F38D3013B7A0100C401A5EE6D1C3015358104FD280182046C1F38848CA60413001800"
(cul) no eol found yet, partial frame
(serial) received binary "30304334303431334537313630303030383430353133434432323534313530303030433430353133414431333030303038343036313334303132373934363030303043343036313341313130303030303834300D0A"
(cul) checkCULFrame "bAC44243480350123500762FA7A860000202F2F041399180000046D2F689D2D0E3904FD17000000000E7800000000ADD300004413660D0000426CFF2C84011300811C18000082016C1F38D3013B7A0100C401A5EE6D1C3015358104FD280182046C1F38848CA6041300180000C40413E7160000840513CD2254150000C40513AD130000840613401279460000C40613A1100000840<0D><0A>"
(cul) checkCULFrame RSSI_RAW=64
(cul) checkCULFrame LQI=4
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/wmbusmeters/wmbusmeters/issues/390
(cul)         and: https://wmbusmeters.github.io/wmbusmeters-wiki/nanoCUL.html
(cul) bad hex "AC44243480350123500762FA7A860000202F2F041399180000046D2F689D2D0E3904FD17000000000E7800000000ADD300004413660D0000426CFF2C84011300811C18000082016C1F38D3013B7A0100C401A5EE6D1C3015358104FD280182046C1F38848CA6041300180000C40413E7160000840513CD2254150000C40513AD130000840613401279460000C40613A110000"
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) error in received message.
```

Example 2:
```
(serial) received binary "6241433434323433343133343532343232353030373944364237413936303030303230324632463034313345324239303030303034364433"
(cul) checkCULFrame "bAC4424341345242250079D6B7A960000202F2F0413E2B90000046D3"
(cul) no eol found yet, partial frame
(serial) received binary "34303945323244304533393034464431373030303030303030304537383030303030303030414444333030303034343133454535433030303034323643"
(cul) checkCULFrame "bAC4424341345242250079D6B7A960000202F2F0413E2B90000046D3409E22D0E3904FD17000000000E7800000000ADD300004413EE5C0000426C"
(cul) no eol found yet, partial frame
(serial) received binary "464632433834303131333942303346454235303030303832303136433146333844333031334238453032303043343031363241453644304332423035333538313034464432383031383230343643314633383834463431323034313339424235303030304334303431333031423130303030383430353133"
(cul) checkCULFrame "bAC4424341345242250079D6B7A960000202F2F0413E2B90000046D3409E22D0E3904FD17000000000E7800000000ADD300004413EE5C0000426CFF2C8401139B03FEB5000082016C1F38D3013B8E0200C40162AE6D0C2B05358104FD280182046C1F3884F41204139BB50000C4041301B10000840513"
(cul) no eol found yet, partial frame
(serial) received binary "30384641324341423030303043343035313344413944303030303834303631334439393039433237303030304334303631333045383330303030383430"
(cul) checkCULFrame "bAC4424341345242250079D6B7A960000202F2F0413E2B90000046D3409E22D0E3904FD17000000000E7800000000ADD300004413EE5C0000426CFF2C8401139B03FEB5000082016C1F38D3013B8E0200C40162AE6D0C2B05358104FD280182046C1F3884F41204139BB50000C4041301B1000084051308FA2CAB0000C40513DA9D0000840613D9909C270000C406130E830000840"
(cul) no eol found yet, partial frame
(serial) received binary "0D0A"
(cul) checkCULFrame "bAC4424341345242250079D6B7A960000202F2F0413E2B90000046D3409E22D0E3904FD17000000000E7800000000ADD300004413EE5C0000426CFF2C8401139B03FEB5000082016C1F38D3013B8E0200C40162AE6D0C2B05358104FD280182046C1F3884F41204139BB50000C4041301B1000084051308FA2CAB0000C40513DA9D0000840613D9909C270000C406130E830000840<0D><0A>"
(cul) checkCULFrame RSSI_RAW=64
(cul) checkCULFrame LQI=4
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/wmbusmeters/wmbusmeters/issues/390
(cul)         and: https://wmbusmeters.github.io/wmbusmeters-wiki/nanoCUL.html
(cul) bad hex "AC4424341345242250079D6B7A960000202F2F0413E2B90000046D3409E22D0E3904FD17000000000E7800000000ADD300004413EE5C0000426CFF2C8401139B03FEB5000082016C1F38D3013B8E0200C40162AE6D0C2B05358104FD280182046C1F3884F41204139BB50000C4041301B1000084051308FA2CAB0000C40513DA9D0000840613D9909C270000C406130E83000"
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) error in received message.
```

alientec, 2024-09-14:
I tested now some time and wrote on my own a little script to delete the CRC Bytes.

Debug output:
`(cul) bad hex "AC4424341345242250079D6B7A960000202F2F0413E2B90000046D3409E22D0E3904FD17000000000E7800000000ADD300004413EE5C0000426CFF2C8401139B03FEB5000082016C1F38D3013B8E0200C40162AE6D0C2B05358104FD280182046C1F3884F41204139BB50000C4041301B1000084051308FA2CAB0000C40513DA9D0000840613D9909C270000C406130E83000"`

Converted without CRC Bytes:
`AC4424341345242250077A960000202F2F0413E2B90000046D342D0E3904FD17000000000E780000000000004413EE5C0000426CFF2C8401139BB5000082016C1F38D3013B8E0200C4016D0C2B05358104FD280182046C1F388404139BB50000C4041301B100008405132CAB0000C40513DA9D0000840613D9900000C406130E83000`

Added a 0 to the end to get a even lenght:
`AC4424341345242250077A960000202F2F0413E2B90000046D342D0E3904FD17000000000E780000000000004413EE5C0000426CFF2C8401139BB5000082016C1F38D3013B8E0200C4016D0C2B05358104FD280182046C1F388404139BB50000C4041301B100008405132CAB0000C40513DA9D0000840613D9900000C406130E830000`

Output from [https://wmbusmeters.org](wmbusmeters.org):
```
{
    "media":"water",
    "meter":"evo868",
    "name":"",
    "id":"23013580",
    "consumption_at_history_1_m3":6.144,
    "consumption_at_history_2_m3":5.863,
    "consumption_at_history_3_m3":5.46,
    "consumption_at_history_4_m3":5.037,
    "consumption_at_history_5_m3":4.672,
    "consumption_at_history_6_m3":4.257,
    "consumption_at_set_date_m3":3.43,
    "consumption_at_set_date_2_m3":6.144,
    "history_1_date":"2024-08-31",
    "history_2_date":"2024-07-31",
    "history_3_date":"2024-06-30",
    "history_4_date":"2024-05-31",
    "history_5_date":"2024-04-30",
    "history_6_date":"2024-03-31",
    "history_reference_date":"2024-08-31",
    "max_flow_since_datetime_m3h":0.378,
    "total_m3":6.297,
    "current_status":"OK",
    "device_date_time":"2024-09-14 13:47",
    "fabrication_no":"000000000000",
    "max_flow_datetime":"2024-05-21 16:28",
    "set_date":"2023-12-31",
    "set_date_2":"2024-08-31",
    "timestamp":"2024-09-14T14:08:18Z"
}
```

weetmuts, 2024-09-14:
The log says it all. The nano-CUL dongle cannot receive telegrams over a certain length.  There is nothing to fix except buying another dongle. https://github.com/wmbusmeters/wmbusmeters-wiki/blob/master/nanoCUL.md

closed 2024-09-14 by alientec