wmbusmeters/wmbusmeters #1364: 32 bit datetime 1 hour offset

micw, 2024-09-17:
### Type of request

Bug

### OS version

linux

### wmbusmeters version

latest(online)

### Your message goes here

Hello,

I had cross-checked one of my decodings on the online decode and think there is an issue with decoding of 32 bit date time which have an 1-hour-offset.

This is the oms field:

```
036   : 04 dif (32 Bit Integer/Binary Instantaneous value)
037   : 6D vif (Date and time type)
038 C!: 3AA81139 ("meter_datetime":"2024-09-17 08:58")
```

I'm not 100% sure because different online decoders gives me a time of 08:58, 09:58 or 10:58 (lobaro decoder).

But the 2nd Byte (0xA3) has bit 7 set (so the 15th bit of the whole filed, starting with zero) and the Spec (EN13757-3-2018, page 37: Date+Time Type F, CP32) says:

```
SU B1 [15]
SU < 0 > : Normalzeit ;
SU < 1 > : Sommerzeit
```

"Sommerzeit" is german for "Daylight saving time": https://en.wikipedia.org/wiki/Daylight_saving_time

So this should be in "Summer time" which is IMO normal time +1h

weetmuts, 2024-09-17:
You are quite right. I have not implemented this, can you post a full telegram?

The code that needs improvement is src/dvparser.cc --- bool DVEntry::extractDate(struct tm *out) { ... }
where the comment says: 
        // There are also bits for day of week, week of year.                                                          
        // A bit for if daylight saving is in use or not and its offset.                                               
        // A bit if it is a leap year.                                                                                 
        // I am unsure how to deal with this here..... TODO

micw, 2024-09-17:
> can you post a full telegram?

Unfortunately not, it's not my own device and I cannot share the telegram.  But the initial post should contain enough relevant fragment to add a unit test for this particular case.

kzajac83, 2024-09-17:
Similar functionality have this overlay  https://github.com/wmbusmeters/wmbusmeters/issues/1210  example telegram:

`telegram=|68C4C46808637283654616475F031672000000_041300000000046D17101139426C992B44133D00000042EC7EB92B0C785506140082016C9E2B8401133D000000C2016C7F2CC401133D00000082026C9F218402133D000000C2026C9D22C402133D00000082036C9F238403133D000000C2036C9E24C403133D00000082046C9F258404133D000000C2046C9E26C404133D00000082056C9F278405133D000000C2056C9F28C405133D00000082066C9E298406133D000000C2066C9F2AC406133D0000000F010000_6C16|+4
`

`    "meter_datetime":"2024-09-17 16:23",`

(timestamp is wrong, cannot setup properly hours and minuses, but no impact for daily light saving option).

M-Tool for this overlay show this options:

[image]

kzajac83, 2024-09-17:
When setting current timestamp I got tomorrow date and some minsters after midnight however for DLS should be no problem:

[image]

`68C4C46808637283654616475F03167C000000041300000000046D24001239426C992B44133D00000042EC7EB92B0C785506140082016C9E2B8401133D000000C2016C7F2CC401133D00000082026C9F218402133D000000C2026C9D22C402133D00000082036C9F238403133D000000C2036C9E24C403133D00000082046C9F258404133D000000C2046C9E26C404133D00000082056C9F278405133D000000C2056C9F28C405133D00000082066C9E298406133D000000C2066C9F2AC406133D0000000F0100007416`

[image]

`68C4C46808637283654616475F03167F000000041300000000046D27001239426C992B44133D00000042EC7EB92B0C785506140082016C9E2B8401133D000000C2016C7F2CC401133D00000082026C9F218402133D000000C2026C9D22C402133D00000082036C9F238403133D000000C2036C9E24C403133D00000082046C9F258404133D000000C2046C9E26C404133D00000082056C9F278405133D000000C2056C9F28C405133D00000082066C9E298406133D000000C2066C9F2AC406133D0000000F0100007A16`

weetmuts, 2024-12-17:
Anyone cares to help out? 

My old comment:

The code that needs improvement is src/dvparser.cc --- bool DVEntry::extractDate(struct tm *out) { ... }
where the comment says:
// There are also bits for day of week, week of year.
// A bit for if daylight saving is in use or not and its offset.
// A bit if it is a leap year.
// I am unsure how to deal with this here..... TODO