wmbusmeters/wmbusmeters #1370: USB Master Device still supported

manalishi70, 2024-09-20:
### Type of request

Question

### OS version

Ubuntu 22.04

### wmbusmeters version

1.17.1

### Your message goes here

I was using this device [https://de.aliexpress.com/item/1005004540180026.html?spm=a2g0o.detail.1000023.5.5c37cALccALcQ9&gatewayAdapt=glo2deu#nav-specification](url) with an older wmbusmeters version on a raspberry. 
Now I connected it to a Ubuntu 22.04 x86 machine and compiled the latest version of wmbusmeters.

lsusb: Bus 001 Device 015: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC

 wmbusmeters --debug auto:c1,t1:

(main) ignoring /dev/ttyUSB0, it does not respond as any of the supported wmbus devices.
(serial) EVENT thread interrupted
No wmbus device detected, waiting for a device to be plugged in.

Is this device still supported? I'm looking for a solution for a while now, but no luck.
[image]

weetmuts, 2024-09-21:
The USB Master Device is just a serial port. It will not respond to anything. :-) So it cannot be detected.

Just check the README and you will find the correct configuration commandline to use mbus. E.g.

wmbusmeters --pollinterval=60s MAIN=/dev/ttyUSB0:mbus:2400 MyTempMeter piigth:MAIN:mbus 12001932 NOKEY

manalishi70, 2024-09-21:
Thank you @weetmuts. Shame on me. I didn't do any documentation that time and the Raspberry SDCart is broken.
I will read more carefully.

weetmuts, 2024-09-30:
No worries! Good luck!

closed 2024-09-30 by weetmuts