wmbusmeters/wmbusmeters #1372: [M-Bus] Analyze & pollinterval = non-stop reading

kzajac83, 2024-09-22:
### Type of request

Bug

### OS version

Current Debian

### wmbusmeters version

Current

### Your message goes here

Hi

When using parallel flags `analyze `and `pollinterval` for example like below command:

`root@wyse:~# wmbusmeters --analyze pollinterval=600s ----format=json --ppjson  /dev/ttyUSB0:mbus:2400 meter kamheat:mbus p203 NOKEY`

where pool is set 600s (10 min) then response is without delay, if you not break command then non-stop reading and analyzing telegrams, only between each end of processing telegram and giving JSON format data output and start next attempt appear below msg 

`meter p203 did not send a response!`

but this happened for all my wired M-Bus devices from different vendors hence in my opinion this is a bug.

Also when using `analyze `flag why is mandatory `pollinterval` too?  When you analyze data then focusing on one telegram longer time (sometime quick, sometime longer) due to need remember put ctrl+c after each analyze attempt. Can be this changed?tTis mean `poolinterval `is optional when `analyze `is used. 

PS. When analyzing probably wmbusmeters trying assign all drivers to analyzed telegram.  Not found but there is some switch that disable checking other driver except explicitly indicated in the command line? I see lot of the same telegram when used flag `logtelegrams `

BR, Krzysiek

weetmuts, 2024-09-22:
I am not sure --analyze is supposed to work during actual collection.

It was meant to be applied to already collected telegrams using for example --logtelegrams
:-)

Analyze is pretty costly, it will apply all drivers to your telegram to find the best match.

weetmuts, 2024-09-22:
You use it from the command line where the telegram hex is an argument:
wmbusmeters --analyze 68101068........

kzajac83, 2024-09-22:
Understood idea, I was not familiar with the concept using `--analyze` :)

Seems I using not as design (not a bug, a feature) however in my opinion the method of analyzing is quicker because you analyze live data and not waste time for storing telegrams, just trying the driver on actual data.

kzajac83, 2024-11-22:
To clarify it is bug or feature ? If second then please close the issue.

closed 2025-02-23 by github-actions[bot]