wmbusmeters/wmbusmeters #1373: Question to "Home Assistant Community Add-on: wmbusmeters (W-MBus to MQTT)"

thoath2, 2024-09-26:
### Type of request

Question

### HAOS version

2024.9

### wmbusmeters version

---

### Your message goes here

Hello,
i plan to use the HACS-Add-on wmbusmeters (W-MBus to MQTT).
I Have two question to the add-on. 

1. In the forst paragraph of documentation i read "This add-on allows you to acquire utility meter readings without the vendors bridge or gateway as long as they support C1, T1 or S1 telegrams using the wireless mbus protocol (WMBUS).". Do I understand it right, that the add-on is only for radio-m-bus-Devices? I want to use a serial (USB)-Device. Ist it possible?

2. In the documentation i read that the add-on processes the incoming telegrams. But my devices do not send telegrams independently. The devices must be requested to periodically send back a telegram. Is the add-on able to actively query telegrams? Is the add-on able to take over the time control?

I look forward for answers.

regards Thomas

gotsila, 2024-09-27:
Hello,
the device is the Receiver. in my case its the im871a. this is an usb stick. the sender of the telegramm is my e.g. watermeter. this sends periodically telegramms.
this would be received from the im871a. and send it via the usb into to Add-On. the Add-On to MQTT and then to HA.

so the device (im871a) is almost a receiver. this can't sending i believe, and my watermeter can't receive.

so would say: yes,  is only for radio-m-bus-Devices

BIBOLV, 2024-10-06:
Hi,
Yes, USB-to-MBUS is also supported - check documentation, other issues/discussions, like [this](https://github.com/orgs/wmbusmeters/discussions/842).

closed 2025-01-06 by github-actions[bot]