wmbusmeters/wmbusmeters #1380: Driver not compiled / created pollucomf.json

faulesocke85, 2024-10-04:
### Type of request

Bug

### HAOS version

13.1

### wmbusmeters version

1.17.1

### Your message goes here

Hi there, as I read in the documentation there is a driver for pollucom heat meters. It seems the needed driver ist not created even there seems to be a source file. My log shows this: 

[23:51:49] INFO:   File /config/wmbusmeters/etc/mqtt_discovery/pollucomf.json not found. 

And the folder contains only that files: 
[image]

I am using mbus not wmbus. 

There is another user with the problem found something in mqttdiscover.sh 
https://github.com/wmbusmeters/wmbusmeters/issues/711

I am not able to implement this. Can you please check this? 

BR

BIBOLV, 2024-10-06:
Hi,
You can check driver definitions that has been added [here](https://github.com/wmbusmeters/wmbusmeters-ha-addon/tree/main/wmbusmeters-ha-addon-edge/mqtt_discovery). You can create [PR](https://github.com/wmbusmeters/wmbusmeters-ha-addon/blob/main/CONTRIBUTING.md) for driver you are interested in or describe them manually in HA sensors.

faulesocke85, 2024-10-07:
Solved

closed 2024-10-07 by faulesocke85