wmbusmeters/wmbusmeters #1395: Support for Sensus Metering Systems water meter

jm314159, 2024-10-15:
### Type of request

New device support

### Meter description

Please add support for Sensus Metering Systems watermeter (PulseRF, CompactRF). The following is the log from ESPHome:
```
[16:21:15][D][mbus:035]: Received T1 A frame
[16:21:15][V][mbus:046]: Frame: 39671C9B273459A38D5995962EC2D38D63724CE35A4CD34D3659B27345A35936B459659659659C34B8CD69AA69A4B58E59658E2E35965962E5C9AA (59) [RAW]
[16:21:15][VV][3of6:095]: Decode 3 out of 6 OK.
[16:21:15][V][mbus:053]: Frame: 2044AE4C062105003837B01E7216711119AE4C0B076C0000000413B166FFF30200023B000039E6 (39) [with CRC]
[16:21:15][V][mbus:096]: Validating CRC for Block1
[16:21:15][V][crc:031]:     calculated: 0xB01E, read: 0xB01E
[16:21:15][V][mbus:116]: Validating CRC for Block2
[16:21:15][V][crc:031]:     calculated: 0x66FF, read: 0x66FF
[16:21:15][V][mbus:116]: Validating CRC for Block3
[16:21:15][V][crc:031]:     calculated: 0x39E6, read: 0x39E6
[16:21:15][V][mbus:063]: Frame: 2044AE4C0621050038377216711119AE4C0B076C0000000413B1F30200023B0000 (33) [without CRC]
[16:21:15][VV][wmbus:056]: Have data from RF ...
[16:21:15][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 33
[16:21:15][V][Telegram.cpp:1178]: (wmbus) parseELL @10 23
[16:21:15][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 23
[16:21:15][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 23
[16:21:15][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 23
[16:21:15][V][dvparser.cpp:1471]: [DVPARSER] entry 25: 0413 Instantaneous vif=13  st=0 ta=0 su=0
[16:21:15][V][dvparser.cpp:1471]: [DVPARSER] entry 31: 023B Instantaneous vif=3b  st=0 ta=0 su=0
[16:21:15][V][dvparser.cpp:1493]: (dvparser) found new format "0413023B" with hash c944, remembering!
[16:21:15][I][wmbus:092]: Using selected driver iperl (detected driver was )
[16:21:15][I][wmbus:104]: iperl [0x00052106] RSSI: -86dBm T: 2044AE4C0621050038377216711119AE4C0B076C0000000413B1F30200023B0000 (33) T1 A
[16:21:15][D][meters.cpp:1993]: (meter) created ESPHome iperl 00052106 not-encrypted
[16:21:15][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 33
[16:21:15][V][Telegram.cpp:1178]: (wmbus) parseELL @10 23
[16:21:16][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 23
[16:21:16][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 23
[16:21:16][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 23
[16:21:16][V][dvparser.cpp:1471]: [DVPARSER] entry 25: 0413 Instantaneous vif=13  st=0 ta=0 su=0
[16:21:16][V][dvparser.cpp:1471]: [DVPARSER] entry 31: 023B Instantaneous vif=3b  st=0 ta=0 su=0
[16:21:16][V][meters.cpp:635]: (meter) ESPHome: for me? 00052106.M=SEN.V=38.T=37,19117116.M=SEN.V=0b.T=07,00052106.M=SEN.V=38.T=37,19117116.M=SEN.V=0b.T=07 in 00052106
[16:21:16][W][meters.cpp:688]: (meter) ESPHome: meter detection did not match the selected driver iperl! correct driver is: unknown!
(meter) Not printing this warning again for id: 00052106 mfct: (SEN) Sensus Metering Systems, Germany (0x4cae) type: Radio converter (meter side) (0x37) ver: 0x38
[16:21:16][W][meters.cpp:692]: (meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
[16:21:16][W][meters.cpp:693]: (meter) to add support for this unknown mfct,media,version combination
[16:21:16][V][meters.cpp:699]: (meter) ESPHome: yes for me
[16:21:16][D][meters.cpp:922]: (meter) ESPHome(0) iperl  handling telegram from 19117116.M=SEN.V=0b.T=07
[16:21:16][V][Telegram.cpp:1131]: (wmbus) parseDLL @0 33
[16:21:16][V][Telegram.cpp:545]: (telegram) DLL L=20 C=44 (from meter SND_NR) M=4cae (SEN) A=00052106 VER=38 TYPE=37 (Radio converter (meter side)) (driver unknown!) DEV=ESPHome wM-Bus RSSI=-86
[16:21:16][V][Telegram.cpp:1178]: (wmbus) parseELL @10 23
[16:21:16][V][Telegram.cpp:1333]: (wmbus) parseNWL @10 23
[16:21:16][V][Telegram.cpp:1391]: (wmbus) parseAFL @10 23
[16:21:16][V][Telegram.cpp:2078]: (wmbus) parseTPL @10 23
[16:21:16][V][dvparser.cpp:1471]: [DVPARSER] entry 25: 0413 Instantaneous vif=13  st=0 ta=0 su=0
[16:21:16][V][dvparser.cpp:1471]: [DVPARSER] entry 31: 023B Instantaneous vif=3b  st=0 ta=0 su=0
[16:21:16][V][Telegram.cpp:634]: (telegram) TPL CI=72
[16:21:16][V][Telegram.cpp:640]:  ACC=6c STS=00 CFG=0000 ()
[16:21:16][V][Telegram.cpp:649]:  ID=16711119 MFT=ae4c VER=0b TYPE=07 (Water meter)
[16:21:16][V][meters.cpp:986]: (meters) trying field info total(Volume)[0]...
[16:21:16][V][meters.cpp:1008]: (meters) using field info total(Volume)[0] to extract 0413 at offset 25
[16:21:16][V][meters.cpp:2168]: (meter) Volume total_m3 decoded m3 default m3 value 193.457 (scale 1)
[16:21:16][V][meters.cpp:986]: (meters) trying field info max_flow(Flow)[1]...
[16:21:16][V][meters.cpp:1008]: (meters) using field info max_flow(Flow)[1] to extract 023B at offset 31
[16:21:16][V][meters.cpp:2168]: (meter) VolumeFlow max_flow_m3h decoded m3h default m3h value 0 (scale 1)
```

A few existing drivers, e.g. iperl, hydrus, itron, can decode the total volume as well as topaseskr which is proposed on https://wmbusmeters.org/analyze, however, ESPHome discovers a mismatch.

### Logged telegrams from meter in decrypted format

```shell
2044AE4C0621050038377216711119AE4C0B077B0000000413F7BA0200023B0000
2044AE4C0621050038377216711119AE4C0B077C0000000413FEBA0200023B0000
2044AE4C0621050038377216711119AE4C0B07880000000413FFBA0200023B0000
2044AE4C0621050038377216711119AE4C0B0789000000041303BB0200023B0000
2044AE4C0621050038377216711119AE4C0B0797000000041307BB0200023B0000
2044AE4C0621050038377216711119AE4C0B0798000000041323BB0200023B2A00
2044AE4C0621050038377216711119AE4C0B0799000000041340BB0200023B0000
2044AE4C0621050038377216711119AE4C0B07AC000000041345BB0200023B0000
2044AE4C0621050038377216711119AE4C0B076D0000000413B1F30200023B0000
2044AE4C0621050038377216711119AE4C0B076E0000000413BAF30200023B0000
2044AE4C0621050038377216711119AE4C0B076F0000000413BDF30200023B0000
```

jm314159, 2024-10-17:
No, I received the sensor without any key from the water company.

rege245, 2024-10-17:
thats strange because in manual for PulseRF is noted that if no key is set, default key will be used :(

jm314159, 2024-10-18:
The previous transmitter had the key set. Though the company has been using bi-directional protocols SensusRF and/or FlexNet, I was informed that the WMBus key cannot be shared because it is the same for all managed meters. Instead, they offered me opening the WMBus protocol and I asked for that upon the water meter exchange.

rege245, 2024-10-18:
ok, i ask supplier for code, because it deppends on serial number of radio. i got it, but cant decode message ... i need to look at it deeply

BIBOLV, 2024-12-29:
Can you provide more details about your setup? As you mentioned - lot of drivers are capable of decoding your meter message, so can you check that you are running latest wmbusmters version?

jm314159, 2024-12-30:
The log I provided comes from HA > ESPHome. The following is the related setup to read that water meter:
```
time:
  - platform: sntp
    id: sntp_time

esphome:
  name: wmbus-esp32
  friendly_name: wmbus-esp32
  comment: "ESP32 & CC1101"

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

external_components:
  - source: github://SzczepanLeon/esphome-components@main
    refresh: 0d
    components: [ wmbus ]

logger:
  level: VERY_VERBOSE

#======================================
# Water Meter WMBus Section
#======================================
wmbus:
  mosi_pin: GPIO23
  clk_pin:  GPIO18
  miso_pin: GPIO19
  gdo2_pin: GPIO4
  gdo0_pin: GPIO0
  cs_pin:   GPIO5
  
  led_pin:
    number: GPIO2
    inverted: false
  led_blink_time: "250ms"
  
  #frequency: 868.950  #default value, MHz
  frequency: 868.920
  all_drivers: false
  sync_mode:   false
  log_all:     false

#======================================
sensor:
#======================================

  #======================================
  # Water Meter Sensors
  #======================================
  # add sensor with defined type (driver will be also added to compile list)
  - platform: wmbus
    meter_id: 0x00052106
    type: topaseskr
    #type: iperl
    key: ""
    sensors:
      - name:  "Main wM"
        field: "total"
        accuracy_decimals: 3
        unit_of_measurement: "m³"
        device_class: "water"
        state_class:  "total_increasing"
        icon:         "mdi:water"
      - name:  "Main wM RSSI"
        field: "rssi"
        accuracy_decimals: 0
[... 104 lines cut ...]
021   : 0000 tpl-cfg 0000 ( )
023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
024   : 13 vif (Volume l)
025 C!: BB5B0300 ("total_m3":220.091)
029   : 02 dif (16 Bit Integer/Binary Instantaneous value)
030   : 3B vif (Volume flow l/h)
031 C!: 0000 ("current_flow_m3h":0)

{
    "_":"telegram",
    "media":"water",
    "meter":"topaseskr",
    "name":"",
    "id":"19117116",
    "current_flow_m3h":0,
    "total_m3":220.091,
    "timestamp":"2024-12-30T09:51:10Z"
}

Using: wmbusmeters: 1.17.1-70-g41a2b48
41a2b4808f00367af7bbaac57e221539825babbb
```
on `https://wmbusmeters.org` site.

Actually, I am not sure if that is the information you expected. Please let me know.

BIBOLV, 2024-12-30:
Do you know how/where wmbusmeters is installed in your setup? Can you check that it is running latest version? From info your provided, this looks like some custom setup, so I have no knowledge about it.

jm314159, 2024-12-31:
That's the [component to ESPHome](https://github.com/SzczepanLeon/esphome-components) that is used to receive and process wMBus frame with the use of wmbusmeters drivers. Please see a bottom note in there: "Supported drivers: almost all from wmbusmeters version 1.17.1(without drivers from [file](https://github.com/wmbusmeters/wmbusmeters/blob/1.17.1/src/generated_database.cc)).

There is also a [discussion thread](https://forum.arturhome.pl/t/komponent-wm-bus-do-esphome-szczepanleon-wersja-4-x-watek-ogolny/), though in Polish. You can find a piece of news in there regarding that solution, e.g. all drivers form wmbusmeters 1.17.1 are supported. And, for all meters to be configured, users are encouraged to test the frame/telegram and/or the key, if exists, on the wmbusmeters.org site.

The logs I provided from both the ESPHome and wmbusmeters directly, are consistent in the result that the driver is unknown for my meter. Therefore I followed the suggestion of using topaseskr which mapping looks similar to other drivers in terms of the fundamental measure of water volume, however, it does not match the hardware.

BIBOLV, 2024-12-31:
Yeah, this is custom implementation of wmbusmeters parts/drivers, so I will not be able to help in any way. What I can see in logs that values are decoded - `[10:30:59][V][meters.cpp:2160]: (meter) Volume total_m3 decoded m3 default m3 value 220.091 (scale 1)`. I would suggest to ask this question to developer of tool you are using.

jm314159, 2024-12-31:
If you could please be a little more specific on the following: is it completely impossible to add a new wmbusmeters driver for Sensus based on my readings and logs, or, it is just not you who could do this.

I bet the developer's aim was to preserve wmbusmeter's primacy of the source support of different meters. That's why a user like me is encouraged with the following message in the component log: `please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/ to add support for this unknown mfct,media,version combination`. Luckily, there is still an alternative driver for my meter.

Having the opportunity, I wish you happy New Year!

BIBOLV, 2024-12-31:
As you mentioned [data](https://wmbusmeters.org/analyze/2044AE4C0621050038377216711119AE4C0B077B0000000413F7BA0200023B0000) are extracted by wmbusmeters, so if it is not in tool you are using, you have to approach someone who knows it and can help you. If you are referring to message about unsupported meter - yes, it is shown once for meters that are not recognized, but if data is extracted and are correct, it can be ignored.

closed 2025-04-01 by github-actions[bot]