wmbusmeters/wmbusmeters #1399: Is the iU891A-XL the same as iu891a?

adrai, 2024-10-23:
### Type of request

Question

### OS version

Linux vmubuntu 6.8.0-47-generic #47-Ubuntu SMP PREEMPT_DYNAMIC Fri Sep 27 22:03:50 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

### wmbusmeters version

directly built from master

### Your message goes here

Hi there,

Sorry if this sounds a bit naive, but I'm very new to wmbusmeters...

I have this device: https://wireless-solutions.de/products/iu891a-xl-wmbus/
I assume it's code is: iu891a

Shouldn't I be able to see something when I call: `wmbusmeters auto:t1` ?
I just get: 
```
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!
```

I also tested this directly on my Home Assistant Green, but the result is the same.

Is there something I miss?

btw: This seems to be the mounted device: `/dev/ttyACM0`

adrai, 2024-10-25:
Yes it's iu891a

In HA it does not work, since the latest master was not yet released

closed 2024-10-25 by adrai

ExpressPet, 2024-11-10:
Hi!
I have the same device. Stable version does not work, but EDGE works fine for this device with _device = auto:c1,t1_ in UI. I was not patient enough to set a specific chip, but auto is fast.
comparing to amb8465 from wurth without antenna, much better reach.
Having 100+ sensors, I also went directly to addon config and updated options_custom.json with sensors' details - to do it via UI is fine for a few, but unmanageable for a fleet.

rpdelaat, 2024-11-27:
> Hi! I have the same device. Stable version does not work, but EDGE works fine for this device with _device = auto:c1,t1_ in UI. I was not patient enough to set a specific chip, but auto is fast. [...]

I have the same problem - am new to this all, so please bear with me: 

I get this: [00:30:15] INFO: CONFIG_CONF ...
[00:30:15] INFO: {"loglevel":"normal","device":"auto:t1","donotprobe":"/dev/ttyAMA0","logtelegrams":"false","format":"json","logfile":"/dev/stdout","shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[00:30:15] INFO: CONFIG_METERS ...
[00:30:15] INFO: []
[00:30:15] INFO: Syncing wmbusmeters configuration ...
[00:30:15] INFO: Registering meters ...
[00:30:15] INFO: Generating MQTT configuration ... 
[00:30:15] INFO: Broker core-mosquitto will be used.
[00:30:15] INFO: MQTT Discovery cleanup...
[00:30:16] INFO: Starting web configuration service.
[00:30:16] INFO: Running wmbusmeters ...
[00:30:17] INFO: Starting Nginx...
2024/11/27 00:30:17 [notice] 83#83: using the "epoll" event method
2024/11/27 00:30:17 [notice] 83#83: nginx/1.24.0
2024/11/27 00:30:17 [notice] 83#83: OS: Linux 6.6.31-haos-raspi
2024/11/27 00:30:17 [notice] 83#83: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/11/27 00:30:17 [notice] 83#83: start worker processes
2024/11/27 00:30:17 [notice] 83#83: start worker process 206
Started auto iu880b[00062dbe] on /dev/ttyACM0 listening on t1
Warning! Desired link modes t1 cannot be set for device /dev/ttyACM0:iu880b[00062dbe]

rpdelaat, 2024-11-27:
> Hi! I have the same device. Stable version does not work, but EDGE works fine for this device with _device = auto:c1,t1_ in UI. I was not patient enough to set a specific chip, but auto is fast. [...]

@ExpressPet  - can you please expand on how to go about making it work with HA?  I am really new in this and clearly out of my depth. Thanks for your consideration.

weetmuts, 2024-11-27:
Hi @ExpressPet  @rpdelaat  @adrai 

I have made a new stable release 1.18.0 so the ha-addon should have updated to include support for the iu891a.

Please test!

rpdelaat, 2024-11-27:
> Hi @ExpressPet @rpdelaat @adrai
> 
> I have made a new stable release 1.18.0 so the ha-addon should have updated to include support for the iu891a.
> 
> Please test!

Thanks! I implemented it today with the EDGE addon; now what happens if I try your solution? 
This is what I see in the log: 
[21:58:37] INFO: CONFIG_METERS ...
[21:58:37] INFO: []
[21:58:37] INFO: Syncing wmbusmeters configuration ...
[21:58:37] INFO: Registering meters ...
[21:58:37] INFO: Generating MQTT configuration ... 
[21:58:37] INFO: Starting Nginx...
2024/11/27 21:58:37 [notice] 92#92: using the "epoll" event method
2024/11/27 21:58:37 [notice] 92#92: nginx/1.24.0
2024/11/27 21:58:37 [notice] 92#92: OS: Linux 6.6.31-haos-raspi
2024/11/27 21:58:37 [notice] 92#92: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/11/27 21:58:37 [notice] 92#92: start worker processes
2024/11/27 21:58:37 [notice] 92#92: start worker process 172
[21:58:38] INFO: Broker core-mosquitto will be used.
[21:58:38] INFO: MQTT Discovery cleanup...
[21:58:39] INFO: Running wmbusmeters ...
No wmbus device detected, waiting for a device to be plugged in.

in the [edge] log I get the reading ok. 
Now when I started the [edge] version, I also had to unplug and plug the usb stick. 
not sure whether that would help making it work - and like I said - not sure we get a huge confusion/conflict when two addons are targeting the same. 

I am happy it works now - for a newbie like me it was a challenge to figure out the process steps to get things working. 

I have one challenge left - The encryption key. I do not get any data yet - not clear whether it is the lack of the right key or the lack of water being used. 

I will keep you posted.

rpdelaat, 2024-11-28:
Update: Restarted today - Still get same message. See below for the [Edge] version - which works, but complains about missing encryption key (I tried a generic key, but it does not work - and I asked the utilty company for the key.)

[17:38:01] INFO: Service restart after closing
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nginx.sh
cont-init: info: /etc/cont-init.d/nginx.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun flask (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
services-up: info: copying legacy longrun wmbusmeters (no readiness notification)
[18:38:07] INFO: Starting web configuration service.
[18:38:07] INFO: Starting wmbusmeters service.
s6-rc: info: service legacy-services successfully started
[18:38:07] INFO: CONFIG_CONF ...
[18:38:07] INFO: {"device":"iu891a[00202063]","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[18:38:07] INFO: CONFIG_METERS ...
[18:38:07] INFO: [{"name":"MyTapWater","driver":"topaseskr","id":"71105929","key":"00112233445566778899AABBCCDDEEFF"}]
[18:38:07] INFO: Syncing wmbusmeters configuration ...
[18:38:07] INFO: Registering meters ...
[18:38:07] INFO: Adding meter-0001 ...
[18:38:07] INFO: Starting Nginx...
2024/11/28 18:38:07 [notice] 92#92: using the "epoll" event method
2024/11/28 18:38:07 [notice] 92#92: nginx/1.24.0
2024/11/28 18:38:07 [notice] 92#92: OS: Linux 6.6.31-haos-raspi
2024/11/28 18:38:07 [notice] 92#92: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/11/28 18:38:07 [notice] 92#92: start worker processes
2024/11/28 18:38:07 [notice] 92#92: start worker process 174
[18:38:07] INFO: Generating MQTT configuration ... 
[18:38:07] INFO: Broker core-mosquitto will be used.
[18:38:08] INFO: 
MQTT Discovery ...
[18:38:08] INFO:  Adding meter: MyTapWater ...
[18:38:08] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/71105929_external_temperature_c/config
[18:38:08] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/71105929_flow_m3h/config
[18:38:08] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/71105929_flow_temperature_c/config
[18:38:08] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/71105929_rssi_dbm/config
[18:38:08] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/71105929_total_m3/config
[18:38:08] INFO: MQTT Discovery cleanup...
[18:38:09] INFO: Running wmbusmeters ...
[ALARM SpecifiedDeviceNotFound] the device iu891a[00202063] is not working
No wmbus device detected, waiting for a device to be plugged in.

===> **Note that it is plugged in - and I even rebooted the Raspberry pie and stuck the stick in another USB port.** 
................................................................................................................................................

**EDGE version:**

[17:45:48] INFO: Starting web configuration service.
[17:45:48] INFO: Starting wmbusmeters service.
s6-rc: info: service legacy-services successfully started
[17:45:48] INFO: CONFIG_CONF ...
[17:45:48] INFO: {"device":"device=auto:t1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[17:45:48] INFO: CONFIG_METERS ...
[17:45:48] INFO: [{"driver":"topaseskr","id":"71105929","key":"","name":"MyTapWater"}]
[17:45:48] INFO: Syncing wmbusmeters configuration ...
[17:45:48] INFO: Starting Nginx...
2024/11/28 17:45:48 [notice] 92#92: using the "epoll" event method
2024/11/28 17:45:48 [notice] 92#92: nginx/1.24.0
[... 2 lines cut ...]
2024/11/28 17:45:48 [notice] 92#92: start worker processes
2024/11/28 17:45:48 [notice] 92#92: start worker process 164
[17:45:48] INFO: Registering meters ...
[17:45:49] INFO: Adding meter-0001 ...
[17:45:49] INFO: Generating MQTT configuration ... 
[17:45:49] INFO: Broker core-mosquitto will be used.
[17:45:49] INFO: 
MQTT Discovery ...
[17:45:49] INFO:  Adding meter: MyTapWater ...
[17:45:49] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/71105929_external_temperature_c/config
[17:45:50] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/71105929_flow_m3h/config
[17:45:50] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/71105929_flow_temperature_c/config
[17:45:50] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/71105929_rssi_dbm/config
[17:45:50] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/71105929_total_m3/config
[17:45:50] INFO: MQTT Discovery cleanup...
[17:45:51] INFO: Running wmbusmeters ...
Started auto iu891a[00202063] on /dev/ttyACM1 listening on t1
[... cut ...]
172.30.32.2 - - [28/Nov/2024:18:42:22 +0100] "GET /wmbusmeters.png HTTP/1.1" 200 55338 "http://homeassistant.local:8123/api/hassio_ingress/VpYl0h5rM5azvBaBWm_FZ0tUHTM-8kbjWKtIT-avR2k/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
172.30.32.2 - - [28/Nov/2024:18:42:22 +0100] "GET /get_json HTTP/1.1" 200 394 "http://homeassistant.local:8123/api/hassio_ingress/VpYl0h5rM5azvBaBWm_FZ0tUHTM-8kbjWKtIT-avR2k/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"

rpdelaat, 2024-11-28:
Hi Fredrik,

I tested it - does not recognise the device - EDGE version works (apart
from key missing). see my notes on Github.
kind regards,
Richard