wmbusmeters/wmbusmeters #1409: Compatible with Arduino IDE ?

vincegre, 2024-10-30:
### Type of request

Question

### OS version

Arduino

### wmbusmeters version

not relevant

### Your message goes here

Hi

Is it possible to compile wmbusmeters on Arduino or not at all ? I'd like to use it for a embedded device.

Thanks

Vincèn

kzajac83, 2024-10-30:
Not sure Arduino but there is for esp32 + CC1101 (I using it)

https://github.com/SzczepanLeon/esphome-components

Depending of ESP board you can connect to ESP by PoE Ethernet, Ethernet, WiFi, BT, serial via USB C, 3.3V or TTL 5V.   You can easily integrate it with HA or read telegrams from the device and make what you wish.

vincegre, 2024-10-30:
Thanks a lot @kzajac83  for your answer ;) unhappy in meantime I realised my energy supplier activates only mbus wired and not mbus wireless on electricity counter. Now I have to find an Mbus wired interface auto-powered by Mbus which so far doesn't look to exist :(
I'll close the ticket as it's no more an issue with wmbusmeters ;)

closed 2024-10-30 by vincegre

kzajac83, 2024-10-30:
I have also from AliExpress ~15$ M-Bus wire master device for more than 30 slaves, output is usb, no need extra power supply. Working fine under Windows and Linux.

Have also industrial for ~90$ with full RS232 interface but I thing you are not interested.

vincegre, 2024-10-30:
> I have also from AliExpress ~15$ M-Bus wire master device for more than 30 slaves, output is usb, no need extra power supply. Working fine under Windows and Linux.

Yeah but USB won't make it as I need to connect in URAT (serial) to my LoRa board to transmit datas remotely (counters are in place without electricity and underground so it's the only thing that will make it). Do you know similar interfaces but with UART port ? could not find any on Aliexpress :)

> Have also industrial for ~90$ with full RS232 interface but I thing you are not interested.

Well I might be interested if they are auto powered by mbus ;)

Thanks

kzajac83, 2024-10-30:
https://control.com.pl/en/product-category/m-bus-485-eth-converters/

I have M-Bus 10+ (in real up to 15 slaves can connect) 
https://control.com.pl/en/product/m-bus-10/

vincegre, 2024-10-30:
> https://control.com.pl/en/product-category/m-bus-485-eth-converters/
> I have M-Bus 10+ (in real up to 15 slaves can connect) https://control.com.pl/en/product/m-bus-10/

@kzajac83  Thanks a lot, it's exactly what I needed. Going to order one straight now :-)