wmbusmeters/wmbusmeters #1415: wmbusmeters does not compile on FreeBSD due to libxml2

nefh, 2024-11-10:
### Type of request

Bug

### OS version

FreeBSD 14.1

### wmbusmeters version

1.17.1

### Your message goes here

Hi!
In wmbusmeters Makefile there is hardcoded filepath for libxml2:

```
$(BUILD)/%.o: src/%.c $(wildcard src/%.h)
	$(CXX) -I/usr/include/libxml2 $(CXXFLAGS) $< -c -E > $@.src
	$(CXX) -I/usr/include/libxml2 -fpermissive $(CXXFLAGS)  $< -MMD -c -o $@
```

This causes following error:
`src/xmq.c:938:9: fatal error: 'libxml/tree.h' file not found`

With following modification in Makefile, compilation succeded.

```
$(BUILD)/%.o: src/%.c $(wildcard src/%.h)
	$(CXX) -I/usr/local/include/libxml2 $(CXXFLAGS) $< -c -E > $@.src
	$(CXX) -I/usr/local/include/libxml2 -fpermissive $(CXXFLAGS)  $< -MMD -c -o $@
```

It would be nice to not hardcode path to libxml2, thanks :)

weetmuts, 2024-11-11:
I'll take a look. Perhaps I can have the configure script look for libxml2 in user local.
Did you manually install libxml2 and libxslt1 ?

Also do you have pkg-config on freebsd.  Does this work?
pkg-config --cflags libxml2

Do you succesfully use wmbusmeters on freebsd? Which kind of dongle do you use?

nefh, 2024-11-11:
Thanks. I've installed libxml2 by package manager:
`pkg install libxml2`

The pkg-config is not present in FreeBSD anymore. There is pkgconf ([freshports](https://www.freshports.org/devel/pkgconf), [git](https://gitea.treehouse.systems/ariadne/pkgconf)), but it is not installed by default.

closed 2025-02-10 by github-actions[bot]