wmbusmeters/wmbusmeters #1418: Support for ISTA water meter

MaxXor, 2024-11-13:
### Type of request

New device support

### Meter description

Hello, I'm trying to add a new xmq driver for a ISTA water meter. The water meter can be decoded already successfully by evo868 driver (see logged telegrams below), which my new istawater.xmq is based on.

Now decoding works fine when run with below `--analyze` command, but when I'm running wmbusmeters in daemon mode and feed it with telegrams, it's crashing with segmentation fault exit code 139. I'm sending encrypted telegrams via MQTT and they're successfully decrypted, parsed with evo868 driver and returned on a different MQTT topic. But if I use my istawater.xmq driver it's crashing. So I think something is wrong with my xmq driver below. Any help is appreciated.

```
# debug works
./wmbusmeters --debug --driver=/wmbusmeters_data/etc/wmbusmeters.drivers.d/istawater.xmq --analyze 434474262957702231078C2095900F002C25D744B09E30930AFB644687F97AD7002007102F2F0C136960320082016C1E398C0113626032002F2F2F2F2F2F2F2F2F2F2F2F
```

Content of `wmbusmeters.conf`:

```
loglevel=verbose
donotprobe=all
logtelegrams=true
format=json
logfile=/dev/stdout
device=hex:CMD(/usr/bin/mosquitto_sub -h mosquitto -t wmbusmeters/payload | tr -d "\n")
shell=/usr/bin/mosquitto_pub -h mosquitto -t wmbusmeters/parse -m "$METER_JSON"
alarmshell=/usr/bin/mosquitto_pub -h mosquitto -t wmbusmeters/alarm -m "$ALARM_TYPE $ALARM_MESSAGE"
```

Content of `WaterMeter.conf`:

```
name=WaterMeter
id=22705729
key=####
driver=istawater
```

The `istawater.xmq` looks like this:

```
driver {
    name           = istawater
    meter_type     = WaterMeter
    default_fields = name,id,total_m3,timestamp
    detect {
        mvt = IST,31,07
    }
    library {
        use = total_m3,target_m3,target_date
    }
    fields {
        field {
            name     = total
            quantity = Volume
            match {
                measurement_type = Instantaneous
                vif_range        = Volume
                tariff_nr        = 0
            }
        }
        field {
            name         = target
            quantity     = PointInTime
            display_unit = date
            match {
                measurement_type = Instantaneous
                vif_range        = Date
                storage_nr       = 2
            }
        }
        field {
            name         = target
            quantity     = Volume
            match {
                measurement_type = Instantaneous
                vif_range        = Volume
                tariff_nr        = 0
                storage_nr       = 2
            }
        }
    }
}
```

### Logged telegrams from meter in decrypted format

```shell
https://wmbusmeters.org/analyze/434474262957702231078C2095900F002C25D744B09E30930AFB644687F97AD7002007102F2F0C136960320082016C1E398C0113626032002F2F2F2F2F2F2F2F2F2F2F2F
```

Mexit, 2025-01-13:
@MaxXor 
Any progress?
How and with what do you read the data from the meter? How often is it sent?

MaxXor, 2025-01-16:
No progress yet. It's basically the same as evo868 driver, i just need to translate it to xmq format.

closed 2025-04-17 by github-actions[bot]

weetmuts, 2026-05-03:
Many water meter drivers use the same dif vif field for the total consumption, so for total_m3 any water driver works, so evo868 was perfectly fine for you.

It gets trickier when you want to read the other data, the istawater driver picks up storage nr 2 for target, ie end of last billing period.
Where most other meters seems to use storage nr 1. 

So total_m3 could be picked from the library but target_m3 could not be picked from the library (since that uses storage nr 1) and a new explicit field rule had to be created with storage_nr 2.