wmbusmeters/wmbusmeters #1422: wmbusmeters working but not getting any message from my meter

guibar, 2024-11-20:
### Type of request

Question

### OS version

Ubuntu 22.04

### wmbusmeters version

1.17.1-29-g3af7a4c

### Your message goes here

I have managed to compile and run wmbusmeters on my Ubuntu laptop. I use a Dongle RTL-SDR R820T2 connected on the USB port. When I run 
`wmbusmeters auto:t1`

I get some telegrams from a meter which is not mine but which seems to indicate that wmbusmeters and the dongle are running fine. 

I have a water meter of the iPerl sensus type which is supported by the wmbusmeters project. However, I don't  get any message from my meter even when I am next to the meter with my laptop and dongle.
But since I get water bills with the correct readings, the meter must be sending messages for the water company.
Any idea of what I could be doing wrong or what I could try to get messages from my Sensus meter?
Thanks in advance for your help.

kzajac83, 2024-11-21:
If see other telegrams then maybe T1 is disabled and person reading water-meter remotely send request T2 and in the same time received once T1 telegram.

guibar, 2024-11-23:
> If see other telegrams then maybe T1 is disabled and person reading water-meter remotely send request T2 and in the same time received once T1 telegram.

Thanks for your reply, so you suggest that the meter maybe only sends T1 telegrams upon receiving some request (which is possible since this meter has bidirectional communication). 
If that is the case, I can't do anything because I don't have the ability to send the appropriate information for the meter to emit; I can only to listen to what it sends, not send information.

kzajac83, 2024-11-23:
Not sure wmbusmeters can general generate T2 but if yes then what structure of telegram request? 

But maybe ask your water company when the guy reading the meters and ask him on place he can change setting to send every 5 or 10 min T1 telegram?

Or maybe change from T1 to other "one" modes especially c1

`    --listento=<mode> listen to one of the c1,t1,s1,s1m,n1a-n1f link modes
`
 Or add all after coma.

guibar, 2024-12-30:
> Not sure wmbusmeters can general generate T2 but if yes then what structure of telegram request?
>
> But maybe ask your water company when the guy reading the meters and ask him on place he can change [...]

Thanks for your help. I tried ringing my water company and they are not very cooperative. I seem to remember I had tried with all modes and had seen no message.

closed 2025-03-31 by github-actions[bot]