wmbusmeters/wmbusmeters #1435: Stopped receiving certain telegrams

nieeck, 2024-12-08:
### Type of request

Question

### OS version

HAOS 13.2

### wmbusmeters version

1.18.0-RC1

### Your message goes here

I have some Maddalena Radioevo wmbus water meters (https://www.maddalena.it/en/product/radio-evo/). I was getting telegrams from them a couple of weeks ago. I (miserably) tried to configure the meters in HA/MQTT. Then suddenly they disappeared form the log.

Log as of now:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nginx.sh
cont-init: info: /etc/cont-init.d/nginx.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun flask (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
services-up: info: copying legacy longrun wmbusmeters (no readiness notification)
[08:45:33] INFO: Starting web configuration service.
[08:45:33] INFO: Starting wmbusmeters service.
s6-rc: info: service legacy-services successfully started
[08:45:33] INFO: CONFIG_CONF ...
[08:45:33] INFO: {"device":"auto:s1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[08:45:33] INFO: CONFIG_METERS ...
[08:45:33] INFO: []
[08:45:33] INFO: Syncing wmbusmeters configuration ...
[08:45:33] INFO: Registering meters ...
[08:45:33] INFO: Generating MQTT configuration ... 
[08:45:33] INFO: Starting Nginx...
2024/12/08 08:45:33 [notice] 94#94: using the "epoll" event method
2024/12/08 08:45:33 [notice] 94#94: nginx/1.24.0
2024/12/08 08:45:33 [notice] 94#94: OS: Linux 6.6.54-haos
2024/12/08 08:45:33 [notice] 94#94: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/12/08 08:45:33 [notice] 94#94: start worker processes
2024/12/08 08:45:33 [notice] 94#94: start worker process 192
[08:45:33] INFO: Broker core-mosquitto will be used.
[08:45:33] INFO: 
MQTT Discovery ...
[08:45:33] INFO: MQTT Discovery cleanup...
172.30.32.2 - - [08/Dec/2024:08:45:34 +0100] "GET / HTTP/1.1" 200 1342 "http://homeassistant.local:8123/c8a990ad_wmbusmeters-ha-addon" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
172.30.32.2 - - [08/Dec/2024:08:45:34 +0100] "GET /main.js HTTP/1.1" 200 9981 "http://homeassistant.local:8123/api/hassio_ingress/ZSQDPo_G3kJTqr4wP7q4dJnC9FerH-4u6fdLcgxpUJA/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
172.30.32.2 - - [08/Dec/2024:08:45:34 +0100] "GET /bootstrap.bundle.min.js HTTP/1.1" 200 80762 "http://homeassistant.local:8123/api/hassio_ingress/ZSQDPo_G3kJTqr4wP7q4dJnC9FerH-4u6fdLcgxpUJA/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
172.30.32.2 - - [08/Dec/2024:08:45:34 +0100] "GET /bootstrap.min.css HTTP/1.1" 200 221227 "http://homeassistant.local:8123/api/hassio_ingress/ZSQDPo_G3kJTqr4wP7q4dJnC9FerH-4u6fdLcgxpUJA/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
172.30.32.2 - - [08/Dec/2024:08:45:34 +0100] "GET /wmbusmeters.png HTTP/1.1" 200 55338 "http://homeassistant.local:8123/api/hassio_ingress/ZSQDPo_G3kJTqr4wP7q4dJnC9FerH-4u6fdLcgxpUJA/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
172.30.32.2 - - [08/Dec/2024:08:45:34 +0100] "GET /get_json HTTP/1.1" 200 320 "http://homeassistant.local:8123/api/hassio_ingress/ZSQDPo_G3kJTqr4wP7q4dJnC9FerH-4u6fdLcgxpUJA/" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36"
[08:45:34] INFO: Running wmbusmeters ...
Started auto rtlwmbus[00000001] listening on s1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 12980***
          manufacturer: (KAW) Kamstrup, Denmark (0x2c37)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x3c
                device: rtlwmbus[00000001]
                  rssi: 141 dBm
                driver: flowiq2200

```

Telegrams I was receiving before (i did a small export at some point):

```
Received telegram from: 2306****
--
manufacturer: (MAD) Maddalena,   Italy (0x3424)
type: Warm Water   (30¬∞C-90¬∞C) meter (0x06)
ver: 0x50 |  
device: rtlwmbus[00000001]
rssi: 95 dBm |  
driver: unknown!
Received   telegram from: 1298****
manufacturer: (KAW) Kamstrup,   Denmark (0x2c37)
type: Cold water meter   (0x16) encrypted
ver: 0x3c |  
device: rtlwmbus[00000001]
rssi: 142 dBm |  
driver: flowiq2200
Received   telegram from: 1298****
manufacturer: (KAW) Kamstrup,   Denmark (0x2c37)
type: Cold water meter   (0x16) encrypted
ver: 0x3c |  
device: rtlwmbus[00000001]
rssi: 145 dBm |  
driver: flowiq2200
Received   telegram from: 2305****
manufacturer: (MAD) Maddalena,   Italy (0x3424)
type: Warm Water   (30¬∞C-90¬∞C) meter (0x06)
ver: 0x50 |  
device: rtlwmbus[00000001]
rssi: 99 dBm |  
driver: unknown!

```

My config file:
```
device=auto:s1
donotprobe=/dev/ttyAMA0
format=json
logfile=/dev/stdout
loglevel=normal
logtelegrams=false
shell=/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON"
```

I'm really losing my mind now. Reisntalled wmbusmeter 3x, used edge version, changed to c1, t1, s1, etc. Nothing always only geetting the Kamstrup telegrams.

I generally never ask for help, but am really stuck this time since weeks. Thank you for your wisdom.

kzajac83, 2024-12-23:
General (w)m-bus devices have internal clock, your must have too because long frame carry historical data with dates. 

General if you hear the this device and now no and see other meters (Kamstrup) then probably something wrong with your meter or very specific config of heat meter. 
The overlay can transmitter T1 (Default), C1 (Optional).

Btw check this:
https://bit.ly/AppMaddalenaEVO

Maybe NFC config is possible?

nieeck, 2024-12-26:
Thank you, Kzajac. Really appreciated.
I wonder why i doubted you (why would have you given your hint if you didnt' know?) - I will never get used to this platform :)
Looks like they work from 7am to 7pm - am so impressed how the clock is still accurate after a couple of years.

So the problem was that when i realized i had to remove all meters to be able to receive logs it was at night.

Now I can get all the telegrams. Am so happy!

nieeck, 2024-12-26:
resolved

closed 2024-12-26 by nieeck