wmbusmeters/wmbusmeters #1440: Maddalena radio evo doesn't seem to work properly

Stubenh0cker, 2024-12-12:
### Type of request

Existing device improvement

### Meter description

Hey everyone,

I have Maddalena water meters installed and can receive the telegrams. Unfortunately I receive an error message, that the selected driver isn't the right one. 

When analyzing the telegram in https://wmbusmeters.org/analyze/ it looks like there is something off in line 83 (I have no clue what the expectation is)

telegram=|AC4424349418032150167A39000020_2F2F041356850000046D0A340B3C04FD17000000000E786858000000004413D0620000426CFF2C8401137A83000082016C1E3BD3013BCF0200C4016D09210E338104FD280182046C1E3B8404137A830000C40413BE7E00008405131A7B0000C405136B7600008406132A700000C40613ED6A00008407130F690000C40713536800008408136E670000C40813EE65000084091330640000C40913D0620000|

Auto driver    : not found!
Similar driver : werhlemodwm 52/88
Using driver   : evo868 00/00
000   : ac length (172 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 2434 dll-mfct (MAD)
004   : 94180321 dll-id (21031894)
008   : 50 dll-version
009   : 16 dll-type (Cold water meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 39 tpl-acc-field
012   : 00 tpl-sts-field (OK)(OK)
013   : 0020 tpl-cfg 2000 (synchronous )
015   : 2F skip
016   : 2F skip
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 13 vif (Volume l)
019 C!: 56850000 ("total_m3":34.134)
023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
024   : 6D vif (Date and time type)
025 C!: 0A340B3C ("device_date_time":"2024-12-11 20:10")
029   : 04 dif (32 Bit Integer/Binary Instantaneous value)
030   : FD vif (Second extension FD of VIF-codes)
031   : 17 vife (Error flags (binary))
032 C!: 00000000 ("current_status":"OK")
036   : 0E dif (12 digit BCD Instantaneous value)
037   : 78 vif (Fabrication no)
038 C!: 685800000000 ("fabrication_no":"000000005868")
044   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
045   : 13 vif (Volume l)
046 C!: D0620000 ("consumption_at_set_date_m3":25.296)
050   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
051   : 6C vif (Date type G)
052 C!: FF2C ("set_date":"2023-12-31")
054   : 84 dif (32 Bit Integer/Binary Instantaneous value)
055   : 01 dife (subunit=0 tariff=0 storagenr=2)
056   : 13 vif (Volume l)
057 C!: 7A830000 ("consumption_at_set_date_2_m3":33.658)
061   : 82 dif (16 Bit Integer/Binary Instantaneous value)
062   : 01 dife (subunit=0 tariff=0 storagenr=2)
063   : 6C vif (Date type G)
064 C!: 1E3B ("set_date_2":"2024-11-30")
066   : D3 dif (24 Bit Integer/Binary Maximum value storagenr=1)
067   : 01 dife (subunit=0 tariff=0 storagenr=3)
068   : 3B vif (Volume flow l/h)
069 C!: CF0200 ("max_flow_since_datetime_m3h":0.719)
072   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
073   : 01 dife (subunit=0 tariff=0 storagenr=3)
074   : 6D vif (Date and time type)
075 C!: 09210E33 ("max_flow_datetime":"2024-03-14 01:09")
079   : 81 dif (8 Bit Integer/Binary Instantaneous value)
080   : 04 dife (subunit=0 tariff=0 storagenr=8)
081   : FD vif (Second extension FD of VIF-codes)
082   : 28 vife (Storage interval month(s))
083 C?: 01
084   : 82 dif (16 Bit Integer/Binary Instantaneous value)
085   : 04 dife (subunit=0 tariff=0 storagenr=8)
086   : 6C vif (Date type G)
087 C!: 1E3B ("history_reference_date":"2024-11-30")
089   : 84 dif (32 Bit Integer/Binary Instantaneous value)
090   : 04 dife (subunit=0 tariff=0 storagenr=8)
091   : 13 vif (Volume l)
092 C!: 7A830000 ("consumption_at_history_1_m3":33.658)
096   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
097   : 04 dife (subunit=0 tariff=0 storagenr=9)
098   : 13 vif (Volume l)
099 C!: BE7E0000 ("consumption_at_history_2_m3":32.446)
103   : 84 dif (32 Bit Integer/Binary Instantaneous value)
104   : 05 dife (subunit=0 tariff=0 storagenr=10)
105   : 13 vif (Volume l)
106 C!: 1A7B0000 ("consumption_at_history_3_m3":31.514)
110   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
111   : 05 dife (subunit=0 tariff=0 storagenr=11)
112   : 13 vif (Volume l)
113 C!: 6B760000 ("consumption_at_history_4_m3":30.315)
117   : 84 dif (32 Bit Integer/Binary Instantaneous value)
118   : 06 dife (subunit=0 tariff=0 storagenr=12)
119   : 13 vif (Volume l)
120 C!: 2A700000 ("consumption_at_history_5_m3":28.714)
124   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
125   : 06 dife (subunit=0 tariff=0 storagenr=13)
126   : 13 vif (Volume l)
127 C!: ED6A0000 ("consumption_at_history_6_m3":27.373)
131   : 84 dif (32 Bit Integer/Binary Instantaneous value)
132   : 07 dife (subunit=0 tariff=0 storagenr=14)
133   : 13 vif (Volume l)
134 C!: 0F690000 ("consumption_at_history_7_m3":26.895)
138   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
139   : 07 dife (subunit=0 tariff=0 storagenr=15)
140   : 13 vif (Volume l)
141 C!: 53680000 ("consumption_at_history_8_m3":26.707)
145   : 84 dif (32 Bit Integer/Binary Instantaneous value)
146   : 08 dife (subunit=0 tariff=0 storagenr=16)
147   : 13 vif (Volume l)
148 C!: 6E670000 ("consumption_at_history_9_m3":26.478)
152   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
153   : 08 dife (subunit=0 tariff=0 storagenr=17)
154   : 13 vif (Volume l)
155 C!: EE650000 ("consumption_at_history_10_m3":26.094)
159   : 84 dif (32 Bit Integer/Binary Instantaneous value)
160   : 09 dife (subunit=0 tariff=0 storagenr=18)
161   : 13 vif (Volume l)
162 C!: 30640000 ("consumption_at_history_11_m3":25.648)
166   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
167   : 09 dife (subunit=0 tariff=0 storagenr=19)
168   : 13 vif (Volume l)
169 C!: D0620000 ("consumption_at_history_12_m3":25.296)

{
    "_":"telegram",
    "media":"cold water",
    "meter":"evo868",
    "name":"",
    "id":"21031894",
    "consumption_at_history_1_m3":33.658,
    "consumption_at_history_10_m3":26.094,
    "consumption_at_history_11_m3":25.648,
    "consumption_at_history_12_m3":25.296,
    "consumption_at_history_2_m3":32.446,
    "consumption_at_history_3_m3":31.514,
    "consumption_at_history_4_m3":30.315,
    "consumption_at_history_5_m3":28.714,
    "consumption_at_history_6_m3":27.373,
    "consumption_at_history_7_m3":26.895,
    "consumption_at_history_8_m3":26.707,
    "consumption_at_history_9_m3":26.478,
    "consumption_at_set_date_m3":25.296,
    "consumption_at_set_date_2_m3":33.658,
    "history_1_date":"2024-11-30",
    "history_10_date":"2024-02-29",
    "history_11_date":"2024-01-31",
    "history_12_date":"2023-12-31",
    "history_2_date":"2024-10-31",
    "history_3_date":"2024-09-30",
    "history_4_date":"2024-08-31",
    "history_5_date":"2024-07-31",
    "history_6_date":"2024-06-30",
    "history_7_date":"2024-05-31",
    "history_8_date":"2024-04-30",
    "history_9_date":"2024-03-31",
    "history_reference_date":"2024-11-30",
    "max_flow_since_datetime_m3h":0.719,
    "total_m3":34.134,
    "current_status":"OK",
    "device_date_time":"2024-12-11 20:10",
    "fabrication_no":"000000005868",
    "max_flow_datetime":"2024-03-14 01:09",
    "set_date":"2023-12-31",
    "set_date_2":"2024-11-30",
    "timestamp":"2024-12-12T08:26:58Z"
}

Would be great, if you can have a look into that matter.

Best
Jan

### Logged telegrams from meter in decrypted format

```shell
Started auto iu891a[00202006] on /dev/ttyACM0 listening on t1,c1
(meter) Kaltwasser Wohnzimmer: meter detection did not match the selected driver evo868! correct driver is: unknown!
(meter) Not printing this warning again for id: 21031894 mfct: (MAD) Maddalena, Italy (0x3424) type: Cold water meter (0x16) ver: 0x50
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|AC4424349418032150167A39000020_2F2F041356850000046D0A340B3C04FD17000000000E786858000000004413D0620000426CFF2C8401137A83000082016C1E3BD3013BCF0200C4016D09210E338104FD280182046C1E3B8404137A830000C40413BE7E00008405131A7B0000C405136B7600008406132A700000C40613ED6A00008407130F690000C40713536800008408136E670000C40813EE65000084091330640000C40913D0620000|+25
```

Stubenh0cker, 2024-12-22:
Just a short update: according to the documentation the problem is the storage interval value. Right now, I have no idea where to start but I'll try my best to solve the issue.

btw: is there any way to shorten the telegrams? I probably only need the part before the storage interval.

[image]

nieeck, 2024-12-26:
Am getting the same issue at line 83 with the same product it seems (MAD,50,16).

Did you happen to solve it?
I started playing around with custom xmq drivers by loading them via HA WEB UI, but with no success.

telegram=|AC4424345408062350167ACA000020_2F2F04130E120200046D37321A3C04FD17000400000E785181048223004413ED160100426CFF2C84011348FF010082016C1E3BD3013B1C0500C4016D202B09388104FD280182046C1E3B84041348FF0100C404133CF10100840513CFD30100C4051391A00100840613CE650100C406133B2D0100840713901B0100C407132519010084081374180100C40813891701008409134F170100C40913ED160100|

Auto driver    : not found!
Similar driver : werhlemodwm 52/88
Using driver   : evo868 00/00
000   : ac length (172 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 2434 dll-mfct (MAD)
004   : 54080623 dll-id (23060854)
008   : 50 dll-version
009   : 16 dll-type (Cold water meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : ca tpl-acc-field
012   : 00 tpl-sts-field (OK)(OK)
013   : 0020 tpl-cfg 2000 (synchronous )
015   : 2F skip
016   : 2F skip
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 13 vif (Volume l)
019 C!: 0E120200 ("total_m3":135.694)
023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
024   : 6D vif (Date and time type)
025 C!: 37321A3C ("device_date_time":"2024-12-26 18:55")
029   : 04 dif (32 Bit Integer/Binary Instantaneous value)
030   : FD vif (Second extension FD of VIF-codes)
031   : 17 vife (Error flags (binary))
032 C!: 00040000 ("current_status":"ERROR_FLAGS_400")
036   : 0E dif (12 digit BCD Instantaneous value)
037   : 78 vif (Fabrication no)
038 C!: 518104822300 ("fabrication_no":"002382048151")
044   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
045   : 13 vif (Volume l)
046 C!: ED160100 ("consumption_at_set_date_m3":71.405)
050   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
051   : 6C vif (Date type G)
052 C!: FF2C ("set_date":"2023-12-31")
054   : 84 dif (32 Bit Integer/Binary Instantaneous value)
055   : 01 dife (subunit=0 tariff=0 storagenr=2)
056   : 13 vif (Volume l)
057 C!: 48FF0100 ("consumption_at_set_date_2_m3":130.888)
061   : 82 dif (16 Bit Integer/Binary Instantaneous value)
062   : 01 dife (subunit=0 tariff=0 storagenr=2)
063   : 6C vif (Date type G)
064 C!: 1E3B ("set_date_2":"2024-11-30")
066   : D3 dif (24 Bit Integer/Binary Maximum value storagenr=1)
067   : 01 dife (subunit=0 tariff=0 storagenr=3)
068   : 3B vif (Volume flow l/h)
069 C!: 1C0500 ("max_flow_since_datetime_m3h":1.308)
072   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
073   : 01 dife (subunit=0 tariff=0 storagenr=3)
074   : 6D vif (Date and time type)
075 C!: 202B0938 ("max_flow_datetime":"2024-08-09 11:32")
079   : 81 dif (8 Bit Integer/Binary Instantaneous value)
080   : 04 dife (subunit=0 tariff=0 storagenr=8)
081   : FD vif (Second extension FD of VIF-codes)
082   : 28 vife (Storage interval month(s))
**083 C?: 01**
084   : 82 dif (16 Bit Integer/Binary Instantaneous value)
085   : 04 dife (subunit=0 tariff=0 storagenr=8)
086   : 6C vif (Date type G)
087 C!: 1E3B ("history_reference_date":"2024-11-30")
089   : 84 dif (32 Bit Integer/Binary Instantaneous value)
090   : 04 dife (subunit=0 tariff=0 storagenr=8)
091   : 13 vif (Volume l)
092 C!: 48FF0100 ("consumption_at_history_1_m3":130.888)
[... 66 lines cut ...]
    "history_10_date":"2024-02-29",
    "history_11_date":"2024-01-31",
    "history_12_date":"2023-12-31",
    "history_2_date":"2024-10-31",
    "history_3_date":"2024-09-30",
    "history_4_date":"2024-08-31",
    "history_5_date":"2024-07-31",
    "history_6_date":"2024-06-30",
    "history_7_date":"2024-05-31",
    "history_8_date":"2024-04-30",
    "history_9_date":"2024-03-31",
    "history_reference_date":"2024-11-30",
    "max_flow_since_datetime_m3h":1.308,
    "total_m3":135.694,
    "current_status":"ERROR_FLAGS_400",
    "device_date_time":"2024-12-26 18:55",
    "fabrication_no":"002382048151",
    "max_flow_datetime":"2024-08-09 11:32",
    "set_date":"2023-12-31",
    "set_date_2":"2024-11-30",
    "timestamp":"2024-12-26T18:10:36Z"
}

Using: wmbusmeters: 1.17.1-70-g41a2b48
41a2b4808f00367af7bbaac57e221539825babbb

Stubenh0cker, 2024-12-29:
Hi Nieeck, 

unfortunately I'm still at the very beginning. I think the problem is, that the implementation only reads short frame telegrams. I think I would have to add the storage interval value or maybe ignore it. I don't have any c++ expertise so at the moment I really don't know where to start.

Stubenh0cker, 2024-12-29:
Short Update: 

I think I found the cause for the problem: 

the evo868 implementation matches on water meters  (0x07)

`di.addDetection(MANUFACTURER_MAD,  0x07,  0x50);`

My meter, however, specifically sends **cold water** and **hot water** values. 

Locally after adding: 

`di.addDetection(MANUFACTURER_MAD,  0x16,  0x50);` 

the reading seems to be fine. My understanding is that adding cold and warm water meter should do the job...

Stubenh0cker, 2025-01-04:
After adding the cold and warm water detection it works like a charm. weetmuts already merged it into the repository. Thanks for all the great work!

majherek, 2025-01-21:
Hi, I have similar problem.

I have [Maddalena RADIO EVO wM-BUS RADIO MODULE 868 MHz](https://www.maddalena.it/en/product/radio-evo/). 

I receive telegrams:

```
(serial) received ascii "C1;1;1;2025-01-21 13:14:44.000;148;132;23068266;0x4e44ae4c6682062372077a80334025782bbea0f6dd5698085f9711123673011a1c7e7f1bab47c151101dd7a62c5b64c365e7e193b50546a8745395a1e5f63effb35151fd810e3b5f27127b37f5db45<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2025-01-21 13:14:44.000;148;132;23068266;0x4e44ae4c6682062372077a80334025782bbea0f6dd5698085f9711123673011a1c7e7f1bab47c151101dd7a62c5b64c365e7e193b50546a8745395a1e5f63effb35151fd810e3b5f27127b37f5db45<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 0 templates.
(wmbus) parseDLL @0 79
(wmbus) parseELL @10 69
(wmbus) parseNWL @10 69
(wmbus) parseAFL @10 69
(wmbus) parseTPL @10 69
(wmbus) parseDLL @0 79
(telegram) DLL L=4e C=44 (from meter SND_NR) M=4cae (SEN) A=23068266 VER=72 TYPE=07 (Water meter) (driver unknown!) DEV=rtlwmbus[012345671] RSSI=148
(wmbus) parseELL @10 69
(wmbus) parseNWL @10 69
(wmbus) parseAFL @10 69
(wmbus) parseTPL @10 69
(TPL) num encrypted blocks 4 (64 bytes and remaining unencrypted 0 bytes)
(telegram) TPL CI=7a ACC=80 STS=33 CFG=2540 (synchronous AES_CBC_IV nb=4 cntn=0 ra=0 hc=0)
Received telegram from: 23068266
          manufacturer: (SEN) Sensus Metering Systems, Germany (0x4cae)
                  type: Water meter (0x07) encrypted
                   ver: 0x72
                device: rtlwmbus[012345671]
                  rssi: 148 dBm
                driver: unknown!
(wmbus) 000   : 4e length (78 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : ae4c dll-mfct (SEN)
(wmbus) 004   : 66820623 dll-id (23068266)
(wmbus) 008   : 72 dll-version
(wmbus) 009   : 07 dll-type (Water meter)
(wmbus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 011   : 80 tpl-acc-field
(wmbus) 012   : 33 tpl-sts-field (ALARM TEMPORARY_ERROR)
(wmbus) 013   : 4025 tpl-cfg 2540 (synchronous AES_CBC_IV nb=4 cntn=0 ra=0 hc=0 )
(wmbus) 015 CE: 782BBEA0F6DD5698085F9711123673011A1C7E7F1BAB47C151101DD7A62C5B64C365E7E193B50546A8745395A1E5F63EFFB35151FD810E3B5F27127B37F5DB45 encrypted
telegram=|_4E44AE4C6682062372077A80334025782BBEA0F6DD5698085F9711123673011A1C7E7F1BAB47C151101DD7A62C5B64C365E7E193B50546A8745395A1E5F63EFFB35151FD810E3B5F27127B37F5DB45|+64485
(wmbus) telegram from 23068266 ignored by all configured meters!
```

As you can see from picture, ID for RadioEVO matchs: `23068266`.

[image]

When trying to decode using `wmbusmeters` (compiled yesterday from github) I get:
```
~/wmbusmeters/build$ ./wmbusmeters 4e44ae4c6682062372077a7f334025f747519B0eeB6e91474Bdc080B13d48Be253B76d733a690034aBd0eedc3532dB722a1595e983a30a9829e38e7391d203dc44df1d2e173f0cfd99dfccB60ecf7B test evo868 23068266 NOKEY
(meter) test: meter detection did not match the selected driver evo868! correct driver is: unknown!
(meter) Not printing this warning again for id: 23068266 mfct: (SEN) Sensus Metering Systems, Germany (0x4cae) type: Water meter (0x07) ver: 0x72
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (test 23068266 evo868) did not handle telegram!
```
So which driver should I use? 
```
mfct: (SEN) Sensus Metering Systems, Germany (0x4cae) type: Water meter (0x07) ver: 0x72
```

majherek, 2025-01-21:
Ok, 

I think it's another system:

My water meter is: [Sensus DomoJet Water Meter](https://www.xylem.com/en-uk/products--services/metrology-equipment-for-utilities/meters/domojet/) with [Domo R – Wireless M-Bus OMS radio module](https://www.xylem.com/en-uk/products--services/metrology-equipment-for-utilities/reading-devices/domo-modules/)

> The Sensus Domo R module adds wireless communication capabilities to the DomoJet. It is wireless M-Bus OMS generation 4 certified and 100% interoperable with OMS (Open Metering System) compatible systems and the Sensus DIAVASO Mobile App Suite. With an easily exchangeable battery and several alarm functions, the Domo R is an innovative and durable device with an IP67 certification.

majherek, 2025-01-21:
I opened new issue:  #1468. Sorry for spam.

closed 2025-04-22 by github-actions[bot]