wmbusmeters/wmbusmeters #1442: Qwater - looks like there are some changes in the protocol. The identify some telegrams as unencrypted

mel525, 2024-12-13:
Hi guys, resently i discovered wmbus world. First of all let me express my gratitude to you. You did a greate job. 
I copiled a scanner based on ESP32+CC1101+ESPHOME. In my appartment I have two counters for water and one for heat from QUNDIS: Qwater 5.5 and Qheat 5.5
I managed to capture Telegrams from the counters. Here is a sample:

[11:50:26][D][mbus:014]: Received C1 A frame
[11:50:26][I][wmbus:104]: wme5 [0x14771853] RSSI: -61dBm T: 48449344531877141A077AFF0030256C1D9707360531D2BA11EB9482DC910D605280AE7C44A403E83100207DEF5740B6D5F0805B53F0AC84D9D30E3FAF6BBB326CFFFF046D310B0C3C (73) C1 A
[11:50:33][D][mbus:014]: Received C1 A frame
[11:50:34][I][wmbus:104]: qwater [0x14770942] RSSI: -52dBm T: 49449344420977141A06780DFF5F35008214003533EE06BDDB15853D27144EA1D25A15336BD60455612E5B338D025C87C3B0D3B134857DFD19FFFF2F4631E537F1E9AB10046D320B0C3C (74) C1 A
[11:52:02][D][mbus:014]: Received C1 A frame
[11:52:02][D][crc:035]:     calculated: 0x96FF, read: 0x9141  !!!
[11:52:16][D][mbus:014]: Received C1 A frame
[11:52:16][I][wmbus:104]: wme5 [0x14771853] RSSI: -61dBm T: 49449344531877141A07780DFF5F350082FF00357D437084303A5216CF1339E648E82E96B7D02EA6D4965FA565C51DE1366324F4ECA9D045EBC095BE85240A556A1AD5EC046D330B0C3C (74) C1 A
[11:52:21][D][mbus:014]: Received C1 A frame
[11:52:21][D][crc:035]:     calculated: 0xF354, read: 0xF356  !!!
[11:53:21][D][mbus:014]: Received C1 A frame
[11:54:09][I][wmbus:104]: wme5 [0x14771853] RSSI: -60dBm T: 49449344531877141A07780DFF5F350082FF00350EE04BD4F75392F8C4D8041D26CF8243DCDDFD5BE4D02A9C438A0BB375B366FEF9483C598CE9812B4A324214DF957A64046D350B0C3C (74) C1 A
[11:54:18][D][mbus:014]: Received C1 A frame
[11:58:01][D][crc:035]:     calculated: 0xFD0E, read: 0x3EBE  !!!
[11:58:10][D][mbus:014]: Received C1 A frame
[11:58:10][I][wmbus:104]: wme5 [0x14771853] RSSI: -60dBm T: 34449344531877141A077A0000082004ED3902090C3301FD0C11046D390B0C3C02FD3CC2010DFF5F0C000885FF000006130705FFFC (53) C1 A
[11:58:41][D][mbus:014]: Received C1 A frame
[11:58:41][D][crc:035]:     calculated: 0x9DCC, read: 0x14CA  !!!
[11:58:50][D][mbus:014]: Received C1 A frame
[12:00:02][I][wmbus:104]: wme5 [0x14771853] RSSI: -64dBm T: 49449344531877141A07780DFF5F350082000035606F405FE289CE586070F3CA067FF4CC0543AF481D3C6044D664D119683C2C8827534D272E1B5B17326F19C9DCEDCB2A046D3B0B0C3C (74) C1 A
[12:00:17][D][mbus:014]: Received C1 A frame
[12:01:55][D][mbus:014]: Received C1 A frame
[12:01:55][I][wmbus:104]: wme5 [0x14771853] RSSI: -59dBm T: 49449344531877141A07780DFF5F3500820000358659CE83D8A72E9706CEE75E7766810883BFBBA434D9F8033E91C9EBFFFC213122DC63B8669E35CCC5700FBD507B92B8046D010C0C3C (74) C1 A
[12:01:56][D][mbus:014]: Received C1 A frame

What I notised: 
1. There are 3 type of telegrams 74 bytes, 73 bytes, 53 bytes
2. The regularity of telegrams is different 74 bytes (around 15 telegrams per hour, but only from 6am to 6pm), 73 bytes (2-4 telegramp per hour 24/7) and 53 bytes (3-4 telegrams per day). 
3. I have two similar water counters, one is detected as qwater another wme5

Now to the issues:
1. 74 bytes telegrams looks unencrypted on https://wmbusmeters.org/, however the decoded data is wrong.
2. 73 bytes telegrams looks encrypted

Here are samples of 74 bytes:
https://wmbusmeters.org/analyze/49449344531877141a07780dff5f350082d100352d5c8B7a06497ca34f2dc88Bea47279a71837fd290482822BBBc3e24ca5cB85d2f62866a33dad02e0dc63685B7dc7Bd5046d01060c3c 
06:02:19	wme5	14771853	-60	49449344531877141A07780DFF5F350082D100352D5C8B7A06497CA34F2DC88BEA47279A71837FD290482822BBBC3E24CA5CB85D2F62866A33DAD02E0DC63685B7DC7BD5046D01060C3C	 (74) C1 A
06:04:13	wme5	14771853	-60	49449344531877141A07780DFF5F350082D10035DFA7060B6A6DF61DA09F857ED1848C4B25B7C69A0C305D43C2EDD6721BCD7E095066DDEDF0610186DBCAF2C57D800AC1046D03060C3C	 (74) C1 A
07:21:51	wme5	14771853	-59	49449344531877141A07780DFF5F350082DB0035C65C9D5BE9A7F37D955873FCA01DC0572296D95D5A880F867FECB514C4CCEC1BA908B0A5837653F53A660B0B58F6BF2B046D15070C3C	 (74) C1 A
07:25:46	wme5	14771853	-60	49449344531877141A07780DFF5F350082DC0035F81BA2A1BA1C68E472FFF9082D435A1903D11801D88E4822EDDCB91E336703438E72E74DA65980FEDEF6F7F71DB03CE8046D19070C3C	 (74) C1 A
08:45:43	wme5	14771853	-57	49449344531877141A07780DFF5F350082E60035BDB0AC0B463C9BEDFD468B38F4E1131233367A206434E5D027044E09ECAD0AB1D9BEC2048749C6028E48216923F318A7046D2D080C3C	 (74) C1 A
08:47:46	wme5	14771853	-59	49449344531877141A07780DFF5F350082E70035A7CDDFDB8AA3434EFB85F378448215B73D63A676C0C40DDAE66109EB0B3695C85071B3A0F0C4B1304831CB92306D67D4046D2F080C3C	 (74) C1 A
09:23:57	wme5	14771853	-58	49449344531877141A07780DFF5F350082EB0035D43A4459224AC24291E14FB0171FBD8073AE9E2569A64762854879CF97A58F8DA318A70DA62CC3129EA035AFBF1E8EDC046D17090C3C	 (74) C1 A
09:26:02	wme5	14771853	-57	49449344531877141A07780DFF5F350082EC00354B3379E6A86DBFA1A228168408AC7FEA39EF8145C9969867A52E4114390720D1FA0EEA26ACE08C5D872FE94C9E0AECF3046D19090C3C	 (74) C1 A
10:04:22	wme5	14771853	-61	49449344531877141A07780DFF5F350082F10035466EF75691F34C235D520BBB401DBF5C94F91E6D4E3C87ECDE2542A281A576123B223386BDE1A51F8576BDA88A68485C046D030A0C3C	 (74) C1 A
10:10:01	wme5	14771853	-61	49449344531877141A07780DFF5F350082F10035FA03755E330A152E0292B6A2C7CA01B9B53B4FD5EE38F024149803EB8C6EDB66979646AF057F99262F2BCBB4CC18EA5B046D090A0C3C	 (74) C1 A
11:21:25	wme5	14771853	-60	49449344531877141A07780DFF5F350082FB0035261863A60D10ED0F67BF82847B46613A6CE1E7CF7D38D3E329AD257AEF201D28C121BFCAC9BFF871B3B33AFCB152B73D046D140B0C3C	 (74) C1 A
11:25:08	wme5	14771853	-60	49449344531877141A07780DFF5F350082FB0035484AC7594A8557A05498D18270E2C6F5179280DE43EDBA7AF4F977F4BEA5BEE58C95CA0185073EBAB68C60F94DEC5EF1046D180B0C3C	 (74) C1 A
12:40:32	wme5	14771853	-58	49449344531877141A07780DFF5F3500820500357322E6540A20002C3191F95E9BE429480C26F1B11978F300645E923AAF3D60DB34511A69C91E6E090E10FED3E439CA3E046D280C0C3C	 (74) C1 A
12:44:15	wme5	14771853	-58	49449344531877141A07780DFF5F35008205003521CE56DA8035929B42D9ED864AD488FF204B7E48A9137DAF7703B339B376F431599DB6F2137F654D00E0D3A917580A35046D2B0C0C3C	 (74) C1 A
13:38:10	wme5	14771853	-60	49449344531877141A07780DFF5F3500820C0035D97FEA2E2DA5C7033A710E8044DFBED8DA69CD1EB73A03C23CE64DFD7AB30E03241CCE456204472A5CA5696A06451B40046D250D0C3C	 (74) C1 A
13:43:59	wme5	14771853	-60	49449344531877141A07780DFF5F3500820D0035AFDA014D8122B3B92120F3767674F88271E2ECE3906B483B7596A026DF53670E1B41EEDBA1C2A44655982D7551BF0526046D2B0D0C3C	 (74) C1 A
14:35:51	wme5	14771853	-60	49449344531877141A07780DFF5F350082140035B28358EC5FC6557F349C32F4DF92F463E7732F50C32D96AA78091B6CC0F1786A08628E638D5E58AB42725B9EB0A4725C046D230E0C3C	 (74) C1 A
14:37:41	wme5	14771853	-60	49449344531877141A07780DFF5F350082140035BAA8DA5508478208D887E2A1A61C3917EB683E26E3743F85336637B75B91D5BBED450EB8C1877A7E6B9217B910A76986046D250E0C3C	 (74) C1 A
15:08:18	wme5	14771853	-60	49449344531877141A07780DFF5F3500821800350983253D7DBDFFBDE17760667050BFED7E09CDFD927EE2996832C4CB637ED22E881F3A057003CAFC7AC62F84DCC34FFF046D070F0C3C	 (74) C1 A
15:12:11	wme5	14771853	-58	49449344531877141A07780DFF5F3500821900355A2F8D432C90FF77E48A3A9A076474ACCA6458FDA1EBB46FB49525E94245110B7C8F9B3D5CE13416C6F7269C29608DA7046D0B0F0C3C	 (74) C1 A
16:15:05	wme5	14771853	-58	49449344531877141A07780DFF5F3500822100354286D3B61DE1FF66C027ED7AB140809162C9D111C276E7FAC613795B4FB21C66DAB64C3ECF8107E54B66B4AE6CC6E181046D0E100C3C	 (74) C1 A
16:16:56	wme5	14771853	-60	49449344531877141A07780DFF5F350082210035DF96B5F9C2C222D92A2CC7AF917641682EB7168AFCFD560DF87F39EA9361E79446FDF6320952C5B14B8B299072F98206046D10100C3C	 (74) C1 A
17:15:50	wme5	14771853	-58	49449344531877141A07780DFF5F3500822900355C173DB73142765F1769221155005C9E4A1F57CC09C2D080BBCF03CCC1E6A363067F81FF240B7F8B952B376064793F69046D0F110C3C	 (74) C1 A
17:17:44	wme5	14771853	-57	49449344531877141A07780DFF5F3500822900358155B4453F98DE5C69085F9565155CA9E2C2B5E3ECE53559FB6C6A84AA8757E032123734F43D1AC2C5AB72CBD3C4E4C4046D11110C3C	 (74) C1 A
18:05:01	wme5	14771853	-61	49449344531877141A07780DFF5F3500822F003552A9137A5DE3269171DBCD8D56BFFB7E2187EAAFB18F9B89646E9F98F03361D01CC6561709FFF0D72CA653286E09EECC046D04120C3C	 (74) C1 A
18:07:00	wme5	14771853	-60	49449344531877141A07780DFF5F35008230003551CE0A3840A1FEED71F426A1C99A7E50367F84D65F002831E18A2F25B9E75E4BD7A2FE41A9908E796A9779B5BF371CC2046D06120C3C	 (74) C1 A
18:48:28	wme5	14771853	-57	49449344531877141A07780DFF5F35008235003513BDD86EDB9A5FEC977DFC6302C312E81F55A30193E5409DD66E3B90C8BF9F8BF9BEC85B61DDB46D5A26FEAF9245B8AB046D2F120C3C	 (74) C1 A
[... cut ...]
11:58:10	wme5	14771853	-60	34449344531877141A077A0000082004ED3902090C3301FD0C11046D390B0C3C02FD3CC2010DFF5F0C000885FF000006130705FFFC	 (53) C1 A
03:50:46	wme5	14771853	-59	34449344531877141A077AC000082004ED3902090C3301FD0C11046D32030C3C02FD3CC2010DFF5F0C000885FF000006130705FFFC	 (53) C1 A
20:05:47	wme5	14771853	-56	34449344531877141A077A4000082004ED3902090C3301FD0C11046D05140C3C02FD3CC2010DFF5F0C000885FF000006130705FFFC	 (53) C1 A

I believe that the data is encrypted, but maybe the tool can be adjusted to detect it.

MatthiasFriese, 2024-12-17:
I see a similar behavior and opened the following issue: #1441

MatthiasFriese, 2025-02-11:
Any update on this issue?

rbnhln, 2025-02-24:
Hey there, 

I have somehow the same problem. I am getting to different types of telegrams most like mel525. 
I get 73 bytes, and 52 bytes long telegrams.

I assume the shorter ones are the Q AMR ones, which should include according to Qundis (current consumption, last month consumtion + date, due date value + date and Status + ErrCode + Date).

And the longer ones are Q walk-by telegrams, which should include the data form Q AMR + 12 additional month values.

The time period of the Q wlak-by ones is 112 s,  between 8:00 and 18:00. The other ones (Q AMR) period is 450 s.

One thing I do not get, if the values need to be decrypted or the driver needs to be adjusted for that version of the Qundis Q water. 

Example Q AMR:
`34449344385571141a077ac000082004ed39170d1e3101fd0c11046d220f383202fd3cc2010dff5f0c000879ff000006130705fffc`

Example Q walk by:
`49449344385571141A0778_0DFF5F350082BE00359624641884D3EDE973E1C2A446F77C4D448556038CE4F94966862B6270913FF8C5F1EE4D357CDB99414F4AE6DFE34B67046D190F3832`

fuu0, 2025-03-27:
Hey!

same here:
`[D][mbus:014]: Received C1 A frame
[I][wmbus:092]: Using selected driver lse_07_17 (detected driver was wme5)
[I][wmbus:104]: lse_07_17 [0x52968411] RSSI: -55dBm T: 48449344118496521A077A8B00302502A968D2DFE5DDFA7C28A43D4CEADD339C9E461AAE9044FA16065F9912B103406BFB89EF0DC96845954A09A744B7A4B9326CFFFF046D0C0C3B33 (73) C1 A
[W][wmbus:124]: Link mode C1 not supported in driver lse_07_17
[D][mbus:014]: Received C1 A frame
[I][wmbus:092]: Using selected driver lse_07_17 (detected driver was wme5)
[I][wmbus:104]: lse_07_17 [0x52968411] RSSI: -55dBm T: 49449344118496521A07780DFF5F3500828B00354C44980ADF51AB6D338C02C7BCEEB623D04E5189F4DAA971DF9DE6D2B8BB9B5F4592A3A930E165500AAD7C53E5EB1F67046D0D0C3B33 (74) C1 A
[W][wmbus:124]: Link mode C1 not supported in driver lse_07_17
[D][mbus:014]: Received C1 A frame
[I][wmbus:092]: Using selected driver lse_07_17 (detected driver was wme5)
[I][wmbus:104]: lse_07_17 [0x52968411] RSSI: -54dBm T: 49449344118496521A07780DFF5F3500828B0035ADEEC592051A0136BEDD3F4831AE5EAB84D1C15BF6928E9BA0C4583898F38451323AE5D4D6B87DEC97CE438088DF9F90046D110C3B33 (74) C1 A
[W][wmbus:124]: Link mode C1 not supported in driver lse_07_17
[D][mbus:014]: Received C1 A frame
[I][wmbus:092]: Using selected driver lse_07_17 (detected driver was wme5)
[I][wmbus:104]: lse_07_17 [0x52968411] RSSI: -55dBm T: 49449344118496521A07780DFF5F3500828B003559F544315DFCF0829A5198CEEF92B7CEEA90F3C8148347B40C1F55D2F793E46AFB07B99231C48B9E629F3B0129BACF35046D130C3B33 (74) C1 A
[W][wmbus:124]: Link mode C1 not supported in driver lse_07_17
[D][mbus:014]: Received C1 A frame
[I][wmbus:092]: Using selected driver lse_07_17 (detected driver was wme5)
[I][wmbus:104]: lse_07_17 [0x52968411] RSSI: -58dBm T: 48449344118496521A077A8C003025EBA707C8FE0057609A27D8F52E16C856C4C18C7805622F4A08F2175A728B6636F16071C065EE7B719B5D3CB64F18D7EA326CFFFF046D130C3B33 (73) C1 A
[W][wmbus:124]: Link mode C1 not supported in driver lse_07_17
[D][mbus:014]: Received C1 A frame
[I][wmbus:092]: Using selected driver lse_07_17 (detected driver was wme5)
[I][wmbus:104]: lse_07_17 [0x52968411] RSSI: -53dBm T: 49449344118496521A07780DFF5F3500828C0035B94907452DB1257C6FBBFEF4DDCA95241E7E0ACF985F47BC2800FBEF65BBC487233922DAE6EE11F1EBAF46BDEBBD7DBF046D140C3B33 (74) C1 A
[W][wmbus:124]: Link mode C1 not supported in driver lse_07_17`

This should be the "Q walk-by" frames, I cant see the "Q AMR" Frames

Any hints on how to continue?

rbnhln, 2025-03-27:
No sadly not. 

I observed the following:

[image]

- around 7, 15 and 19 o'clock: I get always the same value of 555,519 (but the time is drifting)
- at approximately 6 o'clock: I receive the first data point of the day with a much higher value (eg. 85,559,104). This value rises through the day with each new data point. It does not matter if the actual value on the meter rises or not. The next day "highvalue" can be lower or higher.

So, sorry I do not understand whats going on there.

closed 2025-06-26 by github-actions[bot]