wmbusmeters/wmbusmeters #1446: Support for heat meter Siemens WFM503-J000H0 with Qundis add-on radio module WFZ16.OK

crpld, 2024-12-27:
### Type of request

New device support

### Meter description

Hi, I have Siemens WFM503-J000H0 heat meter with Qundis add-on radio module WFZ16.OK. I'm using iu891a usb module in s1 mode to capture telegrams in Home Assistant using the wmbusmeters plugin.  I'm using sensostar driver which is translating the data more or less ok.
I've used the following link to decode the telegram based on which I've selected the driver:

https://wmbusmeters.org/analyze/334465B2515041661A047A3E000000046D08121B3C0C0572051500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956

The fields that are green are correct and are matching the readout on the meter display. Field 039 I'm not sure what should be. Field 048 is the serial number of the add-on radio module while field 004 is the heatmeter serial number and the ID.

However I'm getting warning message in the logs. Maybe with minor modifications the sensostar driver along with the json for the homeassistat could be modified to fully match this meter. But I dont have the skills to do it :).

### Logged telegrams from meter in decrypted format

```shell
Started config iu891a[00202045] on /dev/ttyACM0 listening on s1
(meter) HeatMeter: meter detection did not match the selected driver sensostar! correct driver is: unknown!
(meter) Not printing this warning again for id: 66415051 mfct: (LSE) Landis Staefa electronic (0xb265) type: Heat meter (0x04) ver: 0x1a
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|334465B2515041661A047A3E000000_046D08121B3C0C0572051500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+9215

telegram=|334465B2515041661A047A32000000_046D1C13193C0C0536011500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+175
telegram=|334465B2515041661A047A33000000_046D1917193C0C0524021500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+10245
telegram=|334465B2515041661A047A34000000_046D17031A3C0C0558021500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+24520
telegram=|334465B2515041661A047A35000000_046D3A061A3C0C0579021500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+37412
telegram=|334465B2515041661A047A36000000_046D230A1A3C0C0518031500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+4065
telegram=|334465B2515041661A047A37000000_046D050F1A3C0C0575031500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+12049
telegram=|334465B2515041661A047A38000000_046D1E131A3C0C0572041500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+11201
telegram=|334465B2515041661A047A39000000_046D21171A3C0C0501051500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+25731
telegram=|334465B2515041661A047A3A000000_046D22031B3C0C0501051500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+40212
telegram=|334465B2515041661A047A3B000000_046D12071B3C0C0501051500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+53659
telegram=|334465B2515041661A047A3C000000_046D040B1B3C0C0501051500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+67186
telegram=|334465B2515041661A047A3D000000_046D220E1B3C0C0528051500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+79802
telegram=|334465B2515041661A047A3E000000_046D08121B3C0C0572051500426CFF2C4C0570901200326CFFFF02FD1700000C7828898956|+9215
```

BIBOLV, 2024-12-28:
Hi,
Nicely structured message. If you get correct readings and all necessary information needed, then do not worry about warning messages, those are just letting you know that something might not work as driver is not written for your meter specifically. So just ignore warning and enjoy data flow 😄 
But if you can't live with warning messages in logs - take [driver](https://github.com/wmbusmeters/wmbusmeters/blob/master/drivers/src/sensostar.xmq), paste it in ha-addon-edge config sidebar Drivers tab, add `mvt = LSE,1a,04` after line 8 and you should have no more warnings in logs. It should work - you can test and let us know 😉

crpld, 2024-12-30:
Hi. Thanks for the reply. I did a test with the driver mod and I get no more warnings :).

closed 2024-12-30 by crpld