wmbusmeters/wmbusmeters #1447: Water meter with Siemens WFZ661 add on radio module

crpld, 2024-12-27:
### Type of request

New device support

### Meter description

Hi, I have water meter with Siemens WFZ661 add-on radio module. I'm not sure what the meter manufacturer is as the radio module is covering a big part of the meter, but this is printed on the meter: DE-08-MI001-PTB018. I'm using iu891a usb module in s1 mode to capture telegrams in Home Assistant using the wmbusmeters plugin.  This meter is sending telegrams with two different lengths in a regular pattern (4x46 followed by 2x42 bytes length). I've used the following link to decode the telegrams:

46bytes:
https://wmbusmeters.org/analyze/2E4465B20646651733077AD70000000C13113655004C1359604600426CFF2C326CFFFF046D0115193C0C7864622835

The 46bytes telegram is decoded ok using minomess driver.  Field 017 is showing the total m3 of the meter and is matching with the counter on the meter. 023 and 029 is matching with the value of the meter at end of year 2023. 043 is the serial number printed on the radio add-on module and 004 is the serial number printed on the meter.

42bytes: https://wmbusmeters.org/analyze/2A4465B20646651733077ADB0000008C04133581540082046C1E3B01FD0C10046D150D1A3C0C7864622835

On the 42 bytes telegram the field 018 and 025 are matching with the value of the meter at the end of last month. The 42 byte telegram does not show info of the current total m3 of the meter.

I wasnt able to test this in the home assistant using minomess driver as the json is missing in the home assistant plugin. I would like to test this meter in home assistant using minomess driver but I cant find the json for this driver.

### Logged telegrams from meter in decrypted format

```shell
telegram=|2E4465B20646651733077AD7000000_0C13113655004C1359604600426CFF2C326CFFFF046D0115193C0C7864622835|+254
telegram=|2E4465B20646651733077AD8000000_0C13583655004C1359604600426CFF2C326CFFFF046D1E011A3C0C7864622835|+12287
telegram=|2E4465B20646651733077AD9000000_0C13673655004C1359604600426CFF2C326CFFFF046D20051A3C0C7864622835|+26804
telegram=|2E4465B20646651733077ADA000000_0C13033755004C1359604600426CFF2C326CFFFF046D23091A3C0C7864622835|+41369
telegram=|2A4465B20646651733077ADB000000_8C04133581540082046C1E3B01FD0C10046D150D1A3C0C7864622835|+337
telegram=|2A4465B20646651733077ADC000000_8C04133581540082046C1E3B01FD0C10046D06111A3C0C7864622835|+13809
telegram=|2E4465B20646651733077ADD000000_0C13673855004C1359604600426CFF2C326CFFFF046D28141A3C0C7864622835|+9879
telegram=|2E4465B20646651733077ADE000000_0C13073955004C1359604600426CFF2C326CFFFF046D0B001B3C0C7864622835|+22533
telegram=|2E4465B20646651733077ADF000000_0C13163955004C1359604600426CFF2C326CFFFF046D21041B3C0C7864622835|+38271
telegram=|2E4465B20646651733077AE0000000_0C13273955004C1359604600426CFF2C326CFFFF046D3A081B3C0C7864622835|+54185
telegram=|2A4465B20646651733077AE1000000_8C04133581540082046C1E3B01FD0C10046D110D1B3C0C7864622835|+69733
telegram=|2A4465B20646651733077AE2000000_8C04133581540082046C1E3B01FD0C10046D25111B3C0C7864622835|+1918
```

BIBOLV, 2024-12-28:
Hi,
HA addon json template is not created for minomess driver, but as messages from your device is mostly general, you can use lse_07_17 driver for example and it has json template.

crpld, 2024-12-30:
Hi. I've tried lse_07_17 driver and it works good for my needs :).
Thanks!

closed 2024-12-30 by crpld