wmbusmeters/wmbusmeters #1448: could not find driver

EMET1984, 2024-12-29:
### Type of request

New device support

### Meter description

i have new installed HA on a RPI 5, connected a iU891A-XL IMST M-Bus dongle to read the Kamstub Flow iq 2200, but i can not find any telegrams due the message unknown driver.. se some copy past out of the log. could you guide me pls?

--------------
Started config iu891a[00202087] on /dev/ttyACM0 listening on t1,c1
(meter) Wasser: meter detection could not find driver for id: 53203349 mfct: (KAW) Kamstrup, Denmark (0x2c37) type: Cold water meter (0x16) ver: 0x42
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (Wasser 53203349 auto) did not handle telegram!

-----------------------
[16:02:41] INFO: Starting wmbusmeters service.
[16:02:42] INFO: CONFIG_CONF ...
[16:02:42] INFO: {"device":"/dev/ttyACM0:iu891a:c1,t1","donotprobe":"/dev/ttyACM0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[16:02:42] INFO: CONFIG_METERS ...
[16:02:42] INFO: []
[16:02:42] INFO: Syncing wmbusmeters configuration ...
[16:02:42] INFO: Registering meters ...
[16:02:42] INFO: Generating MQTT configuration ... 
[16:02:42] WARNING: No internal MQTT service found or configured
[16:02:42] INFO: Running wmbusmeters ...
Started config iu891a[00202087] on /dev/ttyACM0 listening on t1,c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 53203349
          manufacturer: (KAW) Kamstrup, Denmark (0x2c37)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x42
                device: iu891a[00202087]
                  rssi: -51 dBm
                driver: unknown!
Received telegram from: 53203349
          manufacturer: (KAW) Kamstrup, Denmark (0x2c37)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x42
                device: iu891a[00202087]
                  rssi: -51 dBm
                driver: unknown!
[image]
[image]
[image]

### Logged telegrams from meter in decrypted format

```shell
n/a no logs available
```

BIBOLV, 2024-12-29:
Firstly verify that 53203349 is your meter, then you need encryption key, when you have all that - press Add New meter button and enter all details, then messages from meter will be decrypted and information can be extracted.

EMET1984, 2024-12-29:
thx for your comment, attached i have you some pics after i add the meter. 
[image]
[image]
[image]
thanks again for helping

BIBOLV, 2024-12-29:
add to config `logtelegrams=true` and in logs will be printed actual telegrams received from meter. Paste few of them here.

EMET1984, 2024-12-30:
good morning BIBOLV, ohh how dumb i couldnt see this, and sure it makes sense, i did that entry in the editor in the config but did not realise its not writing in the addon entry ui. at this moment it seems working fine. may i ask you on last thing pls? in the mqtt i receive now also some values but there are some values (in the red cyrcle in the pic) who doesent show up, can i modifie them or activate them ? thank you so much and have a great day, elia
[image]

BIBOLV, 2024-12-30:
It depends from what meter actually sends and what is configured in json template you use. If all data is not sent from meter, if I'm not mistaken, you can change meter card to only show values you have/want.

EMET1984, 2024-12-31:
Dear BIBOLV
Thank you for your support and inputs. I really appreciate it, and it has been very helpful to me. I wish you a wonderful day and all the best in the New Year. Happy New Year! 🎉

Best regards from Switzerland

BIBOLV, 2024-12-31:
All the best to you too!

closed 2024-12-31 by BIBOLV