wmbusmeters/wmbusmeters #1450: Is my water meter encrypted? IZAR RC 868 i W R4

kmlZ, 2024-12-31:
### Type of request

Question

### OS version

x

### wmbusmeters version

x

### Your message goes here

Hello, i'm struggling to read data from my water meter. I'm able to recive telegrams but when i try to analyze them in [wmbusmeters.org](https://wmbusmeters.org/analyze/1944304C83A940219801A2411700136BFD4E0200184DFB8D8497) they don't contain any utility data. I've tried to use NOKEY and other default keys but with same effect.
Can someone tell me if telegrams are simply encrypted or not?

BIBOLV, 2024-12-31:
Hi, post few telegrams.

kmlZ, 2024-12-31:
Thank you for reply! Here are more telegrams:
1944304C83A940219801A2711700136BC93FD8FD7A3991E0AF90 
1944304C83A940219801A2111700136BBBDC6D07BED1453AF99F 
1944304c83a940219801a2411700136Bf04e0200184dfB8d8497
1944304C83A940219801A2611700136BDE1091565BEA483BB692 
1944304C83A940219801A2711700136BC93FD8FD7A3991E0AF90 
1944304C83A940219801A2111700136BBBDC6D07BED1453AF99F 
1944304C83A940219801A2211700136B82ADB7FADCA52F57D298

kmlZ, 2024-12-31:
I put antenna outside and listen for telegrams from other devices for a while. Every single telegram don't contain data as before. Some long ones are encrypted according to [wmbusmeters.org](https://wmbusmeters.org/analyze/6E4401066023600005077A590060857BF544C0565EA58984AA692454E5BD0B6699A6A4D193606348FF9EBF28998E4C22BC9128E193A852159A26154DF36E8159243382B36DAA5A7DAF69F591B7EFBF16C3BEFFA9C7A8732CB66F4D87A3D3AB0485E16F13767D63C5BBC1E422501378) output. I will try to do tournee around my neighborhood with laptop and listen for more telegrams. I attache pastebin with filtred telegrams (device ID and telegram). https://pastebin.com/fj8iv3Nt

I'm using rtl-sdr NOT ONLY TV dongle (RTL2838UHIDIR)

kmlZ, 2025-01-02:
I did tournee around my city and found out that some izar water meters can be deocded without key. Here are two telegrams, mine that i'm unable to decode and other not encrypted.

Mine encrypted water meter
```
telegram=|1944304C499540219801A2_11160013517B20A7B979AF019D6D2C|

Auto driver    : izar
Similar driver : unknown 00/00
Using driver   : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 304c dll-mfct (SAP)
004   : 49954021 dll-id (21409549)
008   : 00 dll-version
009   : 07 dll-type (Water meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 11160013517B20A7B979AF019D6D2C mfct specific

{
    "_":"telegram",
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"21409549",
    "timestamp":"2025-01-02T09:37:26Z"
}
``` 

Other IZAR not encrypted 

```
Telegram: 1944304C31C231219801A231150013FE4DF91E057791B28C7C3A
Wynik:
Auto driver    : izar
Similar driver : unknown 00/00
Using driver   : izar 00/00
000   : 19 length (25 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 304c dll-mfct (SAP)
004   : 31c23121 dll-id (2131c231)
008   : 00 dll-version
009   : 07 dll-type (Water meter)
010   : a2 tpl-ci-field (Mfct specific)
011 C?: 31150013FE4DF91E057791B28C7C3A mfct specific

{
    "_":"telegram",
    "media":"water",
    "meter":"izar",
    "name":"",
    "id":"2131c231",
    "last_month_total_m3":127.16,
    "remaining_battery_life_y":10.5,
    "total_m3":127.16,
    "transmit_period_s":8,
    "current_alarms":"no_alarm",
    "last_month_measure_date":"2024-12-31",
    "manufacture_year":"2020",
    "prefix":"C20FA",
    "previous_alarms":"no_alarm",
    "serial_number":"038193",
    "timestamp":"2025-01-02T09:30:10Z"
}
``` 

It seems to me that my water meter is just encrypted and i have to contact my utility provider to get key. I'm closing issue.

closed 2025-01-02 by kmlZ