wmbusmeters/wmbusmeters #1451: Update wmbusmeters version

hngfngzl, 2024-12-31:
### Type of request

Question

### OS version

debian bookworm

### wmbusmeters version

1.16.1-29

### Your message goes here

Hello,

I installed and compiled the version 1.16.1-29.
I this also the way to update to version 1.18.x that ist actual or is there another way to do. 
Better to delete wmbusmeters before downloading and compiling it new?

(Other programs on debian get their updates by apt-get update/upgrade)

I wish a happy new year.

hngfngzl, 2024-12-31:
If my computerskills would be good enough for this I'd do - but as I am a windows child I am struggeling enough to get this running at all.

weetmuts, 2024-12-31:
Ah, what is your setup? Ubuntu, Raspian, something else?

hngfngzl, 2024-12-31:
raspberry pi zero 4b or raspberry pi zero 2 w both running raspian (what i guess is a changed debian) bookworm

weetmuts, 2024-12-31:
It should be easy to build on a raspberry, though it might take ~30 minutes.

sudo apt-get install build-essential
sudo apt-get install librtlsdr-dev libusb-dev libxml2-dev libxslt1-dev
git clone  https://github.com/wmbusmeters/wmbusmeters.git
cd wmbusmeters
./configure
make
sudo make install

Stop the daemon, so that you can test manually:
sudo systemctl stop wmbusmeters 

wmbusmeters --analyze 1844AE4C4455223368077A55000000_041389E20100023B0000
and other manual commands to receive telegrams
If all that works, then you can can configure /etc/wmbusmeters.conf and add your meters in /etc/wmbusmeters.d

Good Luck and Happy New Year!

hngfngzl, 2024-12-31:
ok, than deleting what i have a and download + new compiling is the solution what I guessed.
compiling on raspberry pi 4 b takes about this time but worked fine last time, so I think will work again. I'll do so the next days.

Thanks for your reply and happy new year to you too.

hngfngzl, 2025-01-01:
Worked fine!

closed 2025-01-01 by hngfngzl