wmbusmeters/wmbusmeters #1452: Unknow Device

Gummibaer, 2024-12-31:
### Type of request

New device support

### Meter description

Qunidis 

The Cold Water Meter are deteced with driver lse_07_17

when i try driver lse_07_17 for warm water, then i get this
```json
{"_":"telegram","media":"warm water","meter":"lse_07_17","name":"WW","id":"12121258","what_date_m3":58.776,"device_date_time":"2024-12-31 17:42","meter_version":"11","what_date":"2024-11-30","timestamp":"2024-12-31T16:39:51Z","device":"cul","rssi_dbm":-61}```

### Logged telegrams from meter in decrypted format

```shell
Received telegram from: 12121258
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Warm Water (30°C-90°C) meter (0x06)
                   ver: 0x17
                device: cul
                  rssi: -74 dBm
                driver: unknown!
telegram=|_2D4465325812121217067A910000000C13049205004C1381680400426CFF2C02BB560000326CFFFF046D09131F3C|+490
```

BIBOLV, 2024-12-31:
For me it shows correctly 
```
$ ./wmbusmeters --format=json 2d4465325812121217067a910000000c13049205004c1381680400426cff2c02BB560000326cffff046d09131f3c WW lse_07_17 12121258 NOKEY
...
{"media":"warm water","meter":"lse_07_17","name":"WW","id":"12121258","due_date_m3":46.881,"total_m3":59.204,"device_date_time":"2024-12-31 19:09","due_date":"2023-12-31","error_code":"OK","error_date":"2127-15-31","timestamp":"2024-12-31T18:58:43Z"}
```

same as [here](https://wmbusmeters.org/analyze/2D4465325812121217067A910000000C13049205004C1381680400426CFF2C02BB560000326CFFFF046D09131F3C).

Gummibaer, 2024-12-31:
it works but i got this message
```
(meter) WW: meter detection did not match the selected driver lse_07_17! correct driver is: unknown!
(meter) Not printing this warning again for id: 12121258 mfct: (LSE) Landis Staefa electronic (0x3265) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x17
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
```

BIBOLV, 2024-12-31:
If you get all necessary data and it is correct, you can ignore that message.

closed 2024-12-31 by Gummibaer