wmbusmeters/wmbusmeters #1453: Raspian/Debian Bookworm in qemu on Windows host - Problems with rtlwmbus

hngfngzl, 2025-01-01:
### Type of request

Question

### OS version

debian bookworm

### wmbusmeters version

1.18.0

### Your message goes here

Hello,

I am running all fine on raspberry pi zero 2 w and raspberry pi 4b. But this are stationary solutions. I'd like to have a portable one on my windows notebook. I gut the image from the sd-card running in qemu and i got the usb rtl-sdr-stick to passthrough into raspian. It is not the fastest solution but faster then on pi zero 2.

I don't geht wmbusmeters running. It brings:
No meters configured. Printing id:s of all telegrams heard!
Warning! Detected rtl_wmbus exit due to stopped data flow. Resetting pipeline!
(wmbus) resetting 00000001:rtlwmbus[00000001]
(wmbus) reset completed 00000001:rtlwmbus[00000001]

A debug protocoll i made to but didn't get it out of the machine yet.

I searched and found #493 and another one:

Running: time /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - | grep -qE '.*'

Found 1 device(s):
  0:  RTLSDRBlog, Blog V4, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
Sampling at 1600000 S/s.
Tuned to 868625000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Signal caught, exiting!
Short write, samples lost, exiting!

User cancel, exiting...

real    0m29.664s
user    0m16.467s
sys     0m2.030s

running: rtl_433 -R 0 -f 868.95M -w - | rtl_wmbus -p s -a

rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 0 out of 223 device decoding protocols [ ]
Found Rafael Micro R828D tuner
r82xx_write: i2c wr failed=-9 reg=1a len=1
r82xx_init: failed=-9
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
[R82XX] PLL not locked!
Tuned to 868.950MHz.
Allocating 15 zero-copy buffers

Running: tl_433 -R 0 -f 868.95M -w -
fills the cli in seconds with whatever it is...

Any ideas?
If wished I'll share my qume settings (which took quite some time to find out)

Best Regards

hngfngzl, 2025-01-06:
I tried /usr/bin/rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | /usr/bin/rtl_wmbus
no reply

I could exctract the debug log:
(wmbusmeters) logging started 2025-01-06 10:51:54 1.18.0-23-g7174759
(wmbusmeters) version: 1.18.0-23-g7174759
(config) store meter files in: "/var/wfiot/log/meter/meterfiles"
(config) using device: rtlwmbus 
(config) number of meters: 0
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	[-D enable direct sampling (default: off)]
	filename (a '-' dumps samples to stdout)

<<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 4041 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 4043 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus
[2025-01-06_10:51:54] Started config rtlwmbus[00000001] listening on s1,t1,c1
(shell) exec (capture output) "/usr/bin/rtl_wmbus"
(shell) arg "-h"
(shell) output: >>>/usr/bin/rtl_wmbus: invalid option -- 'h'
rtl_wmbus: d2be82c-25-g34684e6

Usage /usr/bin/rtl_wmbus:
	-o remove DC offset
	-a accelerate (use an inaccurate atan version)
	-r 0 to disable run length algorithm
	-t 0 to disable time2 algorithm
	-d 2 set decimation rate to 2 (defaults to 2 if omitted)
	-v show used algorithm in the output
	-V show version
	-s receive S1 and T1/C1 datagrams simultaneously. rtl_sdr _MUST_ be set to 868.625MHz (-f 868.625M)
	-p [T,S] to disable processing T1/C1 or S1 mode
	-f exit if incoming data stalled for 5 seconds
	-h print this help
<<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 4049 exited
(rtlwmbus) help /usr/bin/rtl_wmbus: invalid option -- 'h'
rtl_wmbus: d2be82c-25-g34684e6

Usage /usr/bin/rtl_wmbus:
	-o remove DC offset
	-a accelerate (use an inaccurate atan version)
	-r 0 to disable run length algorithm
[... 175 lines cut ...]
(shell) sending SIGTERM to process group 4105
(bgshell) sent sigterm, now waiting for child 4105 to exit.
(bgshell) 4105 terminated due to signal 15
(serialcmd) closed /bin/sh pid=4105 fd=7 (rtlwmbus)
[2025-01-06_10:52:23] (wmbus) resetting 00000001:rtlwmbus[00000001]
(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "ERRFILE=$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus -s -f"
(serial) TIMER thread interrupted
(serialcmd) opened /bin/sh pid 4125 fd 7 (rtlwmbus)
(serial) reset completed
[2025-01-06_10:52:25] (wmbus) reset completed 00000001:rtlwmbus[00000001]
(serial) event loop stopped!
(wmbusmeters) shutting down
(wmbus) deleted rtlwmbus
(serial) closing 1 devices
(shell) sending SIGTERM to process group 4125
(bgshell) sent sigterm, now waiting for child 4125 to exit.
(bgshell) 4125 terminated due to signal 15
(serialcmd) closed /bin/sh pid=4125 fd=7 (rtlwmbus)

Any hints?

hngfngzl, 2025-01-12:
The old Version 1.6. works fine the new Version 1.8. not. I am trying to sort out the errors:

(rtlwmbus) help /usr/bin/rtl_wmbus: invalid option -- 'h'
this has been posted by others too (#1360). Error is not connected to qemu - same problem on raspberry. 

Any chance to correct this?

weetmuts, 2025-01-13:
wmbusmeters uses the -h (help option) to see if rtl_wmbus supports the option "-f".
If rtl_wmbus -h does not work, you might to upgrade rtl_wmbus.
 
Can you manually check if "rtl_wmbus -h" works.

hngfngzl, 2025-01-13:
`rtl_wmbus -h`
brings: 
rtl_wmbus: invalid option -- 'h'
rtl_wmbus: d2be82c-25-g34684e6

Usage rtl_wmbus:
        -o remove DC offset
        -a accelerate (use an inaccurate atan version)
        -r 0 to disable run length algorithm
        -t 0 to disable time2 algorithm
        -d 2 set decimation rate to 2 (defaults to 2 if omitted)
        -v show used algorithm in the output
        -V show version
        -s receive S1 and T1/C1 datagrams simultaneously. rtl_sdr _MUST_ be set to 868.625MHz (-f 868.625M)
        -p [T,S] to disable processing T1/C1 or S1 mode
        -f exit if incoming data stalled for 5 seconds
        -h print this help

so far I'd say this is fine also the result of -h or --h is the same both starting with an error text... Version of rtl_wmbus is up to date. 

When I see the log after this:

(rtlwmbus) using command: ERRFILE=$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus -s -f
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "ERRFILE=$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus -s -f"
(serialcmd) opened /bin/sh pid 2850 fd 7 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams
(serial) waiting for stop
(serial) received ascii "ERRFILE=/tmp/wmbusmeters_rtlsdr.YhOvJTGbFkUY<0A>rtl_wmbus: monitoring flow<0A>2025-01-13T09:49:19+01:00<0A>Found 1 device(s):<0A>  0:  RTLSDRBlog, Blog V4, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>[R82XX] PLL not locked!<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) checkRTLWMBusFrame "ERRFILE=/tmp/wmbusmeters_rtlsdr.YhOvJTGbFkUY<0A>rtl_wmbus: monitoring flow<0A>2025-01-13T09:49:19+01:00<0A>Found 1 device(s):<0A>  0:  RTLSDRBlog, Blog V4, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>[R82XX] PLL not locked!<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "rtl_wmbus: monitoring flow<0A>2025-01-13T09:49:19+01:00<0A>Found 1 device(s):<0A>  0:  RTLSDRBlog, Blog V4, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>[R82XX] PLL not locked!<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "2025-01-13T09:49:19+01:00<0A>Found 1 device(s):<0A>  0:  RTLSDRBlog, Blog V4, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>[R82XX] PLL not locked!<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Found 1 device(s):<0A>  0:  RTLSDRBlog, Blog V4, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>[R82XX] PLL not locked!<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "  0:  RTLSDRBlog, Blog V4, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>[R82XX] PLL not locked!<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "<0A>Using device 0: Generic RTL2832U OEM<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>[R82XX] PLL not locked!<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Using device 0: Generic RTL2832U OEM<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>[R82XX] PLL not locked!<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>[R82XX] PLL not locked!<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
[... cut ...]
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
[2025-01-13_09:49:21] (memory) rss 5103616 peak 5.11 MiB

There is still no meters found. Any further hint?

closed 2025-04-14 by github-actions[bot]