wmbusmeters/wmbusmeters #1456: XMQ hydrocalm4 power enhance

kzajac83, 2025-01-06:
### Type of request

Existing device improvement

### Meter description

Hi 

Have telegram:

https://wmbusmeters.org/analyze/3e44B409418012051a0d8c20f17a9d000020046d0d3126310c0a481878330c13098405000B3B0000000B300000000a5929360a5d94230f6402000000000000

There is field:

```
041   : 0B dif (6 digit BCD Instantaneous value)
042   : 30 vif (Power J/h)
043 C?: 000000

```
but cannot extract power using XMQ, in the definition (BTW in code is "dmatch", not "match", no sure what difference but both not working)  I have:

```
        field {
            name     = power
            quantity = Power
            info     = 'The current power consumption.'
            match {
                measurement_type = Instantaneous
                vif_range        = AnyPowerVIF
            }
        }

```
Try VifRange use PowerW too but also not work, probably need change XQM dictionary UoM for Power?? 

Can someone help?

BR, Krzysiek

### Logged telegrams from meter in decrypted format

```shell
3e44B409418012051a0d8c20f17a9d000020046d0d3126310c0a481878330c13098405000B3B0000000B300000000a5929360a5d94230f6402000000000000
```

weetmuts, 2025-01-11:
Thanks! 

Indeed, a new unit for power that has not been implemented. Well it is now. :-) Fix pushed.

This also surfaced another problem, the printing of the json values is done with %f (to_string(v)) which 
uses 6 decimals after the decimal point by default. This potentially throws away precision since the inner 
calculations are done using double (15-17 digits of significant digits) 

If you provide a 1 J/h value as in your telegram and you provide 1 as the content (your telegram has 0)
then the 1 J/h will translate to 0.0000002777777..... kW and this is rounded to 0.000000  aka 0.

If you really want to track a single J/h then you can always add --calculate_power_jh=power_kw and it will add power_jh=1 to the json so the precision exists before it is printed.

The number printing should be improved to print all the significant digits....
but I did not want to to this today, since this changes almost every single test case I have..... ie
0.777778 becomes 0.77777777777777778 etc.

kzajac83, 2025-01-12:
Thank you, working fine :)

Example telegram with some power value:

`3E44B409418012051A0D8C20017AAD120020046D17302C310C0A208891370C13682006000B3B1801000B310010350A5978710A5D78460F6402010000000000`

BR, Krzysiek

BTW - drop list drivers is not updated from longer time, also changes are not visible in your page:

[image]

PS. Also wiki page link not work after click

[image]

closed 2025-04-13 by github-actions[bot]