wmbusmeters/wmbusmeters #1457: Landis & Gyr T550 Ultraheat - Heat Meter Support

bix2702, 2025-01-07:
### Type of request

Existing device improvement

### Meter description

Hi,

I'm trying to readout our Landis+Gyr T550 Ultraheat by WMBUS using a iu891a 868MHz dongle. The telegrams are AES encrypted, I got the AES-key from my heat provider. I can receive sucessfully telegrams, but the current driver "ultraheat" (nor any other driver) is not decoding the telegram content.

Please find attached an example telegram:
[simulation_t550.txt](https://github.com/user-attachments/files/18332695/simulation_t550.txt)

Without using the key `wmbusmeters --analyze simulation_t550.txt` it correctly shows the meter datetime and vendor.
With the AES-key `wmbusmeters --analyze=0BA9E20C3A134B55D374D0C8E51C8A54 simulation_t550.txt` it shows only nonsense...

Could someone please help me to get the correct output?

I also tried [this](https://wmbusmeters.org/analyze/8344A7323336207204048C2005900F002C2559191600AD518D16B3E553DD7A05006007102F2F3C2200000000C48F0F6D0000E1FF9A115A93099411DA6F0C0B013C9A115E20079411DE6F2F140A3C9B113B2403009411BB6F0006053C9B112D3501009411AD6F000B013C8C0106988800008C011410350200066D26350D2531000274B601:auto:0BA9E20C3A134B55D374D0C8E51C8A54)
to get some information about the telegram.

Kind Regards,
Mario

### Logged telegrams from meter in decrypted format

```shell
8344A7323336207204048C2005900F002C2559191600AD518D16B3E553DD7A05006007102F2F_3C2200000000C48F0F6D0000E1FF9A115A93099411DA6F0C0B013C9A115E20079411DE6F2F140A3C9B113B2403009411BB6F0006053C9B112D3501009411AD6F000B013C8C0106988800008C011410350200066D26350D2531000274B601
```

kzajac83, 2025-01-08:
You provide already decrypted telegram, no need AES key, only need build new driver for your meter.

bix2702, 2025-01-08:
Hmm, that is interesting. I got the code from my heat supplier. That would explain why its only crap if I try to decode.

What information do I need to build a new driver? 

How do I get the format for the different (currently C? unknown) sections:

`Auto driver    : ultraheat
Similar driver : qheat_55_us 16/50
Using driver   : ultraheat 00/00
000   : 83 length (131 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a732 dll-mfct (LUG)
004   : 33362072 dll-id (72203633)
008   : 04 dll-version
009   : 04 dll-type (Heat meter)
010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : 05 ell-acc
013   : 90 afl-ci-field (AFL: Authentication and Fragmentation Sublayer)
014   : 0f afl-len (15)
015   : 002c afl-fc (0 MACInFragment MessCounterInFragment MessControlInFragment LastFragment)
017   : 25 afl-mcl (AES_CMAC_128_8 MessCounter)
018   : 59191600 afl-counter (1448281)
022   : AD518D16B3E553DD afl-mac 8 bytes
030   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
031   : 05 tpl-acc-field
032   : 00 tpl-sts-field (OK)(OK)
033   : 6007 tpl-cfg 0760 (AES_CBC_NO_IV )
035   : 10 tpl-cfg-ext (KDFS=1)
036   : 2f2f already decrypted check bytes
038   : 3C dif (8 digit BCD Value during error state)
039   : 22 vif (On time hours)
040 C?: 00000000
044   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
045   : 8F dife (subunit=0 tariff=0 storagenr=31)
046   : 0F dife (subunit=0 tariff=0 storagenr=511)
047   : 6D vif (Date and time type)
048 C?: 0000E1FF
052   : 9A dif (4 digit BCD Maximum value)
053   : 11 dife (subunit=0 tariff=1 storagenr=2)
054   : 5A vif (Flow temperature 10⁻¹ °C)
055 C?: 9309
057   : 94 dif (32 Bit Integer/Binary Maximum value)
058   : 11 dife (subunit=0 tariff=1 storagenr=2)
059   : DA vif (Flow temperature 10⁻¹ °C)
060   : 6F combinable vif (DateTimeOfDELastEnd)
061 C?: 0C0B013C
065   : 9A dif (4 digit BCD Maximum value)
066   : 11 dife (subunit=0 tariff=1 storagenr=2)
067   : 5E vif (Return temperature 10⁻¹ °C)
068 C?: 2007
070   : 94 dif (32 Bit Integer/Binary Maximum value)
071   : 11 dife (subunit=0 tariff=1 storagenr=2)
072   : DE vif (Return temperature 10⁻¹ °C)
073   : 6F combinable vif (DateTimeOfDELastEnd)
074 C?: 2F140A3C
078   : 9B dif (6 digit BCD Maximum value)
079   : 11 dife (subunit=0 tariff=1 storagenr=2)
080   : 3B vif (Volume flow l/h)
081 C?: 240300
084   : 94 dif (32 Bit Integer/Binary Maximum value)
085   : 11 dife (subunit=0 tariff=1 storagenr=2)
086   : BB vif (Volume flow l/h)
087   : 6F combinable vif (DateTimeOfDELastEnd)
088 C?: 0006053C
092   : 9B dif (6 digit BCD Maximum value)
093   : 11 dife (subunit=0 tariff=1 storagenr=2)
094   : 2D vif (Power 10² W)
095 C?: 350100
098   : 94 dif (32 Bit Integer/Binary Maximum value)
099   : 11 dife (subunit=0 tariff=1 storagenr=2)
100   : AD vif (Power 10² W)
101   : 6F combinable vif (DateTimeOfDELastEnd)
102 C?: 000B013C
106   : 8C dif (8 digit BCD Instantaneous value)
107   : 01 dife (subunit=0 tariff=0 storagenr=2)
108   : 06 vif (Energy kWh)
109 C?: 98880000
113   : 8C dif (8 digit BCD Instantaneous value)
114   : 01 dife (subunit=0 tariff=0 storagenr=2)
115   : 14 vif (Volume 10⁻² m³)
116 C?: 10350200
120   : 06 dif (48 Bit Integer/Binary Instantaneous value)
121   : 6D vif (Date and time type)
122 C!: 26350D253100 ("meter_datetime":"2025-01-05 13:53:38")
128   : 02 dif (16 Bit Integer/Binary Instantaneous value)
129   : 74 vif (Actuality duration seconds)
130 C?: B601`

kzajac83, 2025-01-08:
https://github.com/wmbusmeters/wmbusmeters?tab=readme-ov-file#how-to-add-a-new-driver

There is no documentation, just check existing xmq drivers files and try create your own driver, if something goes wrong you should get info what and where is wrong.

bix2702, 2025-01-08:
But do I need the informations from the heat meter vendor about the content of blocks, or not?

kzajac83, 2025-01-08:
All you need is in the telegram.

bix2702, 2025-01-08:
Ok, I do my best. I'm about to generate a new .xmq derived from other. so far I could decode some values.

Since I seems to be an expert, just a questions:
What should I put into the detect field? My Meter shortcut is "LUG", but what are the two numbers (a9, 04) meaning (I copied from istaheat.xmp.
    
detect {
        mvt = LUG,a9,04
    }

bix2702, 2025-01-08:
ok, now I wrote my own driver. You can find it attached in the zip file.
[lug.zip](https://github.com/user-attachments/files/18346609/lug.zip)

With analyze it shows some values, but volume and temperature is missing:

wmbusmeters --debug --analyze=drivers/src/lug.xmq 8344A7323336207204048C2030900F002C2584491600F57F0E567C34B5ED7A30006007102F2F_3C2200000000C48F0F6D0000E1FF9A115A93099411DA6F0C0B013C9A115E20079411DE6F2F140A3C9B113B2403009411BB6F0006053C9B112D3501009411AD6F000B013C8C0106988800008C011410350200066D0B1E0A28310002745E00

output:

`
Auto driver    : t550
Similar driver : qheat_55_us 16/50
Using driver   : t550 00/00
000   : 83 length (131 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : a732 dll-mfct (LUG)
004   : 33362072 dll-id (72203633)
008   : 04 dll-version
009   : 04 dll-type (Heat meter)
010   : 8c ell-ci-field (ELL: Extended Link Layer I (2 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : 30 ell-acc
013   : 90 afl-ci-field (AFL: Authentication and Fragmentation Sublayer)
014   : 0f afl-len (15)
015   : 002c afl-fc (0 MACInFragment MessCounterInFragment MessControlInFragment LastFragment)
017   : 25 afl-mcl (AES_CMAC_128_8 MessCounter)
018   : 84491600 afl-counter (1460612)
022   : F57F0E567C34B5ED afl-mac 8 bytes
030   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
031   : 30 tpl-acc-field
032   : 00 tpl-sts-field (OK)
033   : 6007 tpl-cfg 0760 (AES_CBC_NO_IV )
035   : 10 tpl-cfg-ext (KDFS=1)
036   : 2f2f already decrypted check bytes
038   : 3C dif (8 digit BCD Value during error state)
039   : 22 vif (On time hours)
040 C?: 00000000
044   : C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
045   : 8F dife (subunit=0 tariff=0 storagenr=31)
046   : 0F dife (subunit=0 tariff=0 storagenr=511)
047   : 6D vif (Date and time type)
048 C?: 0000E1FF
052   : 9A dif (4 digit BCD Maximum value)
053   : 11 dife (subunit=0 tariff=1 storagenr=2)
054   : 5A vif (Flow temperature 10⁻¹ °C)
055 C?: 9309
057   : 94 dif (32 Bit Integer/Binary Maximum value)
058   : 11 dife (subunit=0 tariff=1 storagenr=2)
059   : DA vif (Flow temperature 10⁻¹ °C)
060   : 6F combinable vif (DateTimeOfDELastEnd)
061 C?: 0C0B013C
065   : 9A dif (4 digit BCD Maximum value)
066   : 11 dife (subunit=0 tariff=1 storagenr=2)
067   : 5E vif (Return temperature 10⁻¹ °C)
068 C?: 2007
070   : 94 dif (32 Bit Integer/Binary Maximum value)
071   : 11 dife (subunit=0 tariff=1 storagenr=2)
072   : DE vif (Return temperature 10⁻¹ °C)
073   : 6F combinable vif (DateTimeOfDELastEnd)
074 C?: 2F140A3C
078   : 9B dif (6 digit BCD Maximum value)
079   : 11 dife (subunit=0 tariff=1 storagenr=2)
080   : 3B vif (Volume flow l/h)
081 C?: 240300
084   : 94 dif (32 Bit Integer/Binary Maximum value)
085   : 11 dife (subunit=0 tariff=1 storagenr=2)
086   : BB vif (Volume flow l/h)
087   : 6F combinable vif (DateTimeOfDELastEnd)
088 C?: 0006053C
092   : 9B dif (6 digit BCD Maximum value)
093   : 11 dife (subunit=0 tariff=1 storagenr=2)
094   : 2D vif (Power 10² W)
095 C?: 350100
098   : 94 dif (32 Bit Integer/Binary Maximum value)
099   : 11 dife (subunit=0 tariff=1 storagenr=2)
100   : AD vif (Power 10² W)
101   : 6F combinable vif (DateTimeOfDELastEnd)
102 C?: 000B013C
106   : 8C dif (8 digit BCD Instantaneous value)
107   : 01 dife (subunit=0 tariff=0 storagenr=2)
108   : 06 vif (Energy kWh)
109 C!: 98880000 ("Energy_kwh":8898)
113   : 8C dif (8 digit BCD Instantaneous value)
114   : 01 dife (subunit=0 tariff=0 storagenr=2)
115   : 14 vif (Volume 10⁻² m³)
116 C!: 10350200 ("Volume_m3":235.1)
120   : 06 dif (48 Bit Integer/Binary Instantaneous value)
121   : 6D vif (Date and time type)
122 C!: 0B1E0A283100 ("meter_datetime":"2025-01-08 10:30:11")
128   : 02 dif (16 Bit Integer/Binary Instantaneous value)
129   : 74 vif (Actuality duration seconds)
130 C!: 5E00 ("actuality_duration_s":94)

{
    "_":"telegram",
    "media":"heat",
    "meter":"t550",
    "name":"",
    "id":"72203633",
    "Energy_kwh":8898,
    "Volume_m3":235.1,
    "actuality_duration_s":94,
    "meter_datetime":"2025-01-08 10:30:11",
    "timestamp":"2025-01-08T11:29:22Z"
}
`

I don't understand what I do wrong with the temperatures and the volume (and power). the vif_range, tariff and storage_nr is correct.

would you mind to have a look at it?

kzajac83, 2025-01-08:
> Ok, I do my best. I'm about to generate a new .xmq derived from other. so far I could decode some values.
>
> Since I seems to be an expert, just a questions: What should I put into the detect field? [...]

This is this:

```
002 : a732 dll-mfct (LUG)

008 : 04 dll-version
009 : 04 dll-type (Heat meter)
```

Regarding others have no time to check now....

bix2702, 2025-01-08:
Wonderful! Thx a lot. I investigated some more now.

Now I'm manipulating the driver_ultraheat.cc (It' not possible to make a "new" driver with same LUG,04,04 credentials...). 
I found out how to show the current energy and volume just by adding the StorageNr(2)... The other values (Temperatures, Power etc.) I'm still struggeling...

bix2702, 2025-01-08:
I tried at first to "compile" to lug.xmq to like descriped in the readme, but it didn't worked

kzajac83, 2025-01-14:
Sorry, I was very busy (still I'm but need some break). 

Your code is still this? Or have fresh version because a lot data missing?

`wmbusmeters --analyze=terminal:lug.xmq 8344A7323336207204048C2030900F002C2584491600F57F0E567C34B5ED7A30006007102F2F_3C2200000000C48F0F6D0000E1FF9A115A93099411DA6F0C0B013C9A115E20079411DE6F2F140A3C9B113B2403009411BB6F0006053C9B112D3501009411AD6F000B013C8C0106988800008C011410350200066D0B1E0A28310002745E00
`

```

/* Copyright (C) 2025 Mario Grafe (gpl-3.0-or-later) */
driver {
    name           = t550
    meter_type     = HeatMeter
    default_fields = name,id,status,total_kwh,target_kwh,timestamp
    detect {
        mvt = LUG,04,04
    }
    library {
        use = meter_datetime,actuality_duration_s,on_time_h
    }
    fields {
        field {
            name     = FlowTemp
            quantity = Temperature
            info     = 'The instantaneous Volume Flow Temperature.'
            match {
                measurement_type = Instantaneous
                vif_range        = FlowTemperature
                storage_nr       = 2
                tariff           = 1
            }
        }
  
        field {
            name     = FlowTemp
            quantity = Temperature
            info     = 'The instantaneous Volume Return Flow Temperature.'
            match {
                measurement_type = Instantaneous
                vif_range        = ReturnTemperature
                storage_nr       = 2
                tariff           = 1
            }
        }

        field {
            name     = Energy
            quantity = Energy
            info     = 'The instantaneous Energy in kWh.'
            match {
                measurement_type = Instantaneous
                vif_range        = EnergyWh
                storage_nr       = 2
            }
        }
        
        field {
            name     = Volume
            quantity = Volume
            info     = 'The instantaneous volume in cubic meters (10^-2 m³).'
            match {
                measurement_type = Instantaneous
                vif_range        = AnyVolumeVIF
                storage_nr       = 2
            }
        }

        field {
            name     = Power
            quantity = Power
            info     = 'The instantaneous Power.'
            match {
                measurement_type = Instantaneous
                vif_range        = AnyPowerVIF
                storage_nr       = 2
                tariff           = 1
            }
        }

    }

}

```

bix2702, 2025-01-15:
Hi kzajac83,

thx for you support and sorry for the delay. I'm also very busy currently. I have no newer code. I wrote directly a new driver in c because the compiling from XMQ to the driver was not working for me. I still only can decode same data blocks of thwe telegram.

After checking with the heat provider I found out that they didn''t put the instantanious (current) values to the telegram, but only the monthly values from storage. This is stupid... they using an internal data format for the data telegram which is called "Lo Ext". Anyway, I'm in negotiation to change the meter configuration to add the desired current values for energy.

So, i'm a bit knocked out here. They promised me to have a solution until end of January. I'd like to understand the format of the the XMQ file and to be able to compile it directly to a driver. Since my Landis & Gyr UH50 has the same ID (0x04, 0x04) like the ultraheat, I cannot not duplicate the driver, I have to add may data blocks in the current driver. So, probably I have to code directly the driver in the [driver_ultraheat.cc](https://github.com/wmbusmeters/wmbusmeters/blob/master/src/driver_ultraheat.cc) like I did. Here is what I changed so far in the driver:
`/*
 Copyright (C) 2022-2023 Fredrik Öhrström (gpl-3.0-or-later)

 This program is free software: you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation, either version 3 of the License, or
 (at your option) any later version.

 This program is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 GNU General Public License for more details.

 You should have received a copy of the GNU General Public License
 along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#include"meters_common_implementation.h"

namespace
{
    struct Driver : public virtual MeterCommonImplementation
    {
        Driver(MeterInfo &mi, DriverInfo &di);
    };

    static bool ok = registerDriver([](DriverInfo&di)
    {
        di.setName("ultraheat");
        di.setDefaultFields("name,id,heat_kwh,timestamp");
        di.setMeterType(MeterType::HeatMeter);
        di.addDetection(MANUFACTURER_LUG, 0x04,  0x04);
        di.setConstructor([](MeterInfo& mi, DriverInfo& di){ return shared_ptr<Meter>(new Driver(mi, di)); });
    });

    Driver::Driver(MeterInfo &mi, DriverInfo &di) : MeterCommonImplementation(mi, di)
    {
       /* addOptionalLibraryFields("meter_datetime,fabrication_no"); */
        addOptionalLibraryFields("meter_datetime,actuality_duration_s,on_time_h,fabrication_no");

        addNumericFieldWithExtractor(
            "heat",
            "The total heat energy consumption recorded by this meter.",
            DEFAULT_PRINT_PROPERTIES,
            Quantity::Energy,
            VifScaling::Auto, DifSignedness::Signed,
            FieldMatcher::build()
            .set(MeasurementType::Instantaneous)
            .set(VIFRange::AnyEnergyVIF)
            .set(StorageNr(2))
            );

        addNumericFieldWithExtractor(
            "volume",
            "The total heating media volume recorded by this meter.",
            DEFAULT_PRINT_PROPERTIES,
            Quantity::Volume,
            VifScaling::Auto, DifSignedness::Signed,
            FieldMatcher::build()
            .set(MeasurementType::Instantaneous)
            .set(VIFRange::Volume)
            .set(StorageNr(2))
            );

        addNumericFieldWithExtractor(
            "power",
            "The current power consumption.",
            DEFAULT_PRINT_PROPERTIES,
            Quantity::Power,
            VifScaling::Auto, DifSignedness::Signed,
            FieldMatcher::build()
            .set(MeasurementType::Instantaneous)
            .set(VIFRange::PowerW)
            .set(StorageNr(2))
            );

        addNumericFieldWithExtractor(
            "flow",
            "The current heat media volume flow.",
            DEFAULT_PRINT_PROPERTIES,
            Quantity::Flow,
            VifScaling::Auto, DifSignedness::Signed,
            FieldMatcher::build()
            .set(MeasurementType::Instantaneous)
            .set(VIFRange::VolumeFlow)
            .set(StorageNr(2))
            );

        addNumericFieldWithExtractor(
            "flow",
            "The current forward heat media temperature.",
            DEFAULT_PRINT_PROPERTIES,
            Quantity::Temperature,
            VifScaling::Auto, DifSignedness::Signed,
            FieldMatcher::build()
            .set(MeasurementType::Instantaneous)
            .set(VIFRange::FlowTemperature)
            .set(StorageNr(2))
            );

        addNumericFieldWithExtractor(
            "return",
            "The current return heat media temperature.",
            DEFAULT_PRINT_PROPERTIES,
            Quantity::Temperature,
            VifScaling::Auto, DifSignedness::Signed,
            FieldMatcher::build()
            .set(MeasurementType::Instantaneous)
            .set(VIFRange::ReturnTemperature)
            .set(StorageNr(2))
            );

        addStringFieldWithExtractorAndLookup(
            "status",
            "Meter status.",
            PrintProperty::STATUS | PrintProperty::INCLUDE_TPL_STATUS,
            FieldMatcher::build()
            .set(VIFRange::ErrorFlags),
            Translate::Lookup(
            {
                {
                    {
                        "ERROR_FLAGS",
                        Translate::MapType::BitToString,
                        AlwaysTrigger, MaskBits(0xffff),
                        "OK",
                        {
                            /* TODO */
                        }
                    },
                },
            }));

    }
}
`

closed 2025-04-16 by github-actions[bot]