wmbusmeters/wmbusmeters #1460: name of meters file month is missing

pierrotori76, 2025-01-11:
### Type of request

Question

### OS version

bullseye ARMV7

### wmbusmeters version

2.29.7

### Your message goes here

Hi
i put in my meter files this
meterfiles=/wmbusmeters_data/logs/meter_readings
meterfilesaction=overwrite
meterfilesnaming=name
meterfilestimestamp=day

i saw in the parameters of meterfilestimestamp that month is missing, is it possible to add the parameters month to have water_202501 ?
in my point of view, i could keep the last json of each month
Pierrotori

weetmuts, 2025-01-11:
This was an easy fix. If you specify =month then the files are now named: water_2025-01

Nice to see that someone is using the meterfiles. :-)

weetmuts, 2025-01-11:
Fix pushed. Please test.

pierrotori76, 2025-01-12:
good job , i tested and it's work fine, thank you

closed 2025-04-13 by github-actions[bot]