wmbusmeters/wmbusmeters #1461: Decoding a Siemens RDH10RF Thermostat

pierrotori76, 2025-01-13:
### Type of request

Question

### OS version

bullseye ARMV7

### wmbusmeters version

2.29.7

### Your message goes here

Hi,
i have a thermostat RDH10RF / 433 (https://cdn.plomberie-pro.com/upload/medias/plomberie-pro/fiche-technique/rdh10rf_set.pdf)
i sent the command rtl_433 -f 433M -S unknown and i recevied more file
*** Saving signal to file g001_433M_250k.cu8 (163752 samples, 393216 bytes)
*** Saving signal to file g002_433M_250k.cu8 (40924 samples, 131072 bytes)

if i analyse the file, i have this
rtl_433 -A g001_433M_250k.cu8
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 145 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Test mode active. Reading samples from file: g001_433M_250k.cu8
Detected OOK package    @0.196960s
Analyzing pulses...
Total count:  272,  width: 477.93 ms            (119483 S)
Pulse width distribution:
 [ 0] count:    8,  width: 4592 us [4572;4600]  (1148 S)
 [ 1] count:  264,  width:  316 us [312;320]    (  79 S)
Gap width distribution:
 [ 0] count:  136,  width:  904 us [900;912]    ( 226 S)
 [ 1] count:  128,  width:  292 us [288;300]    (  73 S)
 [ 2] count:    4,  width: 31500 us [31004;32708]       (7875 S)
 [ 3] count:    3,  width: 23480 us [23480;23480]       (5870 S)
Pulse period distribution:
 [ 0] count:    8,  width: 5492 us [5476;5500]  (1373 S)
 [ 1] count:  128,  width:  608 us [608;616]    ( 152 S)
 [ 2] count:  128,  width: 1220 us [1216;1228]  ( 305 S)
 [ 3] count:    4,  width: 31816 us [31320;33024]       (7954 S)
 [ 4] count:    3,  width: 23796 us [23796;23796]       (5949 S)
Pulse timing distribution:
 [ 0] count:    8,  width: 4592 us [4572;4600]  (1148 S)
 [ 1] count:  392,  width:  308 us [288;320]    (  77 S)
 [ 2] count:  136,  width:  904 us [900;912]    ( 226 S)
 [ 3] count:    4,  width: 31500 us [31004;32708]       (7875 S)
 [ 4] count:    3,  width: 23480 us [23480;23480]       (5870 S)
 [ 5] count:    1,  width: 46004 us [46004;46004]       (11501 S)
Level estimates [high, low]:  15881,    132
RSSI: -0.1 dB SNR: 20.8 dB Noise: -20.9 dB
Frequency offsets [F1, F2]:   -7196,      0     (-27.5 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB030060111F0013403887B0C5BB8B3B48291929292919291919291919192929291919191919191919291919191919191929355+AAB052060311F0013403887B0C5BB8B3B4829291919192919292919292929191919292929292929292919292929292929291948291929292919291919291919192929291919191919191919291919191919191929355+AAB030060111F0013403887B0C5BB8B3B48292919191929192929192929291919192929292929292929192929292929292919555
Attempting demodulation... short_width: 316, long_width: 4592, reset_limit: 32712, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=316,l=4592,r=32712,g=916,t=1710,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 8
[00] {34} 7f ff ff ff c0 : 01111111 11111111 11111111 11111111 11
[01] {34} 7f ff ff ff c0 : 01111111 11111111 11111111 11111111 11
[02] {34} 7f ff ff ff c0 : 01111111 11111111 11111111 11111111 11
[03] {34} 7f ff ff ff c0 : 01111111 11111111 11111111 11111111 11
[04] {34} 7f ff ff ff c0 : 01111111 11111111 11111111 11111111 11
[05] {34} 7f ff ff ff c0 : 01111111 11111111 11111111 11111111 11
[06] {34} 7f ff ff ff c0 : 01111111 11111111 11111111 11111111 11
[07] {34} 7f ff ff ff c0 : 01111111 11111111 11111111 11111111 11

and second file
rtl_433 -A g002_433M_250k.cu8
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 145 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Test mode active. Reading samples from file: g002_433M_250k.cu8
Detected OOK package    @0.163984s
Analyzing pulses...
Total count:    1,  width: 0.06 ms              (   14 S)
Pulse width distribution:
 [ 0] count:    1,  width:   56 us [56;56]      (  14 S)
Gap width distribution:
Pulse period distribution:
Pulse timing distribution:
 [ 0] count:    1,  width:   56 us [56;56]      (  14 S)
 [ 1] count:    1,  width: 10004 us [10004;10004]       (2501 S)
Level estimates [high, low]:   1034,    104
RSSI: -12.0 dB SNR: 10.0 dB Noise: -22.0 dB
Frequency offsets [F1, F2]:    1290,      0     (+4.9 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
view at https://triq.org/pdv/#AAB102003827148155

how can i check, that the file is for the RDH10RF thermostat ?
Pierrotori

weetmuts, 2025-01-14:
It does not seem to be wmbus? Perhaps you can open an issue with the rtl_433 team? https://github.com/merbanan/rtl_433/

@merbanan

pierrotori76, 2025-01-15:
exactly, but when i start the container , how can i stop the process rtl_sdr and rtl_wmbus to test and execute manualy another frequency ?

inside the container, i have theses process
PID   USER     TIME  COMMAND
    1 root      0:00 sh /wmbusmeters/docker-entrypoint.sh
    7 root      0:12 /wmbusmeters/wmbusmeters --useconfig=/wmbusmeters_data
   12 root      0:00 /bin/sh -c rtl_sdr -f 868.9M -s 1600000 - 2>/dev/null | rtl_wmbus -p s -a
   13 root     22:38 rtl_sdr -f 868.9M -s 1600000 -
   14 root      7h54 rtl_wmbus -p s -a

closed 2025-04-16 by github-actions[bot]