wmbusmeters/wmbusmeters #1470: Engelmann Faw is not decoded correctly

dimitarbosovski, 2025-01-22:
I have a telegram with known key - ffffffffffffffffffffffffffffffff
The telegram is as follows:
4146c5144921604403077224861831c51400072f2020052511c20890692f40d7e1B2f122a8BBe9fcB53569aBaaaa36ffdB148B836205d8
the thing is that first, I get message Not a valid meter driver "engelmannfaw"
and second, when I set to auto, I get message 
Auto driver    : engelmann-faw
Similar driver : c5isf 09/09
Using driver   : engelmann-faw 00/00
000   : 36 length (54 bytes)
001   : 46 dll-c (from meter SND_IR)
002   : c514 dll-mfct (EFE)
004   : 49216044 dll-id (44602149)
008   : 03 dll-version
009   : 07 dll-type (Water meter)
010   : 72 tpl-ci-field (EN 13757-3 Application Layer (long tplh))
011   : 24861831 tpl-id (31188624)
015   : c514 tpl-mfct (EFE)
017   : 00 tpl-version
018   : 07 tpl-type (Water meter)
019   : 2f tpl-acc-field
020   : 20 tpl-sts-field ()(UNKNOWN_20)
021   : 2005 tpl-cfg 0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
023   : 2f2f decrypt check bytes (OK)
025   : 04 dif (32 Bit Integer/Binary Instantaneous value)
026   : 6D vif (Date and time type)
027 C?: 0A2B3531
031   : 04 dif (32 Bit Integer/Binary Instantaneous value)
032   : 13 vif (Volume l)
033 C?: 81000000
037   : 01 dif (8 Bit Integer/Binary Instantaneous value)
038   : FD vif (Second extension FD of VIF-codes)
039   : 17 vife (Error flags (binary))
040 C!: 00 ("status":"UNKNOWN_20")
041   : 2F skip
042   : 2F skip
043   : 2F skip
044   : 2F skip
045   : 2F skip
046   : 2F skip
047   : 2F skip
048   : 2F skip
049   : 2F skip
050   : 2F skip
051   : 2F skip
052   : 2F skip
053   : 2F skip
054   : 2F skip

{
    "_":"telegram",
    "media":"water",
    "meter":"engelmann-faw",
    "name":"",
    "id":"31188624",
    "status":"UNKNOWN_20",
    "timestamp":"2025-01-21T22:48:59Z"
}

https://wmbusmeters.org/analyze/4146c5144921604403077224861831c51400072f2020052511c20890692f40d7e1B2f122a8BBe9fcB53569aBaaaa36ffdB148B836205d8:auto:ffffffffffffffffffffffffffffffff

dimitarbosovski, 2025-01-28:
Here is python code that can decode correctly the total_m3 at least def 

decode_telegram(hex_data):
    # Convert hex string to byte array
    data = bytearray.fromhex(hex_data)

    decoded = {}

    # Length of the telegram
    decoded["length"] = data[0]

    # Manufacturer ID
    manufacturer = data[2:4].hex()
    decoded["manufacturer"] = "EFE" if manufacturer == "c514" else manufacturer

    # Transmitter ID (reverse order)
    transmitter_id = data[4:8][::-1].hex()
    decoded["transmitter_id"] = transmitter_id

    # Meter ID (reverse order, extracted from tpl-id field)
    meter_id = data[11:15][::-1].hex()
    decoded["meter_id"] = meter_id

    # Meter type
    meter_type = data[9]
    decoded["type"] = "Water meter" if meter_type == 0x07 else f"Unknown ({meter_type})"

    # Status (OK in this example)
    decoded["status"] = "OK"

    # Volume (32-bit integer, reverse order)
    volume_bytes = data[33:37][::-1]
    volume_raw = int(volume_bytes.hex(), 16)
    volume_m3 = volume_raw / 1000  # Convert to cubic meters
    decoded["volume_m3"] = round(volume_m3, 3)

    # Timestamp (Date and time)
  
  
    return decoded

# Example telegram (decoded version)
telegram = "3646C5144921604403077224861831C51400072F2020052F2F046D0A2B353104138100000001FD17002F2F2F2F2F2F2F2F2F2F2F2F2F2F"
decoded_telegram = decode_telegram(telegram)

import json
print(json.dumps(decoded_telegram, indent=4))

{
    "length": 54,
    "manufacturer": "EFE",
    "transmitter_id": "44602149",
    "meter_id": "31188624",
    "type": "Water meter",
    "status": "OK",
    "volume_m3": 0.129
}

danielfaust, 2025-01-30:
I spent last weekend writing some Python code to read out the Engelmann FAW.

https://gist.github.com/danielfaust/3e93300b858022acb364f21b71496ec8#file-funkaufsatzmodul-faw-mbus-py-L209

closed 2025-05-01 by github-actions[bot]