wmbusmeters/wmbusmeters #1472: Not a valid device "iu891a:c1,t1"

blinnibeqiri, 2025-01-23:
### Type of request

Enhancement

### OS version

yocto dunfell

### wmbusmeters version

Latest

### Your message goes here

I have a IU891-XL dongle, but I always get Not a Valid device. For me it never worked. Any help?

weetmuts, 2025-01-23:
You have to post a --debug or loglevel=debug output here. Otherwise it is hard to debug the problem.

dimitarbosovski, 2025-01-24:
I have the same problem. The issue comes when the Active Configuration is set to Scan before wmbusmeters is started. In that case due to coming telegrams, ping,getDeviceId and other commands cannot be invoked and message Expected iu891a on /dev/ttyACM0 but did not find it! Ignoring tty! comes. 
To workaround this make sure that the Active configuration  Link Mode is OFF before starting wmbusmeters. 
Here is log if anyone would like to see the cause. 

(drivers) added builtin driver aquastream
(drivers) added builtin driver elf2
(drivers) added builtin driver elster
(drivers) added builtin driver eltako
(drivers) added builtin driver hydrocalm4
(drivers) added builtin driver ime
(drivers) added builtin driver iperl
(drivers) added builtin driver istaheat
(drivers) added builtin driver itronheat
(drivers) added builtin driver izar
(drivers) added builtin driver kampress
(drivers) added builtin driver maddalena
(drivers) added builtin driver relhca
(drivers) added builtin driver sensostar
(drivers) added builtin driver supercal
(drivers) added builtin driver werhlemodwm
(wmbusmeters) version: 1.18.0-26-ga316380
(config) using device: /dev/ttyACM0:iu891a:t1,c1 
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyACM0" iu891a
(serial) check if /dev/ttyACM0 can be accessed
(serial) tty /dev/ttyACM0 can be accessed
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect iu891a)
(serial) received binary "C009209D2D916703FF02A932446850793832546980A0119E3078029023E504180B8F0E50770177885A5A4629000000000000000000000000000101020601B776C0C009209D2D916703FF02AF32446850067546326980A091BF30FE0480231500060521050002000201000305070300000000000000000000000000005982D93FF1C0C009209E2D916703FF02B132446850"
(serial /dev/ttyACM0) sent "C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0"
(serial) EVENT thread interrupted
(serial /dev/ttyACM0) sent "C001030424C0"
(serial) received binary "714845246980A011BF306E0880237C056A0BE80D3F7701777F706B733B000000000000000000000000005A60556592603FC0C009209E2D9167020505AA294468507936450094087AC00920B92D9167028015AF46442D2C74796282400D8D205A22F6A8219EB3657A7FA0793DB49F447178B1E7FECC5B8162B470A601E577B67D4405398C4A21DF4DB9D7B49134993BA02C79BDE5CADBDD48E4F67A8CF9C0"
(iu891a) bad crc got 3f60 expected 9643
(iu891a) are you there? no.

(serial) stopping manager
(serialtty) closed /dev/ttyACM0 (detect iu891a)
(serial) EVENT thread interrupted
(serial) event loop stopped!
Expected iu891a on /dev/ttyACM0 but did not find it! Ignoring tty!
(verbose) ignoring device t1,c1
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(shell) return code 1
(shell) 10114 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) return code 0
(shell) 10115 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) closing 1 devices
(serial) TIMER thread interrupted
(serial) closing 1 devices

falkio8, 2025-01-28:
So is the IU891-XL officially supported for wmbusmeters integration for HA?

Mercatoro, 2025-03-13:
It seems the mode device=**auto**:c1,t1 is detecting the **iu891a** totally fine, but with specific indication it is not detected.

At least this is the situation in my case. See debug log. HIDDEN_ID is a number for the dongle like 00379275. all commented device lines were tried and failed. (except auto:c1,t1)

#### conf:
```
loglevel=debug
device=/dev/ttyACM0:iu891a[HIDDEN_ID]:c1,t1
#device=auto:c1,t1
#device=/dev/ttyACM0:iu891a HIDDEN_ID:t1,c1
#device=/dev/ttyACM0:iu891a\x5B HIDDEN_ID\x5D:t1,c1
#device=/dev/ttyACM0:iu891a%5B HIDDEN_ID%5D:t1,c1
#device=/dev/ttyACM0:iu891a:t1,c1
#device=/dev/ttyACM0:iu891a\[HIDDEN_ID\]:t1,c1
#device=/dev/ttyACM0:auto:c1,t1
#device="/dev/ttyACM0:iu891a[HIDDEN_ID]:c1,t1"
#device=/dev/ttyACM0:usb-IMST_iU891A-if00:t1
#device=/dev/ttyAMA0:iu891a:c1,t1
#device=iu891a:c1,t1
#device=iu891a:t1
#device=/dev/ttyACM0:iu891a:t1
#device=iu891a[HIDDEN_ID]:t1
#device=/dev/ttyACM0:t1
#device=/dev/ttyUSB0:iu891a:c1,t1
#device=iu891a[HIDDEN_ID]:t1
#donotprobe=/dev/ttyAMA0
#donotprobe=all
logtelegrams=true
format=json
meterfiles=/wmbusmeters_data/logs/meter_readings
meterfilesaction=append
logfile=/wmbusmeters_data/logs/wmbusmeters.log
shell=echo "Got reading from $METER_ID: $METER_JSON"
```

#### Log:
```
(wmbusmeters) version: 
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) using device: /dev/ttyACM0:iu891a[HIDDEN_ID]:t1,c1 
(config) number of meters: 0
(lookup) with file/hex "/dev/ttyACM0" iu891a
(serial) check if /dev/ttyACM0 can be accessed
(serial) tty /dev/ttyACM0 can be accessed
(serial) before config: /dev/ttyACM0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(195,79,211,103,125,202,47,42) modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B115200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars(195,79,211,103,16) modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect iu891a)
(serial) received binary "C00920AF17B86703FF02A3294468509341975376F0A0009F316C3201986C32000077000206006BA1007CB2008DC3009ED4000FE500F723C0C00920B017B867020505C52E44FA121229252300077A3B002025A186942D6A2FAD9F4B8EB653DF0D66A4539A25321A615C272B8360D763D561301340C0C00920B917B86703FF02A4294468504849023376F0A0009F31673200986732"
(serial /dev/ttyACM0) sent "C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0C0"
(serial) EVENT thread interrupted
(serial /dev/ttyACM0) sent "C001030424C0"
(serial) EVENT thread interrupted
(serial) received binary "000077000106006BA1007CB2008DC3009ED4000FE50007E5C0C00920C817B86703FF029F294468508841975376F0A0009F316C3201986C320000770400064ABE2F2BBE2F008DC300C00104006E1C050000BE2D0600BF0500007857C0"
(iu891a) bad crc got e507 expected dc7c
(iu891a) are you there? no.
(serialtty) closed /dev/ttyACM0 (detect iu891a)
Expected iu891a on /dev/ttyACM0 but did not find it! Ignoring tty!
(serial) EVENT thread interrupted
(verbose) ignoring device t1,c1
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 10 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>7
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 11 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted
```

#### Log for device=auto:c1,t1
```
...
(serialtty) closed /dev/ttyACM0 (detect iu891a)
(iu891a) are you there? yes 00202071
(main) opening t1,c1
Started auto iu891a[00202071] on /dev/ttyACM0 listening on t1,c1
...
<telegrams coming in normally>
```

closed 2025-06-12 by github-actions[bot]